esphome/issues #4348: watchdog not fed between each redirect in HTTP request

arthurdent75, 2023-03-24:
### The problem

It looks like the watchdog is not fed during HTTP requests. Should be fed between each redirect.

```
[ 28531][V][HTTPClient.cpp:252] beginInternal(): url: ******************************
[ 28547][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: script.google.com port: 443 url: ******************************
[ 28566][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 0
[ 28573][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 210256
[ 28581][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[ 28717][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
[ 28719][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
[ 28723][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[ 28730][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
[ 28738][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
[ 30299][V][ssl_client.cpp:293] start_ssl_client(): Verifying peer X.509 certificate...
[ 30300][V][ssl_client.cpp:301] start_ssl_client(): Certificate verified.
[ 30303][V][ssl_client.cpp:316] start_ssl_client(): Free internal heap after TLS 159728
[ 30311][D][HTTPClient.cpp:1156] connect():  connected to script.google.com:443
[ 30319][V][ssl_client.cpp:369] send_ssl_data(): Writing HTTP request with 245 bytes...
[ 32133][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'HTTP/1.1 302 Moved Temporarily'
[ 32134][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Content-Type: text/html; charset=UTF-8'
[ 32140][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Access-Control-Allow-Origin: *'
[ 32148][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Cache-Control: no-cache, no-store, max-age=0, must-revalidate'
[ 32159][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Pragma: no-cache'
[ 32166][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Expires: Mon, 01 Jan 1990 00:00:00 GMT'
[ 32175][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Date: Fri, 24 Mar 2023 21:10:02 GMT'
[ 32185][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Location:******************************
[ 32222][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'X-Content-Type-Options: nosniff'
[ 32230][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'X-Frame-Options: SAMEORIGIN'
[ 32238][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Content-Security-Policy: frame-ancestors 'self''
[ 32248][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'X-XSS-Protection: 1; mode=block'
[ 32256][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Server: GSE'
[ 32263][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Alt-Svc: h3=":443"; ma=2592000,h3-29=":443"; ma=2592000'
[ 32273][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Accept-Ranges: none'
[ 32281][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Vary: Accept-Encoding'
[ 32288][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'
[ 32296][V][HTTPClient.cpp:1250] handleHeaderResponse(): RX: ''
[ 32301][D][HTTPClient.cpp:1307] handleHeaderResponse(): code: 302
[ 32307][D][HTTPClient.cpp:1314] handleHeaderResponse(): Transfer-Encoding: chunked
[ 32315][D][HTTPClient.cpp:628] sendRequest(): sendRequest code=302
[ 32321][D][HTTPClient.cpp:671] sendRequest(): following redirect (dropped to GET/HEAD): ****************************** redirCount: 1
[ 32363][D][HTTPClient.cpp:388] disconnect(): still data in buffer (450), clean up.
[ 32370][D][HTTPClient.cpp:393] disconnect(): tcp keep open for reuse
[ 32376][V][HTTPClient.cpp:252] beginInternal(): url:******************************
[ 32414][D][HTTPClient.cpp:297] beginInternal(): switching host from 'script.google.com' to 'script.googleusercontent.com'. disconnecting first
[ 32426][D][HTTPClient.cpp:388] disconnect(): still data in buffer (450), clean up.
[ 32433][D][HTTPClient.cpp:395] disconnect(): tcp stop
[ 32441][V][ssl_client.cpp:324] stop_ssl_socket(): Cleaning SSL connection.
[ 32446][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: script.googleusercontent.com port: 443 url: ******************************
[ 32484][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 1
[ 32492][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 208740
[ 32499][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[ 32588][V][ssl_client.cpp:149] start_ssl_client(): Seeding the random number generator
[ 32590][V][ssl_client.cpp:158] start_ssl_client(): Setting up the SSL/TLS structure...
[ 32593][D][ssl_client.cpp:179] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[ 32601][V][ssl_client.cpp:257] start_ssl_client(): Setting hostname for TLS session...
[ 32609][V][ssl_client.cpp:272] start_ssl_client(): Performing the SSL/TLS handshake...
E (49369) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (49369) task_wdt:  - loopTask (CPU 1)
E (49369) task_wdt: Tasks currently running:
E (49369) task_wdt: CPU 0: IDLE
E (49369) task_wdt: CPU 1: IDLE
E (49369) task_wdt: Aborting.

abort() was called at PC 0x400f9604 on core 0

Backtrace:0x400838a1:0x3ffbeacc |<-CORRUPTED
```

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

stas-sl, 2023-03-25:
Why not just increasing WDT timeout? https://github.com/esphome/issues/issues/2853#issuecomment-1366086914

I believe all redirects are handled internally by Arduino `HTTPClient` class, so you can't really interject there any code from ESPHome in the middle.

arthurdent75, 2023-03-25:
> Why not just increasing WDT timeout? [#2853 (comment)](https://github.com/esphome/issues/issues/2853#issuecomment-1366086914)

@stas-sl thank you for your answer.
I can't find a way to include in the yaml the code you pointed me to.
I try in different ways, even in lambda, but it won't compile.
Would you be so kind as to give me an example?
Thank you so much!

stas-sl, 2023-03-25:
AFAIK, you can't include `esp_task_wdt.h` in config directly, but you can include your own header that includes it. So you have to create e.g. `my_includes.h` file, put a single include there and place it in the same folder where your config is. Then you can call `esp_task_wdt_init` on boot like this:

```yaml
esphome:
  includes:
    - my_includes.h

  on_boot:
    then:
      - lambda: !lambda |-
          esp_task_wdt_init(30, false);

```

stas-sl, 2023-03-25:
Someone has also created [no_watchdog](https://github.com/epiclabs-uc/esphome-nowatchdog-component) component for this matter. Though probably it a bit overkill for 2 lines of code :)