esphome/issues #4350: H801 RGBWW HA API connection constantly reconnects

Rothammel, 2023-03-24:
### The problem

when the H801 is switched on, the API constantly reconnects. when the H801 is switched off, everything works normally.

[image]

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.3

### What platform are you using?

ESP8266

### Board

H801

### Component causing the issue

RGBWW

### Example YAML snippet

```yaml
esphome:
  name: h801-stube
  friendly_name: h801-stube

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "BqHhqa10yWIx......................JhL10I="

ota:
  password: "8d2e4d3c.................2b6b4d9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "H801-Stube Fallback Hotspot"
    password: "5SZLRbCt5CIq"

captive_portal:

web_server:
  port: 80
    
output:
  - platform: esp8266_pwm
    pin: 12
    #frequency: 1000 Hz
    id: pwm_b
  - platform: esp8266_pwm
    pin: 13
    #frequency: 1000 Hz
    id: pwm_g
  - platform: esp8266_pwm
    pin: 15
    #frequency: 1000 Hz
    id: pwm_r
  - platform: esp8266_pwm
    pin: 14
    #frequency: 1000 Hz
    id: pwm_w1
  - platform: esp8266_pwm
    pin: 4
    #frequency: 1000 Hz
    id: pwm_w2

light:
  - platform: rgbww
    name: "H801 Light"
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    cold_white: pwm_w1
    warm_white: pwm_w2
```

ssieb, 2023-03-24:
Is there anything in the HA logs?

Rothammel, 2023-03-25:
nope

`2023-03-25 09:47:32,084 INFO 304 GET /devices (0.0.0.0) 81.78ms
2023-03-25 09:47:36,864 INFO 304 GET /devices (0.0.0.0) 6.93ms
2023-03-25 09:48:32,096 INFO 200 GET /devices (0.0.0.0) 8.73ms`

ssieb, 2023-03-25:
I think that's the addon log.  You need to find the HA logs.

catalin2402, 2023-04-13:
Got the same issue, after adding 
`    cold_white_color_temperature: 6000 K
    warm_white_color_temperature: 2700 K`
the issue dissapeared