esphome/issues #4353: Failed to compile: error: redefinition of 'uint32_t iteration'

AlexeyHimki, 2023-03-26:
### The problem

After v2023.03 update I got the error. Please see the log below.

### Which version of ESPHome has the issue?

v2023.03.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

neopixelbus

### Example YAML snippet

```yaml
esphome:
  name: vanna-led
  friendly_name: vanna_led

esp8266:
  board: nodemcuv2

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

light:
  - platform: neopixelbus
    type: GRB
    variant: WS2812
    pin: D4
    num_leds: 30
    id: led
    name: "NeoPixel Light"

globals:
  - id: timer_sec
    type: int
    restore_value: no
    initial_value: '0'

script:
  - id: led_timer
    then:
      - repeat:
          count: 3
          then:
          - repeat:
              count: 10
              then:
              - light.addressable_set:
                  id: led
                  range_from: !lambda |-
                      return id(timer_sec);
                  range_to: !lambda |-
                      return id(timer_sec);
                  red: 100%
                  green: 100%
                  blue: 100%
              - lambda: |-
                  id(timer_sec) = id(timer_sec) + 1;
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration config/vanna-led.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing vanna-led (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- NeoPixelBus @ 2.7.3
|   |-- SPI @ 1.0
Compiling .pioenvs/vanna-led/src/main.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/Rgb48Color.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/RgbColor.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/RgbColorBase.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/Rgbw64Color.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/RgbwColor.cpp.o
Compiling .pioenvs/vanna-led/lib0b4/NeoPixelBus/internal/SegmentDigit.cpp.o
Archiving .pioenvs/vanna-led/libFrameworkArduinoVariant.a
Indexing .pioenvs/vanna-led/libFrameworkArduinoVariant.a
Compiling .pioenvs/vanna-led/FrameworkArduino/Crypto.cpp.o
Compiling .pioenvs/vanna-led/FrameworkArduino/Esp-frag.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:206:73: error: redefinition of 'uint32_t iteration'
  206 |   light_addressableset->set_range_from([=](uint32_t iteration, uint32_t iteration) -> int32_t {
      |                                                                ~~~~~~~~~^~~~~~~~~
src/main.cpp:206:53: note: 'uint32_t iteration' previously declared here
  206 |   light_addressableset->set_range_from([=](uint32_t iteration, uint32_t iteration) -> int32_t {
      |                                            ~~~~~~~~~^~~~~~~~~
config/vanna-led.yaml:47:71: error: redefinition of 'uint32_t iteration'
config/vanna-led.yaml:47:51: note: 'uint32_t iteration' previously declared here
config/vanna-led.yaml:52:88: error: redefinition of 'uint32_t iteration'
config/vanna-led.yaml:52:68: note: 'uint32_t iteration' previously declared here
config/vanna-led.yaml:53:4: error: no matching function for call to 'esphome::LambdaAction<unsigned int, unsigned int>::LambdaAction(setup()::<lambda(uint32_t)>)'
In file included from src/esphome.h:39,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h:133:12: note: candidate: 'esphome::LambdaAction<Ts>::LambdaAction(std::function<void(Ts ...)>&&) [with Ts = {unsigned int, unsigned int}]'
  133 |   explicit LambdaAction(std::function<void(Ts...)> &&f) : f_(std::move(f)) {}
      |            ^~~~~~~~~~~~
src/esphome/core/base_automation.h:133:54: note:   no known conversion for argument 1 from 'setup()::<lambda(uint32_t)>' to 'std::function<void(unsigned int, unsigned int)>&&'
  133 |   explicit LambdaAction(std::function<void(Ts...)> &&f) : f_(std::move(f)) {}
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
src/esphome/core/base_automation.h:131:32: note: candidate: 'esphome::LambdaAction<unsigned int, unsigned int>::LambdaAction(const esphome::LambdaAction<unsigned int, unsigned int>&)'
  131 | template<typename... Ts> class LambdaAction : public Action<Ts...> {
      |                                ^~~~~~~~~~~~
src/esphome/core/base_automation.h:131:32: note:   no known conversion for argument 1 from 'setup()::<lambda(uint32_t)>' to 'const esphome::LambdaAction<unsigned int, unsigned int>&'
src/esphome/core/base_automation.h:131:32: note: candidate: 'esphome::LambdaAction<unsigned int, unsigned int>::LambdaAction(esphome::LambdaAction<unsigned int, unsigned int>&&)'
src/esphome/core/base_automation.h:131:32: note:   no known conversion for argument 1 from 'setup()::<lambda(uint32_t)>' to 'esphome::LambdaAction<unsigned int, unsigned int>&&'
Compiling .pioenvs/vanna-led/FrameworkArduino/Esp-version.cpp.o
Compiling .pioenvs/vanna-led/FrameworkArduino/Esp.cpp.o
Compiling .pioenvs/vanna-led/FrameworkArduino/FS.cpp.o
Compiling .pioenvs/vanna-led/FrameworkArduino/FSnoop.cpp.o
In file included from src/esphome/components/globals/globals_component.h:4,
                 from src/esphome.h:8,
                 from src/main.cpp:3:
src/esphome/core/automation.h: In instantiation of 'esphome::TemplatableValue<T, X>::TemplatableValue(F) [with F = setup()::<lambda(uint32_t)>; typename std::enable_if<(! std::is_invocable<F, X ...>::value), int>::type <anonymous> = 0; T = int; X = {unsigned int, unsigned int}]':
src/esphome/components/light/automation.h:159:3:   required from 'void esphome::light::AddressableSet<Ts>::set_range_from(V) [with V = setup()::<lambda(uint32_t)>; Ts = {unsigned int, unsigned int}]'
config/vanna-led.yaml:46:4:   required from here
src/esphome/core/automation.h:30:52: error: cannot convert 'setup()::<lambda(uint32_t)>' to 'int' in initialization
   30 |   TemplatableValue(F value) : type_(VALUE), value_(value) {}
      |                                                    ^~~~~
      |                                                    |
      |                                                    setup()::<lambda(uint32_t)>
src/esphome/core/automation.h: In instantiation of 'esphome::TemplatableValue<T, X>::TemplatableValue(F) [with F = setup()::<lambda(uint32_t)>; typename std::enable_if<(! std::is_invocable<F, X ...>::value), int>::type <anonymous> = 0; T = int; X = {unsigned int, unsigned int}]':
src/esphome/components/light/automation.h:160:3:   required from 'void esphome::light::AddressableSet<Ts>::set_range_to(V) [with V = setup()::<lambda(uint32_t)>; Ts = {unsigned int, unsigned int}]'
config/vanna-led.yaml:48:4:   required from here
src/esphome/core/automation.h:30:52: error: cannot convert 'setup()::<lambda(uint32_t)>' to 'int' in initialization
Archiving .pioenvs/vanna-led/lib0b4/libNeoPixelBus.a
Indexing .pioenvs/vanna-led/lib0b4/libNeoPixelBus.a
Compiling .pioenvs/vanna-led/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pioenvs/vanna-led/FrameworkArduino/HardwareSerial.cpp.o
*** [.pioenvs/vanna-led/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.78 seconds ==========================
```

ssieb, 2023-03-26:
That has nothing to do with the neopixelbus.  It's because you have a nested `repeat:`.

AlexeyHimki, 2023-03-28:
Yes I have. But nested repeats are working fine in other cases.