esphome/issues #4355: pzdemdc: Compile fails

pkejval, 2023-03-26:
### The problem

When compiling component "pzdemdc" compile throws error

```
Compiling /data/test/.pioenvs/test/src/esphome/components/pzemdc/pzemdc.cpp.o
In file included from src/esphome/components/pzemdc/pzemdc.cpp:1:
src/esphome/components/pzemdc/pzemdc.h:34:65: error: expected template-name before '<' token
   34 | template<typename... Ts> class ResetEnergyAction : public Action<Ts...> {
      |                                                                 ^
src/esphome/components/pzemdc/pzemdc.h:34:65: error: expected '{' before '<' token
Compiling /data/test/.pioenvs/test/src/esphome/components/sensor/automation.cpp.o
*** [/data/test/.pioenvs/test/src/esphome/components/pzemdc/pzemdc.cpp.o] Error 1
```

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

pzemdc

### Example YAML snippet

```yaml
esphome:
  name: test

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
ota:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Example configuration entry
uart:
  tx_pin: TX
  rx_pin: RX
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzemdc
    current:
      name: "Amps"
    voltage:
      name: "Volts"
    power:
      name: "Watts"
    update_interval: 1s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/test/.pioenvs/test/src/esphome/components/pzemdc/pzemdc.cpp.o
Compiling /data/test/.pioenvs/test/src/esphome/components/sensor/filter.cpp.o
In file included from src/esphome/components/pzemdc/pzemdc.cpp:1:
src/esphome/components/pzemdc/pzemdc.h:34:65: error: expected template-name before '<' token
   34 | template<typename... Ts> class ResetEnergyAction : public Action<Ts...> {
      |                                                                 ^
src/esphome/components/pzemdc/pzemdc.h:34:65: error: expected '{' before '<' token
*** [/data/test/.pioenvs/test/src/esphome/components/pzemdc/pzemdc.cpp.o] Error 1
========================== [FAILED] Took 7.83 seconds ==========================
```

### Additional information

Worked before (last compile in production was on 1st March 2023)

ssieb, 2023-03-26:
This is fixed in dev, so you have to use the esphome dev version until the next release or you can get the pzemdc component from the dev version as an external component.

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ pzemdc ]
```

If you use that, make sure you remove it after the next release.

closed 2023-03-26 by ssieb

ssieb, 2023-03-27:
This was included in the 2023.3.2 point release, so it's available now.