esphome/issues #4356: LD2410B with Wemos d1 mini

tientmit88, 2023-03-26:
### The problem

The Sensors values always unknown.
Here is my yaml file
esphome:
  name: ld2410
  friendly_name: ld2410

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "m5aZzRXsrN124dxdatrvVYn5Pl8+OnXA0UI5ZbfWngw="

ota:
  password: "d750417a5179786e0b5eefaa5c6304bb"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ld2410 Fallback Hotspot"
    password: "KmL7IrpPuASP"

captive_portal:

uart:
  id: ld2410_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  uart_id: ld2410_uart
  id: ld2410_comp

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target

For the hardware I follow this one
[image]

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.3.6

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Example YAML snippet

```yaml
esphome:
  name: ld2410
  friendly_name: ld2410

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant APIx
api:
  encryption:
    key: "m5aZzRXsrN124dxdatrvVYn5Pl8+OnXA0UI5ZbfWngw="

ota:
  password: "d750417a5179786e0b5eefaa5c6304bb"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ld2410 Fallback Hotspot"
    password: "KmL7IrpPuASP"

captive_portal:

uart:
  id: ld2410_uart
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  uart_id: ld2410_uart
  id: ld2410_comp

# sensor:
#   - platform: ld2410
#     moving_distance:
#       name : Moving Distance
#     still_distance:
#       name: Still Distance
#     moving_energy:
#       name: Move Energy
#     still_energy:
#       name: Still Energy
#     detection_distance:
#       name: Detection Distance
binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

### Anything in the logs that might be useful for us?

```txt
========================= [SUCCESS] Took 17.56 seconds =========================
INFO Successfully compiled program.
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting.....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 48:e7:29:49:46:08
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x00079fff...
Compressed 497104 bytes to 348905...
Wrote 497104 bytes (348905 compressed) at 0x00000000 in 8.3 seconds (effective 480.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[01:40:06]\x90nn$`nr\x8e\x92\x92nrr\x92`p\xf2n\x82r\x8c<bbn\xc2nb\xc4\xf2nn'l`\x90nn$`nr\x8e\x92\x92nrl`r\x92\x92nrl`\x9c\xec$ll`\xfc\x82n\x92`[I][logger:258]: Log initialized
[01:40:07][C][ota:469]: There have been 0 suspected unsuccessful boot attempt\xf3\xffh6h05\x88%l\x83\xa0$\xfc\xe7S\x80@\xc4\xfc\xe7\xff\x80\x82\x80\x88\xc0\xe7\xffA\x80\x80A@\x80\x86\xfc\xefC@\x80@\x88\xc0\xe7S@\x80 \xc0\xe7\xffA\x80@ \xc0\xef\xffA\x80 @\x88\xc0\xe7\xffA@\xc0 \x80\xefC\xc0 \x80\xc0 \xc0\xe7K\x90\xc6\xefK\x90\xc6lAUU>Tp\x88l\x83\x818\xf8Ya\xf8\xb0zx|8|9t%S9\xb4z\xf8hq\033\xca
```

Brolsk, 2023-03-28:
Logging in to give a quick response since i found this while having the same issue myself.

Change from wires from GPIO1 and GPIO3
and put them on GPIO13 (RX2 / D7) and GPIO15 (TX2 /D8)

https://www.mischianti.org/wp-content/uploads/2021/04/WeMos-D1-mini-esp8266-pinout-mischianti-low-1024x548.jpg

After that they should pop right up, adding the yaml im using

`esphome:
  name: esphome-web-b8568a
  friendly_name: mmwave-test

esp8266:
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "7ecVANrTSRvwc/afNXMvZFO/0Rs8VvojBDzapWvKtjk="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-B8568A"
    password: "Q0MhqnLQHp4y"

captive_portal:
    
uart:
  id: uart1
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 30s
  max_move_distance : 6m
  max_still_distance: 0.75m
  g0_move_threshold: 50
  g0_still_threshold: 20
  g1_move_threshold: 50
  g1_still_threshold: 20
  g2_move_threshold: 40
  g2_still_threshold: 40
  g3_move_threshold: 40
  g3_still_threshold: 40
  g4_move_threshold: 40
  g4_still_threshold: 40
  g5_move_threshold: 40
  g5_still_threshold: 40
  g6_move_threshold: 30
  g6_still_threshold: 15
  g7_move_threshold: 30
  g7_still_threshold: 15
  g8_move_threshold: 30
  g8_still_threshold: 15

sensor:
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance
binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target`

tientmit88, 2023-03-29:
Hi @Brolsk I did it. But it doesn't work :((

cybuletz, 2023-04-06:
I have the exact issue, all of them unknown and they were working fine few weeks back.

ssieb, 2023-04-06:
What is your config that isn't working?  The original config that was posted at the top was not correct.  The logger was using the same uart as the device.

@Brolsk there's no point disabling serial logging if you're not using those pins.

cybuletz, 2023-04-06:
See below, this was working fine a while ago, have tried multiple versions from GitHub but everything was useless.
Also tried the advice above, with RX2 and TX2, same issue. The board is a D1 Mini.
Weird enough, if I add a binary sensor for presence, that works. But all the other options related to distance are not. So it seems indeed to be an issue with RX/TX but don't know what.
Tried with 3 sensors, 2 of them also have bluetooth which is working fine when used that way.

The Yaml and the .h file are from here: https://github.com/rain931215/ESPHome-LD2410

esphome:
  name: senzor
  friendly_name: senzor
  includes:
    - ld2410_uart.h
  on_boot:
    priority: 600
    # ...
    then:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setNumbers(maxMovingDistanceRange, maxStillDistanceRange, noneDuration);

esp8266:
  board: d1_mini_lite

# Enable logging
logger:
  baud_rate: 0
# Enable Home Assistant API
api:
  encryption:
    key: "g7k2+PMaSJeBEyIWCgE7gSX+TZzkjiJUqIQiKzOe3ok="

ota:
  password: "84c69e68213511392bb81fcd1b598ec9"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Senzor Fallback Hotspot"
    password: "poJSjKxVePu2"

web_server:
  port: 80
  version: 2
  include_internal: true
  ota: false

captive_portal:
uart:
  id: uart1
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 256000 # Change this according to your setting
  parity: NONE
  stop_bits: 1
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: [0xF8,0xF7,0xF6,0xF5]
      
custom_component:
  - lambda: |-
      return {new LD2410(id(uart1))};
    components:
      - id: ld2410
      
binary_sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->hasTarget,uart_component->hasMovingTarget,uart_component->hasStillTarget,uart_component->lastCommandSuccess};
    binary_sensors:
      - name: "Has Target"
      - name: "Has Moving Target"
      - name: "Has Still Target"
      - name: "Last Command Success"
  
sensor:
  - platform: custom
    lambda: |-
      auto uart_component = static_cast<LD2410 *>(ld2410);
      return {uart_component->movingTargetDistance,uart_component->movingTargetEnergy,uart_component->stillTargetDistance,uart_component->stillTargetEnergy,uart_component->detectDistance};
    sensors:
      - name: "Moving Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Moving Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Still Target Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0
      - name: "Still Target Energy"
        unit_of_measurement: "%"
        accuracy_decimals: 0
      - name: "Detect Distance"
        unit_of_measurement: "cm"
        accuracy_decimals: 0

number:        
  - platform: template
    name: "Max Moving Distance Range"
    id: maxMovingDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(x,id(maxStillDistanceRange).state,id(noneDuration).state);
  - platform: template
    name: "Max Still Distance Range"
    id: maxStillDistanceRange
    min_value: 1
    max_value: 8
    step: 1
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,x,id(noneDuration).state);
  - platform: template
    name: "None Duration"
    id: noneDuration
    min_value: 0
    max_value: 32767
    step: 1
    mode: box
    update_interval: never
    optimistic: true
    set_action:
      - lambda: |-
          auto uart_component = static_cast<LD2410 *>(ld2410);
          uart_component->setMaxDistancesAndNoneDuration(id(maxMovingDistanceRange).state,id(maxStillDistanceRange).state,x);
      
button:
  - platform: template
    name: "Reboot LD2410"
    on_press:
      lambda: 'static_cast<LD2410 *>(ld2410)->reboot();'
  - platform: template
    name: "Turn on config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(true);'
  - platform: template
    name: "Turn off config mode"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setConfigMode(false);'
  - platform: template
    name: "Get config"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->queryParameters();'
  - platform: template
    name: "Set baud rate to 256000"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(7);'
  - platform: template
    name: "Set baud rate to 115200"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(5);'
  - platform: template
    name: "Set baud rate to 9600"
    on_press:
      - lambda: 'static_cast<LD2410 *>(ld2410)->setBaudrate(1);'

RubenKelevra, 2023-04-29:
I think this can be closed as it works as described in the docs:

[image]

cybuletz, 2023-04-29:
> [image:
> Screenshot_2023-04-29-19-16-24-496-edit_io.homeassistant.companion.android.jpg]

as mentioned, for me it's not working. tried with different esp boards,
installed another instance of HA, tried ESPHome beta, tried 3 LD2410B
sensors. Sensors work perfevtly when connected through bluetooth.

So, what is not working for me if you think there is no issue? Also, gave
you the YAML I am using.

And it worked great 2 months ago, the only difference is that I was using a
regular, not bluetooth version of
LD2410.

cmille34, 2023-05-02:
I am in the same boat here it seems with a D1 Mini and LD2410b.  This is what i get in the logs when connected via USB in case it helps anyone?  This is with ESPHome 2023.4.2 even.

[C][uart.arduino_esp8266:059]: Setting up UART bus...
[C][ld2410:033]: Setting up LD2410...

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0005b300
~ld

RubenKelevra, 2023-05-02:
Hey @cybuletz, I'm also using a Bluetooth version of the board (accidentally ordered the wrong ones).

I've confirmed, that my wiring is the same as shown above.

Here's my yaml:

```
esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  friendly_name: $friendly_name

mdns:

time:
  - platform: homeassistant

substitutions:
  devicename: presence-x-room
  friendly_name: "Presence - x Room"

wifi:
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_password
    manual_ip:
      static_ip: !secret ip-presence-x-room
      gateway: !secret wifi_gateway
      subnet: !secret wifi_subnet
      dns1: !secret wifi_dns1
  reboot_timeout: 5min
  output_power: 15dB 
  power_save_mode: LIGHT  
  
logger:
  baud_rate: 0
  esp8266_store_log_strings_in_flash: false

api:
  encryption: 
    key: !secret api_key_presence-x-room
    
ota:
  password: !secret ota_password_presence-x-room

uart:
  id: uart_ld2410
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 1s
  max_move_distance : 6m
  max_still_distance: 6m
  g0_move_threshold: 29
  g0_still_threshold: 20
  g1_move_threshold: 29
  g1_still_threshold: 20
  g2_move_threshold: 8
  g2_still_threshold: 3
  g3_move_threshold: 8
  g3_still_threshold: 3
  g4_move_threshold: 11
  g4_still_threshold: 4
  g5_move_threshold: 11
  g5_still_threshold: 4
  g6_move_threshold: 11
  g6_still_threshold: 4
  g7_move_threshold: 11
  g7_still_threshold: 1
  g8_move_threshold: 11
  g8_still_threshold: 1

sensor:
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    entity_category: "diagnostic"
    update_interval: 1s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 0.2
  - platform: uptime
    name: Uptime
    entity_category: "diagnostic"
    update_interval: 300s
  - platform: wifi_signal
    name: "WiFi Signal"
    accuracy_decimals: 1
    entity_category: "diagnostic"
    update_interval: 5s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 2
  - platform: ld2410
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target

```

RubenKelevra, 2023-05-02:
> So, what is not working for me if you think there is no issue?

Could be the power supply, leading to too little voltage on the TX port if it's raised.

Check the voltage on the 3.3 V pin and try a different power supply. Also make sure to *not* power the device via USB from a computer. Use a dumb usb charger.

Check if you may have accidentally swapped TX/RX ports.

Try debugging the UART and have a look what it's logging via the wireless log.

cybuletz, 2023-05-03:
Just used your YAML, connected the board to a 120W phone charger and the
result is the same.
Any other ideas I could try?

Once again, if I use the same D1 mini just for 5V and GND and connect the
sensor through bluetooth, it works fine.
Also, checked and triple checked TX and RX, totally different DuPont
colours, can't mess them up.

[image: image.png]

În mar., 2 mai 2023 la 21:30, @RubenKelevra ***@***.***> a
scris:

morini3andahalf, 2023-05-05:
I've just received a couple of LD2410C ordered from aliexpress. Experiencing the same issues as posted above when using them with some wemos D1 mini's I have lying around. 

I've tried as many yaml files as I can find (originally using the yaml and .h files from https://github.com/rain931215/ESPHome-LD2410) but I can't get anything off the serial tx/rx. Sensors seem to work fine from android phone via bluetooth. It seems the problems are uart related but I haven't had much time to dig too deeply into it.

I did try changing the GPIO pins to D7/D8 as suggested in a couple of threads, but nothing seems to work.

morini3andahalf, 2023-05-05:
> I've just received a couple of LD2410C ordered from aliexpress. Experiencing the same issues as posted above when using them with some wemos D1 mini's I have lying around.
>
> I've tried as many yaml [...]

So, after some experimentation / trail and error I finally got this working. 

Seems that newer versions of esphome cause problems when using the currently available yaml examples available for the LD2410. I set up a new python venv and installed an older esphome (esphome 2023.3.2) then ran the https://github.com/rain931215/ESPHome-LD2410. It still initially gave "unknown" when using the TX/RX ports on D1 mini, but switching over to D7/D8 as suggested in other threads worked. 

[image]

systemerror-codenotfound, 2023-05-06:
> Logging in to give a quick response since i found this while having the same issue myself.
>
> Change from wires from GPIO1 and GPIO3 and put them on GPIO13 (RX2 / D7) and GPIO15 (TX2 /D8) [...]

Hi

I'm trying to use it, but sometimes after a reboot it hang up and not work anymore

After checking, it looks like we shouldn't use that pin, as if it's high on boot, it won't boot

https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp8266

mag1024, 2023-07-10:
I'm also using the D1 Mini v3.0.0, and, after initial issues with "Unknown", I was able to get it to work.
The key was to switch to the "alternative" UART0 pins (13/15).
When using pins 1/3, I noticed that TX was working (when I changed the perimeters in the config, I could see that they took effect via the BT app), but nothing was ever received. I suspect this may be a hardware problem -- like the [pullup being too strong](https://electronics.stackexchange.com/questions/644753/how-to-bypass-strong-pullup-on-rx-pin-of-wemos-d1-mini-esp8266-to-communicate), or some such.

Anyway, here's the config that I ended up with:

```
logger:
  baud_rate: 0

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

ld2410:
  timeout: 30s
  max_move_distance : 6m
  max_still_distance: 6m
  g0_still_threshold: 0
  ...

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

@Princessalbert -- try changing your board type to `d1_mini`

cybuletz, 2023-07-10:
> I'm also using the D1 Mini v3.0.0, and, after initial issues with "Unknown", I was able to get it to work. The key was to switch to the "alternative" UART0 pins (13/15). When using pins 1/3, I [...]

hello,
can you paste the entire YAML config ?

emanuelebeffa, 2023-09-04:
I tried both using SW and HW UART, but the RX won't work. I can turn on engineering mode and see the reflected change on the BT app using UART0 (GPIO 1 and 3) and disabling logging.

Home Assistant 2023.8.4
ESPHome 2023.8.2

ryanhindinger, 2023-09-19:
I'm having this same issue.  It was working before but has suddenly stopped working with the new versions of ESPHome.  Double checked my wiring and have tried renaming the Uart ID and the TX/RX pins in the config.  Always comes back Unknown.  I have two other LD2410 working just fine on ESPHome but on ESP32-S2 devices.

RubenKelevra, 2023-09-20:
Configuration for this device changed a bit around, which gave me configuration errors.

After converting to the new structure, I pushed the latest version of ESPHome 2023.8.3 and it works fine with the Wemos D1:

[image]

cybuletz, 2023-09-20:
> Configuration for this device changed a bit around, which gave me configuration errors.
> 
> After converting to the new structure, I pushed the latest version of ESPHome 2023.8.3 and it works fine with the Wemos D1:
> 
> [image]

what do you mean by "converting to the new structure" ? Mine is still not working on 2023.8.3

emanuelebeffa, 2023-09-20:
> Configuration for this device changed a bit around, which gave me configuration errors.
> 
> After converting to the new structure, I pushed the latest version of ESPHome 2023.8.3 and it works fine with the Wemos D1:
> 
> [image]

Could you please post your configuration? Thanks

cybuletz, 2023-09-20:
> > Configuration for this device changed a bit around, which gave me configuration errors.
> > After converting to the new structure, I pushed the latest version of ESPHome 2023.8.3 and it works fine with the Wemos D1:
> > [image]
> 
> Could you please post your configuration? Thanks

https://pastebin.com/Uv7RY3hp

Thanks!

RubenKelevra, 2023-09-20:
Sure.

<details>

```yaml

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini
  friendly_name: $friendly_name

mdns:
ld2410:

time:
  - platform: homeassistant

substitutions:
  devicename: presence-right-room
  friendly_name: "Presence - Right Room"

wifi:
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_password
    manual_ip:
      static_ip: !secret ip-presence-right-room
      gateway: !secret wifi_gateway
      subnet: !secret wifi_subnet
      dns1: !secret wifi_dns1
  reboot_timeout: 5min
  output_power: 15dB 
  power_save_mode: LIGHT  
  
logger:
  baud_rate: 0
  esp8266_store_log_strings_in_flash: false

api:
  encryption: 
    key: !secret api_key_presence-right-room
    
ota:
  password: !secret ota_password_presence-right-room

uart:
  id: uart_ld2410
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1

sensor:
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    entity_category: "diagnostic"
    update_interval: 1s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 0.2
  - platform: uptime
    name: Uptime
    entity_category: "diagnostic"
    update_interval: 300s
  - platform: wifi_signal
    name: "WiFi Signal"
    accuracy_decimals: 1
    entity_category: "diagnostic"
    update_interval: 5s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 2
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

switch:
  - platform: ld2410
    bluetooth:
      name: "control bluetooth"

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target

```

</details>

emanuelebeffa, 2023-09-20:
> Sure.
> 
> ```yaml
> esphome:
>   name: $devicename
>   platform: ESP8266
>   board: d1_mini
>   friendly_name: $friendly_name
> 
> mdns:
> ld2410:
> 
> time:
>   - platform: homeassistant
> 
> substitutions:
>   devicename: presence-right-room
>   friendly_name: "Presence - Right Room"
> 
> wifi:
>   networks:
>     ssid: !secret wifi_ssid
>     password: !secret wifi_password
>     manual_ip:
>       static_ip: !secret ip-presence-right-room
>       gateway: !secret wifi_gateway
>       subnet: !secret wifi_subnet
>       dns1: !secret wifi_dns1
>   reboot_timeout: 5min
>   output_power: 15dB 
>   power_save_mode: LIGHT  
>   
> logger:
>   baud_rate: 0
>   esp8266_store_log_strings_in_flash: false
> 
> api:
>   encryption: 
>     key: !secret api_key_presence-right-room
>     
> ota:
>   password: !secret ota_password_presence-right-room
> 
> uart:
>   id: uart_ld2410
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   baud_rate: 256000
>   parity: NONE
>   stop_bits: 1
> 
> sensor:
>   - platform: adc
>     pin: VCC
>     name: "VCC Voltage"
>     entity_category: "diagnostic"
>     update_interval: 1s
>     filters:
>       - exponential_moving_average:
>           alpha: 0.05
>           send_every: 2
>           send_first_at: 2
>       - or:
>         - throttle: 600s
>         - delta: 0.2
>   - platform: uptime
>     name: Uptime
>     entity_category: "diagnostic"
>     update_interval: 300s
>   - platform: wifi_signal
>     name: "WiFi Signal"
>     accuracy_decimals: 1
>     entity_category: "diagnostic"
>     update_interval: 5s
>     filters:
>       - exponential_moving_average:
>           alpha: 0.05
>           send_every: 2
>           send_first_at: 2
>       - or:
>         - throttle: 600s
>         - delta: 2
>   - platform: ld2410
>     light:
>       name: light
>     moving_distance:
>       name : Moving Distance
>     still_distance:
>       name: Still Distance
>     moving_energy:
>       name: Move Energy
>     still_energy:
>       name: Still Energy
>     detection_distance:
>       name: Detection Distance
> 
> switch:
>   - platform: ld2410
>     bluetooth:
>       name: "control bluetooth"
> 
> number:
>   - platform: ld2410
>     timeout:
>       name: timeout
>     light_threshold:
>       name: light threshold
>     max_move_distance_gate:
>       name: max move distance gate
>     max_still_distance_gate:
>       name: max still distance gate
>     g0:
>       move_threshold:
>         name: g0 move threshold
>       still_threshold:
>         name: g0 still threshold
>     g1:
>       move_threshold:
>         name: g1 move threshold
>       still_threshold:
>         name: g1 still threshold
>     g2:
>       move_threshold:
>         name: g2 move threshold
>       still_threshold:
>         name: g2 still threshold
>     g3:
>       move_threshold:
>         name: g3 move threshold
>       still_threshold:
>         name: g3 still threshold
>     g4:
>       move_threshold:
>         name: g4 move threshold
>       still_threshold:
>         name: g4 still threshold
>     g5:
>       move_threshold:
>         name: g5 move threshold
>       still_threshold:
>         name: g5 still threshold
>     g6:
>       move_threshold:
>         name: g6 move threshold
>       still_threshold:
>         name: g6 still threshold
>     g7:
>       move_threshold:
>         name: g7 move threshold
>       still_threshold:
>         name: g7 still threshold
>     g8:
>       move_threshold:
>         name: g8 move threshold
>       still_threshold:
>         name: g8 still threshold
> 
> 
> 
> binary_sensor:
>   - platform: ld2410
>     has_target:
>       name: Presence
>     has_moving_target:
>       name: Moving Target
>     has_still_target:
>       name: Still Target
> ```

Unfortunately it doesn't work for me. I think the issue is not code related, since I receive no data from the RX pin

ryanhindinger, 2023-09-25:
Just created a fresh node with a new D1 Mini and a brand new LD2410 with ONLY the LD2410 configured in the yanl and still cannot get anything out of it.  All sensors remain Unknown.

emanuelebeffa, 2023-10-02:
Today I discovered something really strange, I was testing a new LD2410 sensor (the square one instead of rectangular) on a very old D1 mini clone (the one on the left), and it worked perfectly with a basic configuration. So I moved the sensor on a newer D1 mini clone (the one on the right) and it didn't work using the same setup and pinout. Could it be a chipset related problem? Can someone else test on a different D1 mini?

[image]

RubenKelevra, 2023-10-02:
Have you used the same power supply? Maybe the voltage is just on the edge of triggering the 1s and depending on the chip it's enough or not? 🤔

Just a guess in the blind. I've been running two of those without any issues, but yeah I think they cold be "old" if you regard one year old.

emanuelebeffa, 2023-10-03:
> Have you used the same power supply? Maybe the voltage is just on the edge of triggering the 1s and depending on the chip it's enough or not? 🤔
> 
> Just a guess in the blind. I've been running two of those without any issues, but yeah I think they cold be "old" if you regard one year old.

I just tried using the same 5V 2A power supply, and nothing changed. The newer D1 mini receives no data from the RX pin. I also tried with an original LOLIN D1 mini 3.1.0 without any luck. The old one is about 4 years old, and has the ESP8266MOD chip.

RubenKelevra, 2023-10-03:
So does the pin work at all? 🤔

cybuletz, 2023-10-04:
After many months of search I have discovered the issue, just tried with
both an ESP32-S2 mini and a D1 mini.
THE PROBLEM WAS (be prepared): RX and TX are inversed on the sensors. So
the RX of the sensor should go into the TX of the boards and vice versa.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

În mar., 3 oct. 2023 la 15:18, @RubenKelevra ***@***.***> a
scris:

dandek01, 2023-10-05:
Hi guys. I have d1 mini and HLK-LD2410-V1.3 board. This is code that is working for me. This code was already posted here but original poster forgot to include "ld2410:" in the code so this will not work.

```
esphome:
  name: xxx
  friendly_name: xxx

esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:

wifi:
  ssid: "xxxxx"
  password: xxxx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxx"
    password: "xxx"

captive_portal:
    
uart:
  id: uart_ld2410
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1
  
ld2410:

sensor:
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    entity_category: "diagnostic"
    update_interval: 1s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 0.2
  - platform: uptime
    name: Uptime
    entity_category: "diagnostic"
    update_interval: 300s
  - platform: wifi_signal
    name: "WiFi Signal"
    accuracy_decimals: 1
    entity_category: "diagnostic"
    update_interval: 5s
    filters:
      - exponential_moving_average:
          alpha: 0.05
          send_every: 2
          send_first_at: 2
      - or:
        - throttle: 600s
        - delta: 2
  - platform: ld2410
    light:
      name: light
    moving_distance:
      name : Moving Distance
    still_distance:
      name: Still Distance
    moving_energy:
      name: Move Energy
    still_energy:
      name: Still Energy
    detection_distance:
      name: Detection Distance

switch:
  - platform: ld2410
    bluetooth:
      name: "control bluetooth"

number:
  - platform: ld2410
    timeout:
      name: timeout
    light_threshold:
      name: light threshold
    max_move_distance_gate:
      name: max move distance gate
    max_still_distance_gate:
      name: max still distance gate
    g0:
      move_threshold:
        name: g0 move threshold
      still_threshold:
        name: g0 still threshold
    g1:
      move_threshold:
        name: g1 move threshold
      still_threshold:
        name: g1 still threshold
    g2:
      move_threshold:
        name: g2 move threshold
      still_threshold:
        name: g2 still threshold
    g3:
      move_threshold:
        name: g3 move threshold
      still_threshold:
        name: g3 still threshold
    g4:
      move_threshold:
        name: g4 move threshold
      still_threshold:
        name: g4 still threshold
    g5:
      move_threshold:
        name: g5 move threshold
      still_threshold:
        name: g5 still threshold
    g6:
      move_threshold:
        name: g6 move threshold
      still_threshold:
        name: g6 still threshold
    g7:
      move_threshold:
        name: g7 move threshold
      still_threshold:
        name: g7 still threshold
    g8:
      move_threshold:
        name: g8 move threshold
      still_threshold:
        name: g8 still threshold

binary_sensor:
  - platform: ld2410
    has_target:
      name: Presence
    has_moving_target:
      name: Moving Target
    has_still_target:
      name: Still Target
```

RubenKelevra, 2023-10-05:
> After many months of search I have discovered the issue, just tried with both an ESP32-S2 mini and a D1 mini.
> 
> THE PROBLEM WAS (be prepared): 
> RX and TX are inversed on the sensors. So the RX of the sensor should go into the TX of the boards and vice versa.
> 
> 

Yep that's correct. Transmit goes into receive on the other board and vice versa.

But this can't be the issue here, as some boards work and some not.

RubenKelevra, 2023-10-05:
> Hi guys. I have d1 mini and HLK-LD2410-V1.3 board. This is code that is working for me. This code was already posted here but original poster forgot to include "ld2410:" in the code so this will not work.

@dandek01 I'm sorry but your conclusion is wrong. There was no such requirement at the time the ticket was created. The structure of the config changed recently.

shz-tr-sr, 2023-11-19:
Hi
I connected LD2410 To ESP15-F and it worked fine for 2 weeks. last night I changed the PCB with the same circuit. ofcourse the power changed. after that it doesnt work any more and all sensors is unknown. even I used the old power. it is so strange behaviour.

diederik-maes, 2023-11-20:
I've tried several things since the August update but can't get it to work anymore.
No config seems to work, weird

kriger1965, 2023-12-12:
I also have this issue - no data in HomeAssistant -  do anyone have a solution for? Di you find out if it is a problem to make it run with a D1 Mini - Board ?

michele-ferri-iubenda, 2023-12-14:
+1, tried with 2 different wemos d1 mini (old one and a v3), 2 different types of ld2410 sensors, all sensors report unknown, using Esphome 2023.11.6. Tried compiling the firmware with a macbook, a thinkpad running ubuntu, and even the home assistant green board itself, to no avail.
Meanwhile my friend with exact YML, sensor and esphome version got it working first time he tried.

cybuletz, 2023-12-14:
Make sure you connect it properly to your ESP: RX on the sensor goes to TX
on the D1 mini and viceversa, TX goes to RX.

davidw3591, 2023-12-15:
I was also having troubles setting this up. Using the TX and RX pins on the Wemos D1 mini (v3.0.0) would show no data in HASS.
Changed to Wemos D7 and D8 pins works for me.

`uart:
  tx_pin: D8 # --> LD2410 RX pin
  rx_pin: D7 # --> LD2410 TX pin
  baud_rate: 256000
  parity: NONE
  stop_bits: 1`

michele-ferri-iubenda, 2023-12-18:
@davidw3591 I already tried the alternative UART pins, sadly the wemos doean't even connect to the wifi if I use those.
I enabled the UART logs, and I noticed that using GPIO1 and GPIO3 I receive a stream of zeroes only.

```
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 256000
  parity: NONE
  stop_bits: 1
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

The log:
```
[15:21:47][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
[15:21:47][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
[15:21:48][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
[15:21:48][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
[15:21:48][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
```

michele-ferri-iubenda, 2023-12-22:
I think I solved it at last. It was a baud rate issue. Fiddling with the baud rate in the android app has fixed it.
- still using GPIO1 and GPIO3
- crossed (tx to rx, rx to tx)
- baud rate 256000 in the YML
- baud rate of the logger to zero
- install HLKRadarTool app, connect to the sensor using bluetooth
- in the top right select more > parameters settings
- change the baud rate to 115200
- the sensor will reboot.
- after rebooting, go back to ths ettings, and put it back to 256000
- reboot again
- suddenly data transmission to the wemos has started.

crashnbernstein, 2023-12-24:
I was able to use the sensor on UART0, but you have to disconnect the sensor during programming/flashing, then re-connect the sensor after flashing. And you have to have logging set to disabled ( baud_rate: 0). You wont get serial logs, but can get them wirelessly after flashing.

emanuelebeffa, 2024-01-29:
I got it to work properly using this configuration, and as @crashnbernstein said, I disconnected the sensor while flashing the ESP8266.

```
logger:
 baud_rate: 0

ld2410:

uart:
    tx_pin: GPIO15
    rx_pin: GPIO13
    baud_rate: 256000
    parity: NONE
    stop_bits: 1
```

szrudi, 2024-02-23:
For me setting the baud rate to 115200 fixed the issue. I've also tried 230400 and 256000 with multiple TX/RX pin combinations but nothing worked, I got into a `wdt reset` loop every time. Baud rates under 115200 also worked.

```
uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
```

So I've 
- disconnected the UART pins, 
- set the baud rate on LD2410 using BT in the app
- flash the d1 mini with the same baud rate setting
- reconnect UART pins

And it works! 🎉

tkahn, 2025-02-10:
Hi!

Great thread! It was relieving to see that I was not the only one having trouble with the LD2410 and Wemos D1 Mini.
 
Just like many of the other users that have posted in this thread, I can't get the UART data into ESP Home. The OUT-pin on the LD2410c board that I'm using works fine when I have it connected to D5 | GPIO14. It reports a binary state when motion is detected or not. But getting any data via the UART protocol via the RX/TX pins (GPIO 3 and GPIO 1 on the Wemos D1 Mini) and the LDC2410c board is not working.

I have downloaded [HLKRadarTool for iOS](https://apps.apple.com/us/app/hlkradartool/id1638651152) and successfully connected to the sensor via Bluetooth and the sensor itself is working fine since I see the data I expect in HLKRadarTool.

It's been suggested that using GPIO 13 and GPIO 15 for RX/TX will fix the problem, but when I do this the Wemos D1 Mini becomes unresponsive. It's also been suggested that experimenting with the baud rate of the LD2410 and/or the Wemos D1 Mini could solve the issue, but I've not been successful in my combinations so far. It seems as though some combinations are successful for some users but not for others and sometimes the steps are quite elaborate involving disconnecting/reconnecting pins during different stages in the setup and setting and resetting parameters on the LD2410.

I don't have any successful combination to contribute with myself so I guess I'm just posting this to let other users know that this still is an issue for some users (like me). If anyone can shine some more light on the tricks suggested in this thread - why and why not the may or may not work I'd be very grateful.

Thanks!

dandek01, 2025-02-10:
I just went with esp32-c3-devkitm-1 and added a light sensor to it also. I got great presence sensor with illuminance that works much better, in my opinion, than any off the shell one. But I'm probably bias about that.
Wemos d1 mini is not worth of hassle.

RubenKelevra, 2025-02-10:
@tkahn

The relevant documentation:

>The ESP8266 has two UARTs; the second of which is TX-only. Only a limited set of pins can be used. `UART0` may use either `tx_pin: GPIO1` and `rx_pin: GPIO3`, or `tx_pin: GPIO15` and `rx_pin: GPIO13`. `UART1` must use `tx_pin: GPIO2`. Any other combination of pins will result in use of a software UART.

Software UART can only use very very slow speeds which I'm not sure that the module does even support. So if you have configured something different it basically isn't expected to work.

Can you share your yaml config please, where you tried GPIO 1/3?

RubenKelevra, 2025-02-10:
> I just went with esp32-c3-devkitm-1 and added a light sensor to it also. I got great presence sensor with illuminance that works much better, in my opinion, than any off the shell one. But I'm probably bias about that.
> Wemos d1 mini is not worth of hassle.

I've went a similar road, as the LD2410 does not work reliably enough. It's a hassle to configure in sensitivity and will then still pickup a curtain moving in the wind or someone walking in front of a window, outside of the room, despite well beyond the configured maximum range.

I've switched to the LD2450 and so far it's pretty great. 8 seconds delay for shutoff is enough to keep the lights on, and after a maximum of a minute they turn off reliably.

Previously I've used 160 seconds delay on 3 different Zigbee PIR sensors (Aqara, Ikea and noname tuya) and they consistently turned the lights off while someone was standing in a room with low amount of motion, like cooking or talking.

So I can recommend this module paired with an ESP32.

tkahn, 2025-02-10:
Wow! That was quick! 😃 

@RubenKelevra: I've flashed the board ten times or more with different yaml-files, but this is the one I set up originally and none of the variants I tried have worked.

```
esphome:
  name: radar-detection
  friendly_name: Radar Detection

esp8266:
  board: d1_mini

logger:
  level: DEBUG
  baud_rate: 0

api:
  encryption:
    key: !secret api_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "Radar-Detection Fallback Hotspot"
    password: "qQw29xQNtvMs"

captive_portal:

uart:
  id: my_uart
  baud_rate: 256000
  rx_pin: GPIO3
  tx_pin: GPIO1
  parity: NONE
  stop_bits: 1

ld2410:
  id: my_ld2410
  uart_id: my_uart
  throttle: 1s

sensor:
  - platform: ld2410
    light:
      name: "Radar Light"
    moving_distance:
      name: "Moving Distance"
    still_distance:
      name: "Still Distance"
    moving_energy:
      name: "Moving Energy"
    still_energy:
      name: "Still Energy"
    detection_distance:
      name: "Detection Distance"
    g0:
      move_energy:
        name: "G0 Move Energy"
      still_energy:
        name: "G0 Still Energy"
    g1:
      move_energy:
        name: "G1 Move Energy"
      still_energy:
        name: "G1 Still Energy"

binary_sensor:
  - platform: ld2410
    has_target:
      name: "Presence Target"
    has_moving_target:
      name: "Moving Target"
    has_still_target:
      name: "Still Target"
    out_pin_presence_status:
      name: "OUT Pin Presence Status"

  - platform: gpio
    pin:
      number: GPIO5
      mode: INPUT_PULLUP
    name: "Radar Presence (GPIO)"
    device_class: motion
    on_press:
      then:
        - logger.log: "Radar Presence: Motion detected!"
    on_release:
      then:
        - logger.log: "Radar Presence: No motion detected!"

switch:
  - platform: ld2410
    engineering_mode:
      name: "Engineering Mode"
```

tkahn, 2025-02-11:
@dandek01 
I have an ESP32 as well so perhaps I should try that instead of the Wemos D1 Mini?

dandek01, 2025-02-11:
Sure, try that. I settle for esp32-c3 mini as it is smaller then wroom version, have 3.3v and 5v connection so two sensors can be easily connect to it. It also run on much lower temperature then wroom version, 37 C vs. nearly 80 C on wroom. 
I think that running temperature is very important for sensor that is connected 24/7.
And it also is small enough to fit in some prebuild plastic case that can be found cheap on aliexpress.

dandek01, 2025-02-11:
I swapped esp32 wroom modules for c3-mini. I'm using it in home assistant so it have sensor for internal temperature. Maybe it was around 85 C. But I done swap few years ago so I don't remember correct data. But it was around 80 C.

RubenKelevra, 2025-02-11:
@dandek01 that's extremly unlikely, as the bigger ESP32 have a big heat spreader. If it's not scorching hot, the chip is not at 85°C.

The ESP32 is known to return wrong temperature numbers, see [documentation](https://esphome.io/components/sensor/internal_temperature).

dandek01, 2025-02-11:
It was a case on my esp32 wroom. It was a bit odd to me that this is running so hot. That was a reason I looked for another esp32 board as I thought this was a way too hot.
When I tried esp32 c3 temperature significantly drop so i continued to make sensors around that board.

RubenKelevra, 2025-02-11:
@dandek01 I've added the sensor to my ESP32-S3-Wroom, and it reports 39 °C right after the flashing.

Hope this helps, now back to topic please :)

tkahn, 2025-02-11:
Okay! Here's some good new which are on topic. 😃 
Tonight I managed to get the sensor to report back data via UART using an ESP32 and some modifications to pins and the script.

**Pins:**
Instead of using GPIO 1 and 3 (UART0) on the ESP32 I tried GPIO 16 and 17 (UART2). 

**Script:**
I just lowered the baud rate to 115200 so the settings is as follows:

```
uart:
  id: my_uart
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 115200
  parity: NONE
  stop_bits: 1
```
I think I left the project yesterday after having lowered the baud rate of the sensor to 115200 using the  HLKRadarTool, but I can't say for sure. When you've been tinkering with a lot of variables for a couple of days you lose track of what you have done/tried and the various combinations.

So even though this solves the problem for me at the moment it does not really help others that are trying to use the combination Wemos D1 Mini and LD2410. I'll see if I dare to make some more attempts at that combo, but for the moment I dare not touch anything.

Thanks @RubenKelevra and @dandek01 for taking time to read and reply!

RubenKelevra, 2025-02-11:
@tkahn Cool that you found a solution, hope the LD2410 works better in your setup than in mine. Still got a pile of them laying around 🥲

dandek01, 2025-02-11:
It can be set up. I does require a bit tinkering around with settings, but most of them are in yaml file. In my experience some firmware version doesn't work that well. So it might be better not to update it. 
And position of the radar is very important especially if you put it in some plastic box, like I did.

appleimperio, 2025-02-27:
I just managed to make it work. I was having the same issue using this UART configuration
````
uart:
  id: my_uart
  baud_rate: 256000
  rx_pin: GPIO3
  tx_pin: GPIO1
  parity: NONE
  stop_bits: 1
````
I download the HLKRadarTool in my case for iOs and update the firmware to the latest trial version V. 2.44.24073110 as soon as it restart after the upgrade i got values in ESPhome