esphome/issues #4357: Error: Could not find one of 'package.json' manifest files in the package

bobvanderlinden, 2023-03-26:
### The problem

I'm new to ESPHome and tried to use it for the first time using an Wemos D1 Mini and ESPHome Web.

"Prepare for first use", configuring wifi and discovering the device worked fine.

However, each time I try to install I get the following error:

```
INFO Reading configuration /config/esphome/esphome-web-c3ca43.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esphome-web-c3ca43 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
Error: Could not find one of 'package.json' manifest files in the package
```

I found https://github.com/esphome/issues/issues/4272, but the solution doesn't work for me. I cannot find the directories/files that I need to remove/delete. In the terminal add-on I tried:

```

➜  / find . -name '.platform'      
➜  / find . -name 'packages' 
./usr/lib/python3.10/site-packages/pip/_vendor/urllib3/packages
➜  / find . -name 'package.json'
```

These files/directories are no-where to be found.

I did find a `.esphome` directory:

```
➜  / ls -l /config/esphome/.esphome
total 12
drwxr-xr-x    3 root     root          4096 Mar 26 22:05 build
-rw-r--r--    1 root     root           665 Mar 26 22:05 esphome-web-c3ca43.yaml.json
drwxr-xr-x    2 root     root          4096 Mar 26 22:14 trash
```

I already tried deleting `/config/esphome/.esphome`, but to no avail.

What can I do to get this going?

### Which version of ESPHome has the issue?

2023.3.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP8266

### Board

d1_mini

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-c3ca43
  friendly_name: ESPHome Web c3ca43

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "6jtYmqUPYdBxAytOCbY0xb7HH5NsaDj58Yx+sdQXnEc="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-C3Ca43"
    password: "3uZkTSQnCvvI"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
Logs of the ESP aren't applicable. See output above.
```

ssieb, 2023-03-27:
The involved directory is in the `/data` directory of the addon.  Try re-installing the addon.

bobvanderlinden, 2023-03-27:
Arg, wasn't aware add-ons also had a `/data`. Sorry for the trouble, that did the trick :+1:

closed 2023-03-27 by bobvanderlinden

hobojoe, 2023-04-16:
> Arg, wasn't aware add-ons also had a `/data`. Sorry for the trouble, that did the trick 👍

Hi @bobvanderlinden could you please share the location of the data folder? I have the same issue as you but couldn't find the right folder to delete. Thanks!

bobvanderlinden, 2023-04-16:
The data folder wasn't available (Part of a docker container?). I just re-installed the addon, this also removed the data folder that is part of the issue.

hobojoe, 2023-04-16:
@bobvanderlinden Thanks! Just re-installed the addon but the problem persists :(

No idea what I could do next.