esphome/issues #4358: Updating devices causes them to be turned off

kellyatkinson, 2023-03-27:
### The problem

I use ESPHome in Home Assistant on a Raspberry Pi. I have 9 devices: 7 Athom wifi smart plugs (esp8266) and 2 M5stack devices used as Bluetooth proxies. I have kept up with the ESPHome core updates in Home Assistant. Today I installed 2023.3.2. When I first installed the previous update 2023.3.1 I noticed odd behaviour which is persisting in 2023.3.2. My Athom plugs are left "on" because they control my home appliances including a freezer, dishwasher, and washing machine. In the past, flashing the incremental version updates to these devices did not change their state. However, starting with 2023.3.1, flashing ESPHome updates onto each of the Athom smart plugs caused the plug state to change to "off".

The devices function correctly but the state change is _very annoying_ - does it need to happen in this way?

### Which version of ESPHome has the issue?

2023.3.1 and 2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

DAVe3283, 2023-04-19:
I run ESPHome on a number of smart plugs (Sonoff and Athom). When updating they will _momentarily_ turn off as the plug reboots, but then they restore the expected state from flash within under a second.

Can you share the YAML for the Athom plugs? It could be that you are not telling them to restore to an on state or restore from flash.

DAVe3283, 2023-04-20:
Huh, the restore_mode for that Athom config is `RESTORE_DEFAULT_OFF` and it has `restore_from_flash: true`, and your YAML doesn't appear to do anything that should affect that. So yours should be remembering if it was on or off and restoring that state, so long as it has been constantly on or off for 1 minute or longer.

If you always want them to be on after a reboot/update/power outage, you can edit your substitutions:

```yaml
substitutions:
  name: "plug-dishwasher"
  relay_restore_mode: ALWAYS_ON
```

You could also try `RESTORE_DEFAULT_ON`, but if the root problem here is the flash memory is worn out that might not work either.

If you _always_ want the outlet on, you can also replace the Athom config with a YAML of your own that doesn't even expose the switch, just sets it to on at boot and leaves it. Might be a safer option for something like a freezer that you presumably just want energy monitoring for.

I ended up using a simplified version of the Athom YAML because I didn't like how it was writing energy consumption to the flash memory, wearing it out, when that data is already uploaded to Home Assistant. This is the config I run on my Athom plugs, which is easier on the flash (plus calculates power factor):

```yaml
substitutions:
  name: space-heater
  friendly_name: Space Heater
  update_interval: 15s
  relay_restore_mode: RESTORE_DEFAULT_ON

packages:
  pg03v2:
    url: https://PigLab.ReaperLegion.net/home-automation/hass/esphome/packages/athom.git
    file: PG03V2.yaml

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: !secret space_heater_ota
```

kellyatkinson, 2023-05-04:
Wow, I didn't even know that was possible. Thanks for pointing out how I could use different YAML and the potential effects of writes on the flash memory. How can you tell when it is getting worn out?

In updating ESPHome to 2023.4.4 and the last couple of versions, I haven't had problems with the switch power states any longer. I think this is now resolved.

DAVe3283, 2023-05-04:
If the flash is worn out it would fail to erase or program, and potentially start returning bad data when read. I am not sure how that would be exposed via ESPHome, but would likely just cause unexpected behavior like this. Because the ESP8266 has so little flash to wear level across (if it even does wear leveling), I generally try to minimize what gets written there to maximize how long it will last.

If the new version of ESPHome has your plug working as expected again, you should be able to close this issue yourself.

closed 2023-05-04 by kellyatkinson