esphome/issues #4362: wifi 202 auth failed

klmi, 2023-03-27:
### The problem

esp32cam can't connect to wifi

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

not relevant

### What platform are you using?

ESP32

### Board

Esp32-s alias esp32cam

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esphome:
  name: cam1
  friendly_name: cam-1

  platform: ESP32
  board: esp32cam

# Enable logging
logger:
  level: VERY_VERBOSE
 
# Enable Home Assistant API
api:

ota:

wifi:
  ssid: "IoTNetwork"
  password: "dhRk83v4hd8t6G7s#"
```

### Anything in the logs that might be useful for us?

```txt
[20:32:59][D][wifi:386]: Starting scan...
[20:33:01][ 44072][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 132, Status: 0, Results: 28
[20:33:01][ 44072][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[20:33:01][V][wifi_esp32:453]: Event: WiFi Scan Done status=0 number=28 scan_id=132
[20:33:01][D][wifi:401]: Found networks:
[20:33:01][I][wifi:445]: - 'IoTNetwork' (xx:xx:xx:85:68:C2) ▂▄▆█
[20:33:01][D][wifi:446]:     Channel: 6
[20:33:01][D][wifi:447]:     RSSI: -62 dB
[20:33:01][I][wifi:445]: - 'IoTNetwork' (xx:xx:xx:E3:33:C2) ▂▄▆█
[20:33:01][D][wifi:446]:     Channel: 8
[20:33:01][D][wifi:447]:     RSSI: -67 dB
[20:33:01][I][wifi:257]: WiFi Connecting to 'IoTNetwork'...
[20:33:01][V][wifi:259]: Connection Params:
[20:33:01][V][wifi:260]:   SSID: 'IoTNetwork'
[20:33:01][V][wifi:263]:   BSSID: xx:xx:xx:85:68:C2
[20:33:01][V][wifi:283]:   Password: 'dhRk83v4hd8t6G7s#'
[20:33:01][V][wifi:288]:   Channel: 6
[20:33:01][V][wifi:297]:   Using DHCP IP
[20:33:01][V][wifi:299]:   Hidden: NO
[20:33:01][W][wifi:565]: Connecting to WiFi network failed. Are the credentials wrong?
[20:33:01][I][wifi:257]: WiFi Connecting to 'IoTNetwork'...
[20:33:01][V][wifi:259]: Connection Params:
[20:33:01][V][wifi:260]:   SSID: 'IoTNetwork'
[20:33:01][V][wifi:263]:   BSSID: xx:xx:xx:85:68:C2
[20:33:01][V][wifi:283]:   Password: 'dhRk83v4hd8t6G7s#'
[20:33:01][V][wifi:288]:   Channel: 6
[20:33:02][V][wifi:297]:   Using DHCP IP
[20:33:02][V][wifi:299]:   Hidden: NO
[20:33:02][ 45303][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: IoTNetwork, BSSID: xx:xx:xx:85:68:c2, Reason: 202
[20:33:02][ 45303][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[20:33:02][ 45311][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 202 - AUTH_FAIL
[20:33:02][W][wifi_esp32:494]: Event: Disconnected ssid='IoTNetwork' bssid=xx:xx:xx:85:68:C2 reason='Authentication Failed'
[20:33:04][ 46847][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: IoTNetwork, BSSID: xx:xx:xx:85:68:c2, Reason: 202
[20:33:04][ 46848][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[20:33:04][ 46855][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 202 - AUTH_FAIL
[20:33:04][W][wifi_esp32:494]: Event: Disconnected ssid='IoTNetwork' bssid=xx:xx:xx:85:68:C2 reason='Authentication Failed'
```

### Additional information

there are other esps running the same login wifi credentials (this time with !secret) and everything works out of the box
i also tried the same firmware on another esp-cam which also fails to connect to my wifi.
If i change the config to my phones hotspot (with way simple password) esp wifi connection works like a charm, so guess it's no hardware issue

jotaborrajo, 2023-03-28:
I have what i think is the same issue with this board  [ESP32 CH340C TYPE-C](https://es.aliexpress.com/item/32864722159.html)
my configuration is this one:
```yaml
esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bluetooth-Proxy Fallback Hotspot"
    password: "redacted"

captive_portal:
``` 
I constantly have the following log messages: 

```
[I][wifi:257]: WiFi Connecting to 'mynetwork'...
[V][wifi:259]: Connection Params:
[V][wifi:260]:   SSID: 'mynetwork'
[V][wifi:263]:   BSSID: xx:xx:xx:xx:xx:xx
[V][wifi:283]:   Password: [redacted]
[V][wifi:288]:   Channel: 1
[V][wifi:297]:   Using DHCP IP
[V][wifi:299]:   Hidden: NO
[  8026][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: mynetwork, BSSID: xx:xx:xx:xx:xx:xx, Reason: 2
[  8027][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  8034][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  8041][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[  8048][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[W][wifi_esp32:494]: Event: Disconnected ssid='mynetwork' bssid=[redacted] reason='Auth Expired'
[W][wifi:549]: Error while connecting to network.
[W][wifi:585]: Restarting WiFi adapter...
[V][wifi_esp32:041]: Disabling STA.
[  8096][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[  8097][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[V][wifi_esp32:464]: Event: WiFi STA stop
[I][wifi:257]: WiFi Connecting to 'mynetwork'...
``` 

However, sometimes I see messages like 
'Connecting to WiFi network failed. Are the credentials wrong?' or issues related to 4-ways handshakes.

If I install a Bluetooth proxy using [ESPHome Bluetooth Proxies](https://esphome.github.io/bluetooth-proxies/), then it works fine, but I cannot add it to ESPHHome: the board is flashed again and I receive the mentioned logs.

I have this board for other projects, working fine for months, and OTA also works fine (I updated one by accident recently). Fresh installs seem to have a problem. Furthermore, if I make a hotspot with my phone, then it may connect (not always).

Until now I did the following to make it work:

- Deactivated the captive portal
- Avoid verbose logging
- Disable AP
- Flash the board using [esptool](https://github.com/espressif/esptool)
- Set a static IP address
- Change power
- Use a different USB cable
- Press the BOOT button until "installing" appears

The differences before previous flashes with my other boards: now I do not need to press the BOOT button to flash. Before I always had to do it.

klmi, 2023-03-29:
@evolucja might be but it's definetly related to https://github.com/esphome/issues/issues/4192

I setup a network unsing only wpa2, esp could connect directly, when i use wpa2/wps3 mode it can't connect and logs -202 auth failed-

randybb, 2023-03-29:
Could you test it with ESP-IDF framework? Based on the following code it is not supported on Arduino.
https://github.com/esphome/esphome/blob/dev/esphome/components/wifi/wifi_component_esp32_arduino.cpp#L298
https://github.com/esphome/esphome/blob/dev/esphome/components/wifi/wifi_component_esp_idf.cpp#L493

evolucja, 2023-03-30:
> Could you test it with ESP-IDF framework? Based on the following code it is not supported on Arduino. 

Great! It worked at least for my case: https://github.com/esphome/issues/issues/4368

evolucja, 2023-03-30:
I'm sorry I misled you earlier today :( In the morning by mistake, I used the wrong board. So with some of my boards I still have the issue no matter what platform / security mode I choose.

jotaborrajo, 2023-03-31:
> Could you test it with ESP-IDF framework? Based on the following code it is not supported on Arduino. [...]

Thank you for the tip, but unfortunately, it does not work. I changed the framework but I have the same issues. I have other boards of the same kind working with the framework `arduino` for months without network issues. The problem appeared with recent updates of ESPHome,  when flashing using an USB cable. Besides, my wifi network is not WPA3 (it is WPA2-PSK). 

Besides, my wifi network is not WPA3 (it is WPA2-PSK). I am wondering if I should open a new issue, as the title of this one does not correspond with my problem now.

jotaborrajo, 2023-04-06:
After trying different possibilities (different computers, different flashing methods, different OS), I found that my problem was a hardware issue that was somehow fixed by software in previous versions. I solder a 330 Ohm resistor between IO0 (connected to the button labelled as "BOOT) and a 3V3 pin. All my problems disappeared with all the boards I use. I hope this can help someone else. The resistors have some glue to prevent them from moving, but it is not necessary.
[image]
[image]

closed 2023-08-27 by github-actions[bot]