esphome/issues #4363: PCF8574/PCF8575 switch state change causes all inputs to toggle twice.

Aikhjarto, 2023-03-27:
### The problem

Using the PCF8575 port expander with one pin for a [switch component](https://esphome.io/components/switch/index.html) and other pins as [binary_sensor_component](https://esphome.io/components/binary_sensor/index.html).

Changing the state of the switch component causes all binary_sensor inputs to be toggled twice (briefly being reported inverted and after a few seconds reported correctly again).

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

pcf8574 in mode "pcf8575: true"

### Example YAML snippet

```yaml
pcf8574:
  - id: 'pcf8574_hub'
    address: 0x20
    pcf8575: true

switch:
  - platform: gpio
    name: "PCF Pin 0"
    pin: 
      pcf8574: pcf8574_hub
      number: 0
      mode:
        output: true

binary_sensor:
  - platform: gpio
    name: "K99"
    id: k99
    pin:
      pcf8574: pcf8574_hub
      number: 14
      inverted: true  

  - platform: gpio
    name: "Kabelbruchdetektor"
    id: kb
    pin:
      pcf8574: pcf8574_hub
      number: 15
      inverted: false
```

### Anything in the logs that might be useful for us?

```txt
00:17:46	[D]	[switch:013]	
'PCF Pin 0' Turning ON.
00:17:46	[D]	[switch:056]	
'PCF Pin 0': Sending state ON
00:17:46	[D]	[switch:056]	
'PCF Pin 0': Sending state ON
00:17:46	[D]	[binary_sensor:036]	
'K99': Sending state OFF
00:17:46	[D]	[binary_sensor:036]	
'Kabelbruchdetektor': Sending state ON
00:17:46	[D]	[binary_sensor:036]	
'K99': Sending state ON
00:17:46	[D]	[binary_sensor:036]	
'Kabelbruchdetektor': Sending state OFF
00:17:49	[D]	[switch:017]	
'PCF Pin 0' Turning OFF.
00:17:49	[D]	[switch:056]	
'PCF Pin 0': Sending state OFF
00:17:49	[D]	[binary_sensor:036]	
'K99': Sending state OFF
00:17:49	[D]	[binary_sensor:036]	
'Kabelbruchdetektor': Sending state ON
00:17:50	[D]	[binary_sensor:036]	
'K99': Sending state ON
00:17:50	[D]	[binary_sensor:036]	
'Kabelbruchdetektor': Sending state OFF
```

justlikeef, 2023-04-06:
This is the same thing I'm seeing:
[https://github.com/esphome/issues/issues/4182](url)

justlikeef, 2023-04-06:
I see the setup in the serial logs.  Looks like I start getting:

[5223737][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[VV][i2c.arduino:095]: RX 2 from 20 failed with error 0

after some time of running.  Rebooting makes things work again.

justlikeef, 2023-04-06:
More failure logging.

[VV][i2c.arduino:116]: 0x20 RX 0000
[VV][i2c.arduino:116]: 0x20 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[VV][i2c.arduino:116]: 0x21 RX 0000
[286146][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[VV][i2c.arduino:095]: RX 2 from 20 failed with error 0
[VV][i2c.arduino:116]: 0x20 RX 0000
[VV][i2c.arduino:116]: 0x20 RX 0000
[VV][i2c.arduino:116]: 0x20 RX 007F
[VV][i2c.arduino:116]: 0x20 RX 7F7F
[D][binary_sensor:036]: 'Valve3 Opened Sensor': Sending state ON
[V][json:036]: Attempting to allocate 512 bytes for JSON serialization
[V][json:056]: Size after shrink 84 bytes
[VV][i2c.arduino:116]: 0x20 RX 0000
[287192][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[VV][i2c.arduino:095]: RX 2 from 20 failed with error 0
[V][component:204]: Component gpio.binary_sensor took a long time for an operation (1.01 s).
[V][component:205]: Components should block for at most 20-30ms.
[287216][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
[VV][i2c.arduino:095]: RX 2 from 20 failed with error 0
[VV][i2c.arduino:116]: 0x21 RX 0000

I've reduced the logging to verbose and am watching to see what I can catch.

Aikhjarto, 2023-08-05:
This problem is still present in ESPHome 2023.07.1