esphome/issues #4364: xiaomi_lywsd02 needs a bindkey option

Flagun, 2023-03-27:
### The problem

xiaomi_ble doesn't receive data from lywsd02. It looks that in the new firmware xiaomi started to use bindkeys for this device. The key can't be entered for platform: xiaomi_lywsd02 

### Which version of ESPHome has the issue?

2023-03

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

davidcui79, 2023-08-01:
Just ran into this issue today and took me a while(quite) to figure out this might be the reason why I cannot connect to LYWSD02 from ESP32(ESP32Home in Home Assistant).

Additional data
1. I am able to connect to the LYWSD02 briefly from Telink Flasher v6.2(https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) and got these logs
6:58:12 PM: Searching for devices
6:58:18 PM: Get Advertising MAC (Web Experimental Features!)
6:58:18 PM: Connecting to: LYWSD02MMC
6:58:24 PM: MiVer5, DevID: 0x2542-?, FnCnt: 226, CtrID: 0x5858 Not bound, Standard certification, MAC: A4C138******, Data encrypted
6:58:33 PM: Not found Telink OTA service!
6:58:33 PM: Disconnected.

2. Code on ESP32 
esp32_ble_tracker:

xiaomi_ble:

sensor:
  - platform: xiaomi_lywsd02
    mac_address: "A4:C1:38:**:**:**"
    temperature:
      name: "Basement BR Temperature"
    humidity:
      name: "Basement BR Humidity"
    battery_level:
      name: "Basement BR Battery Level"

Nothing related to this device shows up in ESP32 log.

Xinayder, 2023-08-09:
I tried adding a custom component to my ESPHome installation which is pretty much the LYWSD02 implementation but with a `bindkey` and trying to decrypt the payload if the device uses encryption, however, it didn't work.

I got these BLE info to show up by setting my log level to verbose in my ESP32:

```
[V][esp32_ble_client:114]: [0] [A4:C1:38:4E:12:99] gattc_event_handler: event=6 gattc_if=4
[V][esp32_ble_client:189]: [0] [A4:C1:38:4E:12:99] ESP_GATTC_SEARCH_CMPL_EVT
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: 0x1800
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x1  end_handle: 0x7
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: 0x1801
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x8  end_handle: 0xb
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: 0x180A
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0xc  end_handle: 0x1a
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: EBE0CCB0-7A0A-4B0C-8A1A-6FF2997DA3A6
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x1b  end_handle: 0x50
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: FAFAFA00-FAFA-FAFA-FAFA-FAFAFAFAFAFA
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x51  end_handle: 0x54
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: 0xFE95
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x55  end_handle: 0x74
[V][esp32_ble_client:192]: [0] [A4:C1:38:4E:12:99] Service UUID: 00000100-0065-6C62-2E74-6F696D2E696D
[V][esp32_ble_client:194]: [0] [A4:C1:38:4E:12:99]  start_handle: 0x75  end_handle: 0x7d
```

closed 2024-08-13 by jesserockz