esphome/issues #4368: ESP32-S2 (S2 Mini board) can't connect to WiFi 'Auth Expired'

evolucja, 2023-03-29:
### The problem

Hi, recently I've received another pack of few S2 Mini boards.
Unfortunately they are probably different manufacturer and I can't get them to connect to WiFi.

I uploaded the same .bin as works on my other S2 Mini boards.

However on those new I am receiving 'Auth Expired' error.

Searched all around the internet for a solution but nothing work.

Any idea?

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP32

### Board

S2 mini

### Example YAML snippet

```yaml
esphome:
  name: espek-10
  comment: "xxx"
            
esp32:
  board: lolin_s2_mini
  framework:
    type: arduino  

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.0.0.50
    gateway: 10.0.0.1
    subnet: 255.255.255.0
  power_save_mode: none
    
logger:

api:

debug:
  
ota:
```

### Anything in the logs that might be useful for us?

```txt
[D][wifi:401]: Found networks:
[I][wifi:445]: - 'Apek_2G' [redacted]▂▄▆█
[D][wifi:446]:     Channel: 1
[D][wifi:447]:     RSSI: -59 dB
[D][wifi:450]: - [redacted] [redacted]▂▄▆█
[D][wifi:450]: - [redacted] [redacted]▂▄▆█
[D][wifi:450]: - [redacted] [redacted]▂▄▆█
[D][wifi:450]: - [redacted] [redacted]▂▄▆█
[D][wifi:450]: - [redacted] [redacted]▂▄▆█
[I][wifi:257]: WiFi Connecting to 'Apek_2G'...
[W][wifi_esp32:494]: Event: Disconnected ssid='Apek_2G' bssid=[redacted] reason='Auth Expired'
[W][wifi:549]: Error while connecting to network.
[W][wifi:585]: Restarting WiFi adapter...
[I][wifi:257]: WiFi Connecting to 'Apek_2G'...
[W][wifi_esp32:494]: Event: Disconnected ssid='Apek_2G' bssid=[redacted] reason='Auth Expired'
[W][wifi:549]: Error while connecting to network.
[W][wifi:585]: Restarting WiFi adapter...
```

evolucja, 2023-03-30:
It worked after I set wireless security mode from WPA2-PSK/WPA3-SAE AES to WPA/WPA2-PSK AES.

It also worked when WPA3 is enabled and when I change framework from "arduino" to "esp-idf".

Is there any chance to support WPA3 in "arduino" framework or should I stay with "esp-idf" which doesn't support some of the features like captive portal?

evolucja, 2023-03-30:
As an update, it doesn't work no matter if I use: WPA2-PSK/WPA3-SAE AES or pure WPA2 / WPA3.
It also doesn't work when I changed framework from "arduino" to "esp-idf".

badgerwithagun, 2023-04-03:
Same problem here with an ESP32-S3. esp-if just puts the device into a boot loop. Arduino gives the exact same `Auth Expired` issue you're seeting.

snidera, 2023-04-05:
I have same issue with clone S2-mini from Amazon (Hiletgo brand).  All 3 have the same error (Auth Expired) when connecting to Wifi.  Have not tested on 2023.3.x   There are 2 solutions I've come across 1) don't buy clones (always good advice) and 2) add 1-10uf capacitor between 5V & EN.  10 x 0.1uf in parallel did not solve the wifi issue on 2 & I don't have larger values here.  Gave up ~ 3 weeks ago & used WROOM module until S2/S3/C3 are more mature and/or I purchase Wemos/Lolin items.

NODeeJay, 2023-05-01:
> add 1-10uf capacitor between 5V & EN. 

I can confirm, adding the 0.1uF capacitor on the S2 mini between 3V3 and EN resolves the issue immediately.

closed 2023-09-24 by github-actions[bot]

Lassska, 2023-10-04:
After install tasmota and reset S2 mini, are you setting the ssid/pass? If not... it's in ap mode (you will find a wifi network on the phone or PC with the name tasmota-xxxxxx-yyyy;
I associate you with it preferably from the PC and the browser will automatically open at 192.168.4.1 where you can set wifi now)

Tasmota
Successful WiFi Connection

Redirecting to new device's IP address

[192.168........](http://192.168........./)
[Tasmota 13.1.0 by Theo Arends](https://bit.ly/tasmota)