esphome/issues #4370: Preferences lost on flashing over Serial

RoboMagus, 2023-03-29:
### The problem

I could not find a similar issue or open feature request to address this. This could just as well be a feature request, but here it goes:

When flashing new ESPHome firmware to an ESP32 board it seems a full factory reset is performed, as all my preferences / Non-volatile storage parameters have been reset. It took me a long time to figure out that the resetting of these parameter was caused by the Serial flashing...

It makes sense that for some usecases, especially when changing e.g. the partitions of the flash, a full factory default binary is flashed. However, this may not always be desirable. As in my usecase where Serial debug logging was required (to figure out what was happening _before_ WiFi got connected) it's just really convenient to flash over Serial instead of over WiFi (as that tends to time-out every now and then).

Either;
- This is a bug that needs fixing.
- Or its a new feature: `The option to select between complete flash overwrite, or only FW update when flashing over Serial`

### Which version of ESPHome has the issue?

2023.4.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP32

### Board

ESP32-D1-Mini

### Component causing the issue

Serial Flashing

### Example YAML snippet

```yaml
N/a
```

darkpoet78, 2023-04-09:
This seems like expected behavior and normal during Flashing of any firmware to ESP chips.  Arduino and Tasmota both have this behavior.  I assume that OTA updating doesn't lose settings because they can be held in memory while the flashing occurs.

If there was a vote, I would vote against keeping settings during serial flashing because I expect the memory to be cleared completely.  Annoying but because so much can go wrong... I know this from experience with routers, uncleared memory might lead to certain settings being saved but not fully, thus corrupting other settings, leading to all sorts of unexpected, undiagnosable errors.  A full reset and you don't even have to worry about this.

RoboMagus, 2023-04-10:
> If there was a vote, I would vote against keeping settings during serial flashing because I expect the memory to be cleared completely.

That's why it'd have to be an _option_ that the user could select. With the default ofcourse being the full factory reset. The problem with not having an option to keep these settings is that if you can only flash EspHome over Serial there is just no way to persist these settings at all.

RoboMagus, 2023-08-29:
Keeping this one open. 
Might look into a solution myself if time permits.

rspier, 2024-02-04:
Arduino IDE 2.2.1 has an "Erase All Flash Before Sketch Upload" option that I think does (i.e. doesn't) what you're looking for.