esphome/issues #4371: touchscreen xpt2046 doesn't offer x_raw or y_raw

kdorff, 2023-03-29:
### The problem

The [ESPHome xpt2046 documentation](https://esphome.io/components/touchscreen/xpt2046.html) reports being able to calibrate the panel using `x_raw` and `y_raw` to help determine values for `calibration_x_min`,  `calibration_x_max`,  `calibration_y_min`,  `calibration_y_max`.

(see yaml snippet below)

But when try to use these, I'm given a compilation errors:

error: 'struct esphome::touchscreen::TouchPoint' has no member named 'x_raw'
error: 'struct esphome::touchscreen::TouchPoint' has no member named 'y_raw'

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP32

### Board

ESP32 WROOM-32

### Component causing the issue

touchscreen

### Example YAML snippet

```yaml
touchscreen:
  platform: xpt2046
  id: touch
  cs_pin: 32
  interrupt_pin: 21
  swap_x_y: false
  threshold: 400
  update_interval: 50ms
  report_interval: 1s
  on_touch:
    then:
      - lambda: |-
          ESP_LOGD("ili9341-touch", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
              (id(touch)).x,
              (id(touch)).y,
              (id(touch)).x_raw,
              (id(touch)).y_raw
              );
```

### Anything in the logs that might be useful for us?

```txt
error: 'struct esphome::touchscreen::TouchPoint' has no member named 'x_raw'
error: 'struct esphome::touchscreen::TouchPoint' has no member named 'y_raw'
```

### Additional information

This started happening when we transitioned from using `xpt2046:` to `touchscreen:`.

Unfocused, 2023-04-26:
Could you try `id(xpt_touchscreen).x_raw` instead? Where `xpt_touchscreen` is the ID you're using for the touchscreen component. 

That's what the tests use: https://github.com/esphome/esphome/blob/ef832becf1b9490499fbd9132464b0e53650a841/tests/test4.yaml#L542

If that works, then it's just a documentation issue.

kdorff, 2023-04-27:
> Could you try `id(xpt_touchscreen).x_raw` instead? Where `xpt_touchscreen` is the ID you're using for the touchscreen component.
>
> That's what the tests use: [...]

Is that test actually executed? I am guessing it is not. The test creates the component using
```
xpt2046:
  id: xpt_touchscreen
  ...
```
but this changed some time last year to
```
touchscreen:
  platform: xpt2046
  ...
```
At that time, the [ESPHome xpt2046 documentation page](https://esphome.io/components/touchscreen/xpt2046.html) was updated to reflect the new syntax. 

If you try to use the old syntax you will get an error within the ESPHome editor stating **This component should now be used as platform of the Touchscreen component.** 

As best as I understand, before the xpt2046 moved to `touchscreen:`, the `x_raw` and `y_raw` did work (I used them, in fact). But I do not believe this test is up to date and I don't think it will work anymore since it is using the old syntax.

Finally, if you look at my snippet I do give the component an `id` of `touch` so later when I reference the members of touch, I don't have any issue referencing `id(touch).x` or `id(touch).y`, but if I tried to use the documented `id(touch).x_raw` or `id(touch).y_raw` I find these variables no longer exist. Yes, I could rename my component's `id` to `xpt_touchscreen`  but I don't believe it will still be missing `x_raw` and `y_raw` class members as indicated by
```
error: 'struct esphome::touchscreen::TouchPoint' has no member named 'x_raw'
error: 'struct esphome::touchscreen::TouchPoint' has no member named 'y_raw'
```

nielsnl68, 2023-04-29:
@kdorff , it is true, in the last changes i removed the `x_raw` and `y_raw` values. I'm not sure what would be the benefit be to have those values at hand?

kdorff, 2023-04-29:
Hi @nielsnl68 the values of `x_raw` and `y_raw` are there to enable the user of the touchscreen to perform calibration, as documented in the [ESPHome xtp2046 documentation section on Calibration](https://esphome.io/components/touchscreen/xpt2046.html#calibration). Without these values, there is no way to fine tune the values one would provide to `calibration_x_min`, `calibration_x_max`, `calibration_y_min`, and `calibration_y_max`.

Additionally [the test](https://github.com/esphome/esphome/blob/ef832becf1b9490499fbd9132464b0e53650a841/tests/test4.yaml#L542) mentioned by @Unfocused suggests a `z_raw` value which I assume (?) was originally offered to help tune touch sensitivity?? If that is the case, that value could be useful, also.

If these variables could be restored / re-exposed, it might help people who need to more accurately tune their touchscreen.

kdorff, 2023-04-29:
Reviewing the code for xpt2046, I find that the `x_raw`, `y_raw`, and `z_raw` values are output in `VERY_VERBOSE` logging, when a touch is detected. I suppose this could be used in place of the calibration code in the documentation, but VERY_VEROSE is kind of hostile. I hoped I could set logging such as
```
logger:
  level: DEBUG
  logs: 
    xpt2046: VERY_VERBOSE
```
expecting I could have my logs be **relatively** quite but get the `VERY_VERBOSE` calibration log messages from `xpt2046.` Alas, this gives an error:
```
ERROR Error while reading config: The local log level VERY_VERBOSE for xpt2046 must be less severe than the global log level DEBUG.
```
So I continue to wish these values were again exposed.

Unfocused, 2023-05-09:
Sorry, I miss-read your code.

This confused me:
> error: 'struct esphome::touchscreen::TouchPoint' has no member named 'x_raw'
> error: 'struct esphome::touchscreen::TouchPoint' has no member named 'y_raw'

... because that's the TouchPoint struct, the parameter passed to the `on_touch` handler. Which, AFAICT, is not the error you would get with the code you provided. I mentioned `id(..).x_raw` because `x_raw`/`y_raw` are public members of the `XPT2046Component` class:
https://github.com/esphome/esphome/blob/4b664b6f09a6eef16e380a953973eeda59ea2f19/esphome/components/xpt2046/xpt2046.h#L83

> VERY_VEROSE is kind of hostile

Agreed that those seem like they should be output at DEBUG level, not VERY_VERBOSE. Though, looking at the code, there's a separate instance of the `x_raw`/`y_raw` logged at VERBOSE level (logging raw and processed values).

kdorff, 2023-05-10:
@Unfocused after reading your comment above, I decided to change the `id` of my `touchscreen` component from `touch` to `touchid`. And then below, change my `id(touch).x_raw` and similar calls to use the new id `id(touchid).x_raw`.  And this worked. It seems my problem is naming the it `touch` versus something else.

From this, it feels like my naming the component `touch` is clashing with something else of type `struct esphome::touchscreen::TouchPoint`? Perhaps something internal to the component? I'm confused, as well.

nielsnl68, 2023-05-10:
Cool, so this means this **issue** is fixed?
When so, could you close it then ?

closed 2023-05-10 by kdorff