esphome/issues #4372: ESP32-C3 Second UART not receiving anything

6502mos, 2023-03-29:
### The problem

Hello,

I have issues with second UART not being able to receive any data on ESP32-C3. It's an RS485 module connected via MAX485-like chip to the ESP32-C3 chip with both flow control pins being connected which I control in software, pin assigment is:
```
GPIO5: RX
GPIO6: RE
GPIO7: DE
GPIO8: TX
```

To test the functionality I wrote simple YAML configuration which in two second intervals sends specific 8 byte sequence to the RS485 module which should respond back with 9 bytes, yet nothing is shown on debug log. I have verified with an oscilloscope that there are bytes being sent on TX pin, flow control pins are switched as needed and there is response from the RS485 module of exactly 9 bytes present on the RX pin.

I have also written short Arduino code which accomplishes same thing and it works as expected. I have also tried disabling logging UART, but that has made no difference. 

Here's oscilloscope picture of data being sent on TX pin 8 (flow control switchover should be maybe a little bit later, but the RS485 device seem to care and it's hard to precisely control with ESPHome):
[image]

And here's respose on RX pin 5 which is not logged in ESPHome:
[image]

Thanks in advance.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP32

### Board

custom board

### Component causing the issue

uart

### Example YAML snippet

```yaml
esphome:
  name: "uarttest"
  friendly_name: "uarttest"

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

#
#
#

output:
  - platform: gpio
    pin: GPIO6
    id: RE
  - platform: gpio
    pin: GPIO7
    id: DE

uart:
  tx_pin: GPIO8
  rx_pin: GPIO5
  baud_rate: 9600
  debug:

interval:
  - interval: 2s
    then:
      - output.turn_on: RE
      - output.turn_on: DE
      - delay: 10ms
      - uart.write: [0x01, 0x03, 0x00, 0x00, 0x00, 0x02, 0xC4, 0x0B]
      - delay: 8ms #precise delay to switch flow control back to receiving mode
      - output.turn_off: RE
      - output.turn_off: DE
```

### Anything in the logs that might be useful for us?

```txt
...
[22:27:01][C][logger:293]: Logger:
[22:27:01][C][logger:294]:   Level: DEBUG
[22:27:01][C][logger:295]:   Log Baud Rate: 115200
[22:27:01][C][logger:296]:   Hardware UART: UART0
[22:27:01][C][uart.arduino_esp32:108]: UART Bus 1:
[22:27:01][C][uart.arduino_esp32:109]:   TX Pin: GPIO8
[22:27:01][C][uart.arduino_esp32:110]:   RX Pin: GPIO5
[22:27:01][C][uart.arduino_esp32:112]:   RX Buffer Size: 256
[22:27:01][C][uart.arduino_esp32:114]:   Baud Rate: 9600 baud
[22:27:01][C][uart.arduino_esp32:115]:   Data Bits: 8
[22:27:01][C][uart.arduino_esp32:116]:   Parity: NONE
[22:27:01][C][uart.arduino_esp32:117]:   Stop bits: 1
[22:27:01][C][gpio.output:010]: GPIO Binary Output:
[22:27:01][C][gpio.output:011]:   Pin: GPIO6
[22:27:01][C][gpio.output:010]: GPIO Binary Output:
[22:27:01][C][gpio.output:011]:   Pin: GPIO7
[22:27:01][C][captive_portal:088]: Captive Portal:
[22:27:01][C][mdns:108]: mDNS:
[22:27:01][C][mdns:109]:   Hostname: uarttest
[22:27:01][C][ota:093]: Over-The-Air Updates:
[22:27:01][C][ota:094]:   Address: uarttest.local:3232
[22:27:01][C][ota:097]:   Using Password.
[22:27:01][C][api:138]: API Server:
[22:27:01][C][api:139]:   Address: uarttest.local:6053
[22:27:01][C][api:141]:   Using noise encryption: YES
[22:27:03][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:05][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:07][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:09][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:11][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:13][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:15][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:17][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:19][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:21][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
[22:27:23][D][uart_debug:114]: >>> 01:03:00:00:00:02:C4:0B
...
```

### Additional information

Arduino code which is working:
```
#include <Arduino.h>

void flow(bool dir)
{
  digitalWrite(6, dir);
  digitalWrite(7, dir);
}

void setup()
{
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  Serial.begin(9600);
  Serial1.begin(9600, SERIAL_8N1, 5, 8);
  flow(false);
}

void loop()
{
  char bytes[8] = {0x01, 0x03, 0x00, 0x00, 0x00, 0x02, 0xC4, 0x0B};
  flow(true);
  delay(10);
  Serial1.write(bytes, 8);
  delay(8);
  flow(false);
  Serial.println("Sending modbus bytes...");
  while(Serial1.available()>0)
  {
    Serial.print(Serial1.read(), HEX);
    Serial.print(" ");
  }
  Serial.println();
  delay(2000);
 }

```

ssieb, 2023-03-29:
You don't have anything listening, so you have to enable the dummy receiver.
And why are you implementing it yourself instead of using the existing modbus component?

closed 2023-03-29 by ssieb

6502mos, 2023-03-29:
Thanks, I think I tried the dummy listener before but there must have been some different issues at that time that made me think it doesn't work at all. As why I don't I use the modbus component, it's mainly because there's no easy way of setting two flow control pins (I should've wired them together on the PCB, but too late for that now), if there is then please let me know. Also I'm just sending one command and reading the response, nothing else.