esphome/issues #4376: [FastLED] fastpin_esp32.h:69:24: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1'; did you mean 'out'?

lboue, 2023-03-31:
### The problem

### Issue

When adding an RGB LED in the configuration file, the compilation fails.

- Board: [Wemos C3 pico](https://www.wemos.cc/en/latest/c3/c3_pico.html). A mini wifi & Bluetooth5 (LE) boards based ESP32-C3FH4. 
- 1x WS2812 RGB LED
- FastLED @ 3.3.2 (default version)
- https://www.wemos.cc/en/latest/_static/files/sch_c3_pico_v1.0.0.pdf

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: 7
    num_leds: 1
    rgb_order: GRB
    id: status_led
    name: ESP32 Light
    effects:
      - random:
      - flicker:
      - addressable_rainbow:
```

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.6

### What platform are you using?

ESP8266

### Board

esp32-c3-devkitm-1

### Component causing the issue

WS2812B

### Example YAML snippet

```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: 7
    num_leds: 1
    rgb_order: GRB
    id: status_led
[... 304 lines cut ...]
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.h:52,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.cpp:2:
/data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/fastpin_esp32.h:69:24: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'out1'; did you mean 'out'?
       else return GPIO.out1.val & MASK;
                        ^~~~
                        out
In file included from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms.h:36,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.h:52,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.cpp:2:
/data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/clockless_rmt_esp32.h: In member function 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::startOnChannel(int)':
/data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/clockless_rmt_esp32.h:399:52: warning: 'esp_err_t rmt_set_pin(rmt_channel_t, rmt_mode_t, gpio_num_t)' is deprecated: use rmt_set_gpio instead [-Wdeprecated-declarations]
         rmt_set_pin(mRMT_channel, RMT_MODE_TX, mPin);
                                                    ^
In file included from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/clockless_rmt_esp32.h:97,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms/esp/32/fastled_esp32.h:8,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/platforms.h:36,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.h:52,
                 from /data/c3-pico-hp-1dc59c/.piolibdeps/c3-pico-hp-1dc59c/FastLED/FastLED.cpp:2:
/data/cache/platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/driver/include/driver/rmt.h:930:11: note: declared here
 esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
           ^~~~~~~~~~~
*** [/data/c3-pico-hp-1dc59c/.pioenvs/c3-pico-hp-1dc59c/libb79/FastLED/FastLED.cpp.o] Error 1
========================= [FAILED] Took 20.78 seconds =========================
```

lboue, 2023-03-31:
[logs_c3-pico-hp-1dc59c_run.log](https://github.com/esphome/issues/files/11125203/logs_c3-pico-hp-1dc59c_run.log)

wezm, 2023-05-26:
It seems this is fixed in newer FastLED releases https://github.com/FastLED/FastLED/pull/1320

lboue, 2023-05-30:
Do you think yesterday [FastLED 3.6.0 release](https://github.com/FastLED/FastLED/releases/tag/3.6.0) could fix this issue?

nielsnl68, 2023-05-30:
I will test it shortly, some of the fixes are my doing.

closed 2023-10-18 by github-actions[bot]