esphome/issues #4379: ESPHome in podman container does not respond to SIGTERM

wixoff, 2023-04-01:
### The problem

I am running ESPHome in a podman container in Fedora 37, rootful, and it works perfectly.

At times it would be helpful to shut down ESPHome cleanly - for upgrades, reboots, etc.  However, executing the "podman stop" command (which sends a SIGTERM by default) fails to stop the container, and after a timeout period it resorts to SIGKILL.

Even if I extend the timeout period to something like two minutes, it still ends up having to be SIGKILL'ed.

NOTE: I selected "Docker" for the installation type below because something has to be selected in that field - but I'm not using Docker.

### Which version of ESPHome has the issue?

2023.3.2 and various prior versions

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

n/a

### What platform are you using?

ESP32

### Anything in the logs that might be useful for us?

```txt
Looks like this:

[user@host esphome]$ sudo podman stop esphome
WARN[0120] StopSignal SIGTERM failed to stop container esphome in 120 seconds, resorting to SIGKILL 
```
```

wixoff, 2023-08-09:
Still relevant, though admittedly low priority.

drogenlied, 2025-06-19:
This issue still exists with esphome 2025.6.0.

wixoff, 2025-06-22:
Thanks @drogenlied for the confirmation.  It's still true for me too.  I have just been letting the timeout period run, so I appreciate the suggestion re: stop_signal.  Other containers/images have the same problem, so this is by no means limited to esphome.