esphome/issues #4382: PN532 component blocks for 100ms every update

tracestep, 2023-04-02:
### The problem

The `read_data` function on pn532_i2c and pn532_spi components block the code execution for up to 100ms. Since it's called at least once every `update_interval`, it can consume significant processing time (>10% if update_interval is the default of 1s, and a lot more if the interval is less for quicker tag detection), in addition to hogging the I²C / SPI bus while doing that. It will always timeout those 100ms if no tag is presented to the reader. This is especially noticeable when running other timing-sensitive operations (e.g.: rtttl buzzer, fast-updating sensors).
```
bool PN532I2C::read_data(std::vector<uint8_t> &data, uint8_t len) {
  delay(1);

  std::vector<uint8_t> ready;
  ready.resize(1);
  uint32_t start_time = millis();
  while (true) {
    if (this->read_bytes_raw(ready.data(), 1)) {
      if (ready[0] == 0x01)
        break;
    }

    if (millis() - start_time > 100) {
      ESP_LOGV(TAG, "Timed out waiting for readiness from PN532!");
      return false;
    }
  }

  data.resize(len + 1);
  this->read_bytes_raw(data.data(), len + 1);
  return true;
}
```
The ESP will be stuck in the `while(true)` loop until 100ms pass, or a tag is presented thus causing PN532 to respond with `0x01`.

I am working on an asynchronous implementation shifting waiting for a `0x01` from the chip to the `loop()` function, that seems to be working fine.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32-IDF

### Board

WT32-ETH01

### Component causing the issue

pn532, pn532_i2c, pn532_spi

### Example YAML snippet

```yaml
i2c:
  scl: GPIO32
  sda: GPIO33
  scan: True
  frequency: 100kHz
  timeout: 2ms  # see https://github.com/esphome/issues/issues/4340

pn532_i2c:
  update_interval: 250ms
  id: rfid
  on_tag:
    then:
      - binary_sensor.template.publish:
          id: tag_present
          state: ON
```

tracestep, 2023-04-02:
I have implemented a non-blocking version on my fork: [https://github.com/esphome/esphome/compare/dev...tracestep:esphome:pn532-async](https://github.com/esphome/esphome/compare/dev...tracestep:esphome:pn532-async), that seems to work fine on I²C for simple tag UID reads, but I have not tested on SPI (which uses a slightly different approach) or with the fancy mifare stuff, so I'm reluctant to make a pull-request.

sharkydog, 2023-08-02:
My take on this:
https://github.com/esphome/esphome/compare/dev...sharkydog:esphome:pn532_nonblock

closed 2023-11-28 by jesserockz