esphome/issues #4383: EZO Not Working

rspenc29, 2023-04-02:
### The problem

I can't get the EZO pH module to work correctly. I have 2 modules and tried them both. I've tried both 3.3v and 5v with the ezo sensor as well as 2.2k and 4.7k pull-up resistors on both SDA and SCL. I also went all the way back to esphome release 2022.11.0 where the ezo sensor was initially implemented and it doesn't work with that either.

The pH sensor reading seems to work fine. It's reporting a reading in home assistant and the serial logger and the green light on the sensor lights up every time it reads.

Most or all of the automation triggers don't seem to work. `on_t` and `on_custom` are the only ones that actually get called but `x` is always the `?T` or in the case of `on_custom` it's `?i` or whatever command I tried to call. It's never the actual response from the sensor. And none of the other triggers are even called. 

### Which version of ESPHome has the issue?

2023.3.2, 2022.11.0, and probably everything in between

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

N/A

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

EZO

### Example YAML snippet

```yaml
i2c:
  - id: i2cBus1
    sda: 21
    scl: 22
    scan: true
sensor:
  - platform: ezo
    id: ph1
    name: "pH Sensor 1"
    i2c_id: i2cBus1
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 10s
    on_device_information:
      then:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_device_information %s", x.c_str());
    on_led:
      then:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_led %d", x);
    on_slope:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_slope %s", x.c_str());
    on_calibration:
      then:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_calibration %s", x.c_str());
    on_t:
      then:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_t %s", x.c_str());
    on_custom:
      then:
      - lambda: |-
          ESP_LOGI("main", "ph1_on_custom %s", x.c_str());
button:
  - platform: template
    name: "Set pH Calibration Mid (7)"
    on_press:
      - lambda: |-
          id(ph1).set_calibration_point_mid(7);
  - platform: template
    name: "Set pH Calibration Low (4)"
    on_press:
      - lambda: |-
          id(ph1).set_calibration_point_low(4);
  - platform: template
    name: "Get pH Sensor Info"
    on_press:
      - lambda: |-
          id(ph1).get_device_information();
  - platform: template
    name: "Update pH"
    id: phUpdate
    icon: "mdi:ph"
    on_press:
      - lambda: |-
          id(ph1).get_state();
  - platform: template
    name: "Get pH Slope"
    on_press:
      - lambda: |-
          id(ph1).get_slope();
  - platform: template
    name: "Get pH Calibration"
    on_press:
      - lambda: |-
          id(ph1).get_calibration();
  - platform: template
    name: "Get pH Temp Compensation"
    on_press:
      - lambda: |-
          id(ph1).get_t();
```

nevets85, 2023-04-07:
I am having this same issue as well, cannot compile/deploy

 
Compiling /data/poolmon/.pioenvs/poolmon/src/esphome/components/ezo/ezo.cpp.o
src/esphome/components/ezo/ezo.cpp: In member function 'virtual void esphome::ezo::EZOSensor::update()':
src/esphome/components/ezo/ezo.cpp:27:9: error: 'class esphome::ezo::EZOSensor' has no member named 'write_bytes_raw'; did you mean 'write_bytes_16'?
   this->write_bytes_raw(&c, 1);
         ^~~~~~~~~~~~~~~
         write_bytes_16
src/esphome/components/ezo/ezo.cpp:29:23: error: 'millis' was not declared in this scope
   this->start_time_ = millis();
                       ^~~~~~
src/esphome/components/ezo/ezo.cpp:29:23: note: suggested alternative: 'killpg'
   this->start_time_ = millis();
                       ^~~~~~
                       killpg
src/esphome/components/ezo/ezo.cpp: In member function 'virtual void esphome::ezo::EZOSensor::loop()':
src/esphome/components/ezo/ezo.cpp:38:13: error: 'class esphome::ezo::EZOSensor' has no member named 'write_bytes_raw'; did you mean 'write_bytes_16'?
       this->write_bytes_raw(buf, len);
             ^~~~~~~~~~~~~~~
             write_bytes_16
src/esphome/components/ezo/ezo.cpp:40:27: error: 'millis' was not declared in this scope
       this->start_time_ = millis();
                           ^~~~~~
src/esphome/components/ezo/ezo.cpp:40:27: note: suggested alternative: 'killpg'
       this->start_time_ = millis();
                           ^~~~~~
                           killpg
src/esphome/components/ezo/ezo.cpp:45:7: error: 'millis' was not declared in this scope
   if (millis() - this->start_time_ < this->wait_time_)
       ^~~~~~
src/esphome/components/ezo/ezo.cpp:45:7: note: suggested alternative: 'killpg'
   if (millis() - this->start_time_ < this->wait_time_)
       ^~~~~~
       killpg
Compiling /data/poolmon/.pioenvs/poolmon/src/main.cpp.o
*** [/data/poolmon/.pioenvs/poolmon/src/esphome/components/ezo/ezo.cpp.o] Error 1
========================== [FAILED] Took 6.92 seconds ==========================

rspenc29, 2023-04-07:
> I am having this same issue as well, cannot compile/deploy...

That's not the same issue

nevets85, 2023-04-07:
ok thanks ill submit a new issue

closed 2023-08-26 by github-actions[bot]