esphome/issues #4392: issue compiling BSEC 2 for BME688

dbouchabou, 2023-04-06:
### The problem

Hello, i try to compile ESPhome with a module BME688.
As i use a BME 688 and not a BME 680 I need to use the BSEC2 library version.

Is it possible to update the ESPHOME library code ?

### Which version of ESPHome has the issue?

2023.1.7

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.2.3

### What platform are you using?

ESP32

### Board

S3

TamasKazsoki, 2023-04-15:
Just a bit more clarification as I have similar problem however I own a BME680 sensor:

- only Arduino framework is supported by the Bosch BSEC/BSEC2 libraries. ESP-IDF is not supported at all
- to compile to ESP32-C3/S3, using BSEC2 library is a must as the lalgobsec.a files are not included in the BSEC library for S3 and C3 devices, and using the lalgobsec.a from /data/<device name>/.piolibdeps/<device name>/BSEC\ Software\ Library/src/esp32 as you can find this suggestion in some other issues, gives incompatible library error.
- I tried to use only the lalgobsec.a from a newer BSEC2 library and addig it as a build flag. This resulted a successful compile however the sensor was not able to work at all.

I am also waiting for the fix (updating the BSEC library in ESPHome).

ssieb, 2023-04-15:
This is not something to be filed as an issue.  It could possibly be a feature request, I think there already is one filed.  But if the libraries don't support esp-idf then I don't know what you think esphome can do about it.

closed 2023-08-28 by github-actions[bot]