esphome/issues #4393: ESP does not respond on network when MQTT used without HA API, and "climate: thermostat" used

Nuetron06, 2023-04-06:
### The problem

MQTT and "platform: thermostat" are incompatible, yet no errors are generated. ESP fails to respond on network, even though it is connected. In addition, the esp32 has to be re-flashed with corrected code through USB to bring back online.

I now know the cause, but it would have been nice to get an error preventing me from using the thermostat platform if it required the HA API. 

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.0

### What platform are you using?

ESP32

### Board

LilyGo quad relay board

### Component causing the issue

MQTT and "climate: thermostat"

### Anything in the logs that might be useful for us?

```txt
Logs through USB show device attempting to connect to MQTT server, but reports connection failed and tries again.
```

### Additional information

Backstory and troubleshooting:
I was in the process of moving my air conditioner controller from the HA API to MQTT due to network issues (disconnects etc, bad wifi coverage). 
However, when I updated the code from HA API to MQTT, I failed to update the climate platform from thermostat to mqtt.
The firmware compiled, downloaded and the debugger attempted to retrieve data from the MQTT server, but no data was coming through. 
Pinging the IP address of the esp received no response. Ubiquiti Unifi console showed the device connected to wifi, IP address correct, mac was correct, and it had a rock-solid link. Packet capture on PfSense showed the device was sending ARP requests (once per second) for the default gateway (the device is on a different subnet than HA), but would not respond to ping or OTA updates.
After testing removing bits of my code, I had narrowed it down to the actual thermostats (there are 6 of them, all to control one air conditioner...).
None of these thermostats have MQTT options yet, but I just wanted to get started pulling the temps through MQTT and go from there.

ssieb, 2023-04-06:
You didn't include any yaml or any logs.  There's no reason that the thermostat component would interfere with mqtt.  You said it already had bad wifi coverage, how is changing from the API to mqtt going to help that?  If the device keeps sending arp requests then it's not getting a response.  You need to check your network config.

Nuetron06, 2023-04-07:
Sorry, was at work and couldn't post error logs because I didn't have physical access to the device. I'll be updating the original post shortly. 
The thing is, network config is the same, but if I take out the thermostats from the code, it is able to connect (I've verified through mqtt explorer). Packet capture on the router shows arp requests only, and incoming icmp packets but no outgoing data.

Nuetron06, 2023-04-07:
> You said it already had bad wifi coverage, how is changing from the API to mqtt going to help that?
I should have taken screenshots of the Unifi wifi experience console. I think the "bad coverage" was needing to re-flash the device via USB. Since I did that, I have not had a single disconnect in over 24 hours. Before, the device was reconnecting to wifi every 30s to 6m. 
The main difference I was shooting for with the api>mqtt change is that the esp could check in when it could, whereas with the api, I'd have to let HA check in, and it was hit or miss (It's like trying to shoot a fan blade through a hole when you can't see the fan blade coming).
However, since re-flashing, I have not had any more disconnects through the API. It had been progressively getting worse.

Nuetron06, 2023-04-07:
Logs are added now.

>There's no reason that the thermostat component would interfere with mqtt.
>If the device keeps sending arp requests then it's not getting a response. 

My thoughts too. 

>You need to check your network config.

Done. I'm a network engineer, and because I do often overlook stuff, I don't reach out to report bugs lightly because of it. 
I've been able to reproduce on another device, and it's not the network config, since it works without the thermostats.

closed 2023-08-23 by github-actions[bot]