esphome/issues #4395: Ezo Sensor failing to compile

nevets85, 2023-04-07:
### The problem

I've not needed to make changes to this particular sensor deployment for a long time so i'm not sure which version it stopped compiling in. I created a brand new yaml with just the sensor i need in it and it still fails to compile

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.0

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

ezo ph and ezo orp 

### Example YAML snippet

```yaml
esphome:
  name: pooltest
  friendly_name: pooltest

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "redacted"
    password: "redacted"

captive_portal:

i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a

sensor:
  - platform: ezo
    id: ph_ezo
    name: "pH"
    i2c_id: bus_a  
    address: 99
    accuracy_decimals: 1
    unit_of_measurement: "pH"

  - platform: ezo
    id: orp_ezo
    name: "ORP"
    i2c_id: bus_a
    address: 98
    accuracy_decimals: 1
    unit_of_measurement: "orp"
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/pooltest.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing pooltest (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
INFO Installing esphome/AsyncTCP-esphome @ 1.2.2
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@1.2.2 has been installed!
INFO AsyncTCP-esphome@1.2.2 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.4
INFO Installing esphome/noise-c @ 0.1.4
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.4 has been installed!
INFO noise-c@0.1.4 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.1
INFO Installing esphome/libsodium @ 1.10018.1
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.1 has been installed!
INFO libsodium@1.10018.1 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- Wire @ 2.0.0
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/api_connection.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/api_server.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/list_entities.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/proto.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/api/user_services.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/esp32/core.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/esp32/gpio.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/ezo/ezo.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/i2c/i2c.cpp.o
Compiling /data/pooltest/.pioenvs/pooltest/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
src/esphome/components/ezo/ezo.cpp: In member function 'virtual void esphome::ezo::EZOSensor::update()':
src/esphome/components/ezo/ezo.cpp:27:9: error: 'class esphome::ezo::EZOSensor' has no member named 'write_bytes_raw'; did you mean 'write_bytes_16'?
   this->write_bytes_raw(&c, 1);
         ^~~~~~~~~~~~~~~
         write_bytes_16
src/esphome/components/ezo/ezo.cpp:29:23: error: 'millis' was not declared in this scope
   this->start_time_ = millis();
                       ^~~~~~
src/esphome/components/ezo/ezo.cpp:29:23: note: suggested alternative: 'killpg'
   this->start_time_ = millis();
                       ^~~~~~
                       killpg
src/esphome/components/ezo/ezo.cpp: In member function 'virtual void esphome::ezo::EZOSensor::loop()':
src/esphome/components/ezo/ezo.cpp:38:13: error: 'class esphome::ezo::EZOSensor' has no member named 'write_bytes_raw'; did you mean 'write_bytes_16'?
       this->write_bytes_raw(buf, len);
             ^~~~~~~~~~~~~~~
             write_bytes_16
src/esphome/components/ezo/ezo.cpp:40:27: error: 'millis' was not declared in this scope
       this->start_time_ = millis();
                           ^~~~~~
src/esphome/components/ezo/ezo.cpp:40:27: note: suggested alternative: 'killpg'
       this->start_time_ = millis();
                           ^~~~~~
                           killpg
src/esphome/components/ezo/ezo.cpp:45:7: error: 'millis' was not declared in this scope
   if (millis() - this->start_time_ < this->wait_time_)
       ^~~~~~
src/esphome/components/ezo/ezo.cpp:45:7: note: suggested alternative: 'killpg'
   if (millis() - this->start_time_ < this->wait_time_)
       ^~~~~~
       killpg
*** [/data/pooltest/.pioenvs/pooltest/src/esphome/components/ezo/ezo.cpp.o] Error 1
========================= [FAILED] Took 18.68 seconds =========================
```

### Additional information

I've tried commenting out sensors individually in the yaml to see if it is one vs the other, I've also tried commenting out both which then allows it to compile fine but obviously do not have my sensors.

ssieb, 2023-04-07:
Have you ever used custom components?

nevets85, 2023-04-07:
no, ive only followed the guides listed here
https://esphome.io/components/sensor/ezo.html

nevets85, 2023-04-07:
clean build as in a new config in esphome? if so that one is brand new, wanted to test prior to posting here.

ssieb, 2023-04-07:
No, from the dashboard there's an option in the menu to clean the build directory or if you using the command line, it's the `clean` command instead of `compile`.

ssieb, 2023-04-07:
This is really a support issue.  Can you come to discord and ask there instead?

nevets85, 2023-04-07:
joining there, thank you

closed 2023-04-07 by nevets85