esphome/issues #4396: "rmt: RMT RX BUFFER FULL" spam on UART with remote_receiver on ESP32

tracestep, 2023-04-08:
### The problem

I am using the `remote_receiver` component with a 433MHz receiver connected to an ESP32 board. At times (I suppose deppending on the conditions of the 433MHz spectrum) I get several  `rmt: RMT RX BUFFER FULL` messages per second on the UART port.

I get that most 433MHz receivers output random noise on the digital output when they increase the gain because there is no carrier detected. The thing is, _even if I disable serial logging, these messages still appear on the first UART bus I have configured_, interfering in the communication with the serial peripheral I have connected. The messages do not appear on the network log. 

There is another issue on this exact thing: https://github.com/esphome/issues/issues/1143, but it got closed without a solution for being stale. I have found that the message comes from the `rmt_legacy.c` at `components/driver/deprecated` from `esp-idf`. There is a "new" remote receiver driver from Espressif at  `coponents/driver/rmt/`. So it appears that `remote_receiver` component on ESPHome is using a deprecated API from `esp-idf` that will likely not get fixed.

I think, as a workaround, I can compile against a patched fork of the esp-idf with the specific error logging commented-out, but I believe the real solution would be to update `remote_receiver` to use the current API, which I am not sure how to do at the moment.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.5

### What platform are you using?

ESP32-IDF

### Board

WT32-ETH01

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
logger:
  baud_rate: 0

uart:
  - id: myuart
    rx_pin: GPIO5
    tx_pin: GPIO17
    baud_rate: 19200

remote_receiver:
  pin: 
    number: GPIO35
    inverted: false
    mode: INPUT
  filter: 200us
  idle: 4000us
  buffer_size: 2k
  on_raw:
    then:
      lambda: |-
        # . . .
```

### Anything in the logs that might be useful for us?

```txt
E (1039860) rmt: RMT RX BUFFER FULL
E (1039886) rmt: RMT RX BUFFER FULL
E (1039927) rmt: RMT RX BUFFER FULL
E (1039945) rmt: RMT RX BUFFER FULL
E (1039966) rmt: RMT RX BUFFER FULL
...
```

### Additional information

The issue is intermittent (it depends on the 433MHz spectrum, I guess, as I said), so it might be several hours without the messages, and several minutes with constant messages (several per seconds. It is not happening right now, so I copied the log from the issue I mentioned.  I don't have a pattern generator to reliably reproduce the issue...

tracestep, 2023-04-09:
Well, I was not able to compile against a patched esp-idf... setting the `esp32/framework/source` option to my patched fork yielded errors... but I believe setting the `sdkconfig_options` `CONFIG_LOG_DEFAULT_LEVEL_NONE: y` functioned as a workaround:

```
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_LOG_DEFAULT_LEVEL_NONE: y
```

Javiergzhdz, 2023-05-19:
I have the same problem and I don´t find any solution.

Muskanjhawar, 2024-03-15:
Having the same issue. Any solutions?

Javiergzhdz, 2024-03-21:
At least for me, the problem fixed by itself in ESP-IDF V5. A lot of things changed. Nevertheless it is worth giving a shot.