esphome/issues #4397: [name] is an invalid option for [output.modbus_controller]

645340633, 2023-04-09:
### The problem

[name] is an invalid option for [output.modbus_controller]

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.3.2

### What platform are you using?

ESP32

645340633, 2023-04-09:
[name] is an invalid option for [output.modbus_controller]. Please check the indentation.

645340633, 2023-04-09:
[image]

ssieb, 2023-04-09:
That's a bug in the docs, probably copy and paste.  `output`s don't have names.

closed 2023-08-23 by github-actions[bot]