esphome/issues #4405: rpipicow display/ili9xxx 480*320 displays breaks api (HA cannot connect)

nathanp512, 2023-04-12:
### The problem

Home Assistant cannot add esphome RP2040 device when ili9xxx display with 480*320 models ili9481, ili9488 and st7796 enabled.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.4.2

### What platform are you using?

RP2040

### Board

Raspberry Pi Pico W

### Component causing the issue

display/ili9xxx

### Example YAML snippet

```yaml
esphome:
  name: rpidisplaytest

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
#  encryption:
#    key: "YcrSvHb/s+OtivspBfJ52v/N0kCytHKRLqlOFExMG8s="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

rp2040:
  board: rpipicow
  framework:
    # Required until https://github.com/platformio/platform-raspberrypi/pull/36 is merged
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git

spi:
  clk_pin: 10
  mosi_pin: 11
  miso_pin: 12

display:
 - platform: ili9xxx
   model: st7796
   cs_pin: 9
   dc_pin: 8
   reset_pin: 15
   rotation: 270°
   lambda: |-
     it.line(240, 160, 480, 0, COLOR_WHITE);
```

### Anything in the logs that might be useful for us?

```txt
[VV][scheduler:224]: Running interval 'update' with interval=1000 last_execution=4294966519 (now=8413)
[V][ili9xxx:171]: Start display(xlow:0, ylow:0, xhigh:319, yhigh:479, width:320, heigth:480, start_pos:0)
[V][component:204]: Component ili9xxx.display took a long time for an operation (15.29 s).
[V][component:205]: Components should block for at most 20-30ms.
[I][wifi:518]: WiFi Connected!
[C][wifi:362]:   Local MAC: 28:CD:C1:06:0D:1B
[C][wifi:363]:   SSID: 'xxxxxxxx'[redacted]
[C][wifi:364]:   IP Address: xxx.xxx.xxx.86
[C][wifi:365]:   BSSID: xx:xx:xx:xx:xx:xx[redacted]
[C][wifi:367]:   Hostname: 'rpidisplaytest'
[C][wifi:369]:   Signal strength: -47 dB ▂▄▆█
[C][wifi:373]:   Channel: 11
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: xxx.xxx.xxx.1
[C][wifi:376]:   DNS1: xxx.xxx.xxx.59
[C][wifi:377]:   DNS2: xxx.xxx.xxx.42
[C][api:025]: Setting up Home Assistant API server...
[I][app:062]: setup() finished successfully!
[VV][scheduler:224]: Running interval 'update' with interval=1000 last_execution=8223 (now=23712)
[V][ili9xxx:171]: Start display(xlow:0, ylow:0, xhigh:319, yhigh:479, width:320, heigth:480, start_pos:0)
[V][component:204]: Component ili9xxx.display took a long time for an operation (15.29 s).
[V][component:205]: Components should block for at most 20-30ms.
[D][api:102]: Accepted xxx.xxx.xxx.59
[VV][api.service:450]: on_hello_request: HelloRequest {
  client_info: 'aioesphomeapi'
  api_version_major: 1
  api_version_minor: 7
}
[V][api.connection:901]: Hello from client: 'aioesphomeapi (xxx.xxx.xxx.59)' | API Version 1.7
[VV][api.service:013]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 7
  server_info: 'rpidisplaytest (esphome v2023.3.2)'
  name: 'rpidisplaytest'
}
[I][app:102]: ESPHome version 2023.3.2 compiled on Apr 12 2023, 11:26:26
[C][wifi:504]: WiFi:
[C][wifi:362]:   Local MAC: 28:CD:C1:06:0D:1B
[C][wifi:363]:   SSID: 'xxxxxxx'[redacted]
[C][wifi:364]:   IP Address: xxx.xxx.xxx.86
[C][wifi:365]:   BSSID: xx:xx:xx:xx:xx:xx[redacted]
[C][wifi:367]:   Hostname: 'rpidisplaytest'
[C][wifi:369]:   Signal strength: -46 dB ▂▄▆█
[C][wifi:373]:   Channel: 11
[C][wifi:374]:   Subnet: 255.255.255.0
[C][wifi:375]:   Gateway: xxx.xxx.xxx.1
[C][wifi:376]:   DNS1: xxx.xxx.xxx.59
[C][wifi:377]:   DNS2: xxx.xxx.xxx.42
[VV][scheduler:224]: Running interval 'update' with interval=1000 last_execution=23223 (now=39023)
[V][ili9xxx:171]: Start display(xlow:0, ylow:0, xhigh:319, yhigh:479, width:320, heigth:480, start_pos:0)
[V][component:204]: Component ili9xxx.display took a long time for an operation (15.29 s).
[V][component:205]: Components should block for at most 20-30ms.
[VV][api.service:504]: on_device_info_request: DeviceInfoRequest {}
[VV][api.service:049]: send_device_info_response: DeviceInfoResponse {
  uses_password: NO
  name: 'rpidisplaytest'
  mac_address: '28:CD:C1:06:0D:1B'
  esphome_version: '2023.3.2'
  compilation_time: 'Apr 12 2023, 11:26:26'
  model: 'rpipicow'
  has_deep_sleep: NO
  project_name: ''
  project_version: ''
  webserver_port: 0
  bluetooth_proxy_version: 0
  manufacturer: 'Raspberry Pi'
  friendly_name: ''
}
[C][logger:293]: Logger:
[C][logger:294]:   Level: VERY_VERBOSE
[C][logger:295]:   Log Baud Rate: 115200
[C][logger:296]:   Hardware UART: USB_CDC
[VV][scheduler:224]: Running interval 'update' with interval=1000 last_execution=38223 (now=54326)
[V][ili9xxx:171]: Start display(xlow:0, ylow:0, xhigh:319, yhigh:479, width:320, heigth:480, start_pos:0)
[V][component:204]: Component ili9xxx.display took a long time for an operation (15.29 s).
[V][component:205]: Components should block for at most 20-30ms.
[VV][api.socket:809]: aioesphomeapi (xxx.xxx.xxx.59): Socket read failed with errno 104
[W][api.connection:081]: aioesphomeapi (xxx.xxx.xxx.59): Connection reset
[C][spi:101]: SPI bus:
[C][spi:102]:   CLK Pin: GPIO10
[C][spi:103]:   MISO Pin: GPIO12
[C][spi:104]:   MOSI Pin: GPIO11
[C][spi:106]:   Using HW SPI: NO
[VV][scheduler:224]: Running interval 'update' with interval=1000 last_execution=54223 (now=69627)
[V][ili9xxx:171]: Start display(xlow:0, ylow:0, xhigh:319, yhigh:479, width:320, heigth:480, start_pos:0)
```

### Additional information

- When API encryption enabled: HA asks for encryption key but with same result (no connection).
- Port 6053 is open (checked with telnet).
- It does not matter whether the display hardware is connected or not.
- Not sure it is relevant: It takes about 15 seconds to build up the screen as can be seen in the logs.
- What I also did (with same results; cannot connect api when display section enabled):
  - I also locally installed esphome (pip) but with the same results.
  - Downgraded esphome (docker) to 2023.2.4
  - Downgraded home assistant (docker) to 2023.3.6
  - Defined model ili9341 display in rpicow (instead of st7796)
  - Without lambda in YAML
  - 480*320 display models like ili9481 ili9488 st7796

- HA can connect to ESPHome device with 320*240 display models configured (ili9341, ili9342 and TFT 2.4 tested).
 

When I remove/disable the display section (last 9 lines) in YAML HA connects without any issues and it works as expected (except that I cannot display any data).

nielsnl68, 2023-04-12:
Sorry to say, i was not able to test my changes with a Pi Pico. Recently i have ordered one, to see what happens.
I will look into it asap.

nathanp512, 2023-04-12:
I would be happy if I can help.

Offtopic: In case someone has the Waveshare Pico-ResTouch-LCD-3.5 I got this 'ili9488' screen working with some modifications of ST7796 (but takes 15 sec for screen build and no api connection) with this piece of code:

**ili9xxx_init.h:**

static const uint8_t PROGMEM INITCMD_WS35PICO[] = {
  // This Waveshare Pico-ResTouch-LCD-3.5 initilization routine was copied from ST7796
  ILI9XXX_SWRESET,   0x80,         // Soft reset, then delay 150 ms
  ILI9XXX_MADCTL, 1, 0x48,       //Memory Access [00]
  ILI9XXX_PIXFMT, 1, 0x55,         //565 --> is_18bitdisplay_ = true;  //Must be used in combination with ILI9XXX_PIXFMT, 1, 0x66 in init
  ILI9XXX_IFMODE, 1, 0x80,              //Interface     [00]
  ILI9XXX_DFUNCTR, 2, 0x02, 0x02,  // Nomal scan  
  ILI9XXX_ETMOD, 1, 0xC6,              //Entry Mode      [06]

  ILI9XXX_SLPOUT, 0x80, // Exit Sleep, then delay 150 ms
  ILI9XXX_DISPON, 0x80, // Main screen turn on, delay 150 ms

  0x00
};

**ili9xxx_display.cpp**:

//    35_TFT display Waveshare Pico-ResTouch-LCD-3.5
void ILI9XXXWS35PICO::initialize() {
  this->init_lcd_(INITCMD_WS35PICO);
  if (this->width_ == 0) {
    this->width_ = 320;
  }
  if (this->height_ == 0) {
    this->height_ = 480;
  }
  this->invert_display_(true);   // Inverts color
}

nielsnl68, 2023-04-12:
top, do you mind when i add it to the ili9xxx component. 
Although, that said i like to use the driver name of the display.
When you can give me that, then that will be prefert.

nathanp512, 2023-04-12:
Sure you can use the code. 
According Waveshare docs it uses an Ilitek ili9488 driver. But I was not able to use it as an ili9488 with the 'GFX Library for Arduino' and the TFT_eSPI library needs some file replacements (Waveshare [Pico-ResTouch-LCD-3.5 wiki](https://www.waveshare.com/wiki/Pico-ResTouch-LCD-3.5) ). The ST7796 model of the ili9xxx component worked best for my display (although colors were inverted and screen content was mirrored).
So I think there are different implementations of the ili9488 driver.

nielsnl68, 2023-08-17:
@nathanp512 could you make a PR for above changes?

nathanp512, 2023-08-30:
Hi @nielsnl68 I am not sure what driver name to use. According to WaveShare docs it is an ILI9488 but the ST7796S code suits better. Anyway, neither the ili9488 or st7796 code works for this display without modifications. I studied some schematics on http://www.lcdwiki.com/ but my WaveShare Pico-ResTouch-LCD-3.5 differs from other displays (it seems to convert SPI to parallel with some 74HC4040 and 74HC4094 IC's).

What driver name do you suggest for this WaveShare Pico-ResTouch-LCD-3.5 display?

nathanp512, 2023-12-29:
This issue (HA cannot connect API) seems fixed.
Tested successfully with ESPHome 2023.11.6.

closed 2023-12-29 by nathanp512