esphome/issues #4406: shelly 1L not working properly

dflvunoooooo, 2023-04-12:
### The problem

If I set up my shelly 1L with neutral, I get a flashing led light in a specific combination of my two switches / inputs. I would expect the light to flash if no neutral was attached, because then the shelly would draw it's power over the light wire. But this shouldn't happen at all if the neutral is attached. 

Is there some way to tell the shelly how to behave?

Here is my wiring:
[image]

### Which version of ESPHome has the issue?

 2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.04.0

### What platform are you using?

ESP8266

### Board

sheely 1L

### Example YAML snippet

```yaml
substitutions:
  device_name: shelly1l-flur-deckenlicht
  device_description: "Shelly1L Deckenlicht"
  ort: Flur

esphome:
  name: ${device_name}

esp8266:
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption: 
    key: !secret api_encryption_key

ota:
  password: !secret ha_api_password

# Zeit von Home Assistant
time:
  - platform: homeassistant
    id: homeassistant_time

# Buttons
button:
  # Restart ESP32 Butten
  - <<: !include common/button/restart_button.yaml

# Text-Sensor
text_sensor:
# IP-Adresse als Sensor
  - <<: !include common/text-sensor/ip-adresse.yaml

# Sensoren
sensor:
  # WLAN Signalstärke
  - <<: !include common/sensor/wifi_strength.yaml
  
  # Betriebszeit
  - <<: !include common/sensor/uptime.yaml

  - platform: ntc
    sensor: temp_resistance_reading
    name: ${device_name}_temperature
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 298.15K
    on_value_range:
      - above: "80.0"
        then:
          - light.turn_off: lightid
  - platform: resistance
    id: temp_resistance_reading
    sensor: temp_analog_reading
    configuration: DOWNSTREAM
    resistor: 32kOhm
  - platform: adc
    id: temp_analog_reading
    pin: A0

web_server:
  port: 80

status_led:
  pin: GPIO0

output:
  - platform: gpio
    pin: GPIO5
    id: shelly_1l_relay

light:
  - platform: binary
    name: "${ort} Deckenlicht ${device_description}"
    output: shelly_1l_relay
    id: lightid

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO4
    internal: true
    on_state:
      then:
        - light.toggle: lightid
    id: switchid1
    filters:
      - delayed_on_off: 50ms
  - platform: gpio
    pin:
      number: GPIO14
    internal: true
    on_state:
      then:
        - light.toggle: lightid
    id: switchid2
```

veyselr, 2023-06-12:
Not corect wiring. 
[image]

dflvunoooooo, 2023-06-21:
> Not corect wiring. [image]

This is not possible in my case, not enough wires in the wall. And the wiring can not be wrong, since it is an official shelly wiring. 

But never the less should the shelly not draw power over 0 when there is a N attached.