esphome/issues #4407: BME680 Reporting the wrong big value for Gas Resistance (and therefore all the other divided parameters )

tactmaster, 2023-04-12:
### The problem

I have a BME680 on a sense board. https://www.kickstarter.com/projects/nexuselectronics/sense-an-open-source-environment-monitor.

The Gas resistance doesn't change it is always at 12887828.00000. It effects booth the bme680 and bme680_bsec competents 
 
So I have tested the on a raspberry pi with the examples given adafruit lib (non BSEC lib). I have tried with the platform io using this example. 
It work fine. https://registry.platformio.org/libraries/boschsensortec/BSEC%20Software%20Library/examples/basic/basic.ino
It uses BSEC code 1.8.1492.

From looking at the ada fruit code it would appears to be 2 variants of the the BME680

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.2

### What platform are you using?

ESP32

### Board

featheresp32, esp01_1m

### Component causing the issue

bme680,bme680_sec

### Example YAML snippet

```yaml
esphome:
  name: sensor2
  friendly_name: sensor2

esp32:
  board: featheresp32
  framework:
    type: arduino

web_server:
  port: 80

i2c:

bme680_bsec:
    address: 0x77

sensor:
  - platform: bme680_bsec
    gas_resistance:
      name: "BME680 Gas Resistance"
```

### Additional information

I am trying to upgrade the from version BSEC version 1.6.1480 to 1.8.1492. 
However the structure has changed to the lib a little bit, I am not that experience with ESPHome code but i  I would give it a go.

CarlosGS, 2023-05-13:
Hi, you have a BME688, that's why the resistance value is different.
Search for BME688 in ESPHome's feature requests, I think there is a custom component.

tactmaster, 2023-05-15:
Yes you are right. Thanks
Here:
https://github.com/esphome/feature-requests/issues/2134
https://github.com/esphome/feature-requests/issues/1332

The PR is here
https://github.com/esphome/esphome/pull/4585
It seems to work (i compile it from the dev environment and rebased it )

closed 2023-10-06 by github-actions[bot]