esphome/issues #4408: RC522 NFC reader starts on_tag action only once

insoltechdev, 2023-04-13:
### The problem

I try to use the RC522 SPI module's `on_tag` action to detect the tag continuously. Unfortunately, this action runs only once when the TAG is in the sensing field, but the description said this should run several times according to `update_interval` setting.

Formerly I used the RDM6300 in the same way (same variables and lambda codes), and it worked.

The `binary_sensor` way is not working in my solution because I use more than 200 TAGs and these dynamically change their UIDs.

The `on_tag_remove` action could be the other option that works, but the RC522 SPI does not include this action.

### Which version of ESPHome has the issue?

v2023.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

rc522_spi

### Example YAML snippet

```yaml
spi:
  clk_pin: GPIO14
  miso_pin: GPIO12
  mosi_pin: GPIO13

rc522_spi:
  cs_pin: GPIO15
  reset_pin: GPIO17
  update_interval: 500ms
  id: rc522_reader
  on_tag:
    then:
      - lambda: |-
          std::string str = esphome::to_string(x);
          id(toggleBit) = true;
          id(tag_id).publish_state(str.c_str());

globals:
  - id: toggleBit
    type: bool
    initial_value: 'false'

interval:
  - interval: 1000ms
    then:
      - lambda: |-
          if ((!id(toggleBit)) && (id(tag_id).state != "N/A")){
            id(tag_id).publish_state("N/A");
          }
          if(id(toggleBit)){
            id(toggleBit) = false;
          }

text_sensor:
  - platform: template
    name: "Read ID"
    id: tag_id
```

insoltechdev, 2023-04-19:
Hi @jesserockz,

Could you help me with this problem?
Is there a chance that the RC522 component will get the `on_tag_removed` action like PN532 has it?
Thank you in advance for your help.

kbx81, 2023-04-21:
I ordered an RC522 board to play with; hopefully I can find some time to take a look at this in the not-too-distant future.

insoltechdev, 2023-04-24:
> I ordered an RC522 board to play with; hopefully I can find some time to take a look at this in the not-too-distant future.

Thank you so much. I am waiting for your experience.

patrickkeenan, 2023-04-25:
Hi, I'm having the same issue. Ideally I'd be able to use on_tag_removed, but if I can't then the next best thing is to constantly ping the mqtt server. I want to ensure the tag is still present every second, and if its not, I want to know that its empty. Any workarounds? Thanks!

kbx81, 2023-04-26:
Please test https://github.com/esphome/esphome/pull/4742 and let me know how it works.

Add this to your config to test the changes:

```yaml
external_components:
  - source: github://pr#4742
    components: [ rc522 ]
```

kbx81, 2023-04-26:
Updated doc preview is [here](https://deploy-preview-2876--esphome.netlify.app/components/binary_sensor/rc522.html)

insoltechdev, 2023-04-26:
> Please test [esphome/esphome#4742](https://github.com/esphome/esphome/pull/4742) and let me know how it works.
> 
> Add this to your config to test the changes:
> 
> ```yaml
> external_components:
>   - source: github://pr#4742
>     components: [ rc522 ]
> ```

It is working. Thank you so much.
I checked the documentation preview as well, it is also OK.

closed 2023-04-26 by kbx81