esphome/issues #4411: Wemos D1 Mini S2 touch support disabled

spuder, 2023-04-14:
### The problem

The datasheet for the S2 (esp32) chip shows that touch sensors are available on pins 1-14

See section [3.3.4 in the datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf)

If you attempt to enable the `esp32_touch`, it gives the following error: 

```yaml
esp32_touch:

binary_sensor:
  - platform: esp32_touch
    name: "ESP32 Touch Pad"
    pin: 13
    threshold: 1000

```

```
Failed config

esp32_touch: [source config.yaml:274]
  
  This feature is only available on ESP32.
  {}
```

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

Wemos D1 Mini S2 

### Component causing the issue

esp32_touch

### Example YAML snippet

```yaml
https://www.wemos.cc/en/latest/s2/s2_mini.html
```

spuder, 2023-04-14:
Code validation lines: https://github.com/esphome/esphome/blob/6b67acbeb5fc6fae713ed81f7ee549f53a3d42ff/esphome/config_validation.py#L534-L541

spuder, 2023-04-14:
```yaml
esphome:
  name: foobar3
  friendly_name: AnkerMake-M5

esp32:
  board: lolin_s2_mini
  variant: esp32s2
  ```

czechmate777, 2023-04-15:
I'm having the exact same issue. I traced the code a little further, and it seems to be coming from here:
https://github.com/esphome/esphome/blob/6b67acbeb5fc6fae713ed81f7ee549f53a3d42ff/esphome/components/esp32_touch/__init__.py#L79

kbx81, 2023-07-16:
I'm working on this -- should have a PR ready to try later this week. 😇

closed 2023-08-07 by jesserockz

TaruDesigns, 2023-09-06:
@kbx81  I have tried this on the latest Esphome release (2023.8.2) and I can't seem to get any coherent readings with a LOLIN S2 MINI. The project compiles and uploads just fine, and I can see the values with setup_mode: true but the value will never change (whether up or down, it just stays the same) no matter what I do. The same setup is working fine for a regular ESP32 WROOM.

kbx81, 2023-09-08:
@TaruDesigns that's...odd...I'm running it on an S2 here myself and it's working great. Maybe [join us on Discord](https://discord.gg/ejmfyEhK) and some of us can help you there. If the values you see reported with `setup_mode: true` are not changing at all, that suggests something else is out of place somehow -- but we'd need to see your config as well as how you've wired up your device.

TaruDesigns, 2023-09-08:
So thanks to @kbx81  we managed to find the issue:

The default "measurement duration" parameter (which is also used if it's not specified) is simply too fast to get any coherent values.

A value of 100us gives pretty good results, the full config using filters can be as follows:

``` esp32_touch:
  measurement_duration: 100us
  denoise_grade: BIT12
  denoise_cap_level: L0
  debounce_count: 1
  filter_mode: IIR_16
  noise_threshold: 0
  jitter_step: 0
  smooth_mode: IIR_2 ```

kbx81, 2023-09-08:
Thanks for circling back with the solution @TaruDesigns! 🍻

In general, for others/for posterity, the key takeaway here is what's [noted at the bottom of the documentation[(https://esphome.io/components/binary_sensor/esp32_touch#a-note-about-s2-and-s3-variants). In particular:

> if you are moving your configuration from an original ESP32 to an S2 or S3 variant, expect that you’ll need to make some adjustments to your configuration to accommodate [the S2/S3].

The S2 & S3 _are different_ from the original ESP32 when it comes to capacitive touch. This experience reflects that pretty directly. 😇

kbx81, 2023-09-08:
Thanks for circling back with the solution @TaruDesigns! 🍻

In general, for others/for posterity, the key takeaway here is what's [noted at the bottom of the documentation](https://esphome.io/components/binary_sensor/esp32_touch#a-note-about-s2-and-s3-variants). In particular:

> if you are moving your configuration from an original ESP32 to an S2 or S3 variant, expect that you’ll need to make some adjustments to your configuration to accommodate [the S2/S3].

The S2 & S3 _are different_ from the original ESP32 when it comes to capacitive touch. This experience reflects that pretty directly. 😇