esphome/issues #4412: I2C not have any retry mechanism

Szewcson, 2023-04-15:
### The problem

I'm working on GDK101 sensor implementation, and I encountered a problem with I2C write function. With Tasmota implementation of i2c driver everything works. Adding some delay before `wire_->endTransmission(stop);` is a temporary workaround but seems not stable. 

### Which version of ESPHome has the issue?

Firmware: 2023.5.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.4.2

### What platform are you using?

ESP32

### Board

ESP32 Devkit v1

### Component causing the issue

i2c_bus_arduino

### Example YAML snippet

```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "XXX"
  password: "XXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: "wGmReCrulAVP"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time
    
web_server:
  port: 80

status_led:
  pin:
    number: GPIO2
    
i2c:
  sda: GPIO21
  scl: GPIO22
[... 251 lines cut ...]
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x72 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x73 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x74 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x75 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x76 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][i2c.arduino:140]: 0x77 TX 
[22:04:17][VV][i2c.arduino:165]: TX failed: not acknowledged
[22:04:17][VV][scheduler:063]: set_interval(name='', interval=60000, offset=27990)
[22:04:17][VV][scheduler:063]: set_interval(name='update', interval=60000, offset=12898)
[22:04:17][VV][scheduler:063]: set_interval(name='update', interval=900000, offset=211660)
[22:04:17][C][gdk101:039]: Setting up GDK101...
[22:04:17][VV][i2c.arduino:140]: 0x18 TX A0
[22:04:17][VV][i2c.arduino:168]: TX failed: unknown error 5
[22:04:17][E][component:113]: Component gdk101.sensor was marked as failed.
```

### Additional information

I'm actually using local build not the one from HA, but there is no such option to chose.

ssieb, 2023-04-15:
I have no idea what you're asking for.  If you want some help with your code, then come ask on the discord server.

Szewcson, 2023-04-15:
I think this is not a problem with my code but rather with the i2c driver implementation, since Tasmota implementation works just fine, and after adding the delay to the esphome arduino i2c driver my code also work. What I proposing here is adding some while loop with timeout that will do the check instead just checking once if transmision is finished. I will check if the problem is also there for esp-idf and esp8266 build.

ssieb, 2023-04-15:
If your device needs special handling like that, you can implement it in your component.
(I did look at the Tasmota i2c code and I doubt that we would add something like that.)

Szewcson, 2023-04-16:
Ok, I start looking on arduino wire api. I will check if it will be enough to increas timeout value.

closed 2023-08-29 by github-actions[bot]