esphome/issues #4413: HTU21D Fails to Connect on Latest Release

t-az, 2023-04-16:
### The problem

HTU21D temperature and humidity sensor fails to connect over i2c on the latest release of ESPHome (which I believe is 2023.3.2, what ever version has the "latest" tag in Docker). When trying to connect I get an error message saying "Communication with HTU21D failed!" Rolling back to 2023.2.4 resolves this issue for me and gets the sensor working again. Specs for my system are below, let me know if you need any more info. 

Log from latest release:
```
[23:35:50][C][i2c.idf:047]: I2C Bus:
[23:35:50][C][i2c.idf:048]:   SDA Pin: GPIO1
[23:35:50][C][i2c.idf:049]:   SCL Pin: GPIO2
[23:35:50][C][i2c.idf:050]:   Frequency: 50000 Hz
[23:35:50][C][i2c.idf:053]:   Recovery: bus successfully recovered
[23:35:50][I][i2c.idf:063]: Results from i2c bus scan:
[23:35:50][I][i2c.idf:069]: Found i2c device at address 0x40
[23:35:50][C][switch.gpio:076]: GPIO Switch 'Garage Door'
[23:35:50][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[23:35:50][C][switch.gpio:031]:   Pin: GPIO4
[23:35:50][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Garage Door Sensor'
[23:35:50][C][gpio.binary_sensor:016]:   Pin: GPIO6
[23:35:50][C][htu21d:028]: HTU21D:
[23:35:50][C][htu21d:029]:   Address: 0x40
[23:35:50][E][htu21d:031]: Communication with HTU21D failed!
[23:35:50][C][htu21d:033]:   Update Interval: 60.0s
[23:35:50][C][htu21d:034]:   Temperature 'Garage Temperature'
[23:35:50][C][htu21d:034]:     Device Class: 'temperature'
[23:35:50][C][htu21d:034]:     State Class: 'measurement'
[23:35:50][C][htu21d:034]:     Unit of Measurement: '°F'
[23:35:50][C][htu21d:034]:     Accuracy Decimals: 1
[23:35:50][C][htu21d:035]:   Humidity 'Garage Humidity'
[23:35:50][C][htu21d:035]:     Device Class: 'humidity'
[23:35:50][C][htu21d:035]:     State Class: 'measurement'
[23:35:50][C][htu21d:035]:     Unit of Measurement: '%'
[23:35:50][C][htu21d:035]:     Accuracy Decimals: 1
```

Log from 2023.2.4
```
[23:42:42][C][i2c.idf:047]: I2C Bus:
[23:42:42][C][i2c.idf:048]:   SDA Pin: GPIO1
[23:42:42][C][i2c.idf:049]:   SCL Pin: GPIO2
[23:42:42][C][i2c.idf:050]:   Frequency: 50000 Hz
[23:42:42][C][i2c.idf:053]:   Recovery: bus successfully recovered
[23:42:42][I][i2c.idf:063]: Results from i2c bus scan:
[23:42:42][I][i2c.idf:069]: Found i2c device at address 0x40
[23:42:42][C][switch.gpio:076]: GPIO Switch 'Garage Door'
[23:42:42][C][switch.gpio:099]:   Restore Mode: restore defaults to OFF
[23:42:42][C][switch.gpio:031]:   Pin: GPIO4
[23:42:42][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Garage Door Sensor'
[23:42:42][C][gpio.binary_sensor:016]:   Pin: GPIO6
[23:42:42][C][htu21d:028]: HTU21D:
[23:42:42][C][htu21d:029]:   Address: 0x40
[23:42:42][C][htu21d:033]:   Update Interval: 60.0s
[23:42:42][C][htu21d:034]:   Temperature 'Garage Temperature'
[23:42:42][C][htu21d:034]:     Device Class: 'temperature'
[23:42:42][C][htu21d:034]:     State Class: 'measurement'
[23:42:42][C][htu21d:034]:     Unit of Measurement: '°F'
[23:42:42][C][htu21d:034]:     Accuracy Decimals: 1
[23:42:42][C][htu21d:035]:   Humidity 'Garage Humidity'
[23:42:42][C][htu21d:035]:     Device Class: 'humidity'
[23:42:42][C][htu21d:035]:     State Class: 'measurement'
[23:42:42][C][htu21d:035]:     Unit of Measurement: '%'
[23:42:42][C][htu21d:035]:     Accuracy Decimals: 1
```

### Which version of ESPHome has the issue?

Latest

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

SparkFun ESP32-S2 Thing Plus

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esp32:
  board: sparkfun_esp32s2_thing_plus
  framework:
    type: esp-idf

# Enable I2C
i2c:
  sda: 1
  scl: 2
  scan: true
  id: bus_a

# Enable logging
logger:

# Enable temperature and humidity sensor
sensor:
  - platform: htu21d
    temperature:
      name: "Garage Temperature"
      unit_of_measurement: "°F"
      filters:
        - multiply: 1.8
        - offset: 32
    humidity:
      name: "Garage Humidity"
    update_interval: 60s
```

Tonguc-Endem, 2023-05-04:
I have the same issue.  I made many forum posts and issues about this but for some reason the EspHome team persistently ignores this problem.  I had multiples of HTU21D devices in my house.  when HA notifies update, I update.  After updating EspHome and updating the firmware my HTU21D devices stopped working.  There are some I havent updated since EspHome version 2021.9.3 and they are working fine so I can not add any more fetures to those devices.  On my laptop I have installed EspHome 2021.9.3 but for some reason it requires that platformio core 6 must be installed.  That is very stupid.  EspHome 2021.9.3 requires platformio 5.2.0 but it will not compile unless I install platformio 6 which should not even be compatible.  I think the problem about HTU21D with EspHome is with platformio v 6.  

On 2021 I used to be able to install on my laptop with EspHome cli a HTU21D device without problem.  Now for some reason it won't even compile if I don't install platformio core 6 which is totally stupid.  platoformio 5.2.0 should be enough for EspHome 2021.9.3.

Try this yourself on your computer.
install old version of esphome with 
pip install esphome==2021.9.3
pip install platformio==5.2.0
esphome compile yourcode.yaml
esphome upload yourcode.yaml --device 192.168.*.*

try to get this to work maybe it will work for you.  This used to work for me.

for some reason now it complains about platformio6 and maybe platformio6 is the problem, I'm not sure.
When you run esphome compile, it runs platformio run -d yourcode which then complains about core 6.
There has to be a way to run EspHome 2021.9.3 with platformio 5.2.0
If you can get that to work I think, HTU21D will work that way.

t-az, 2023-06-24:
Figured I'd post a resolution I found if anyone else has this issue. I was able to set the "scan" boolean under my i2c configuration to "false" and manually defined the address of the HTU21D and that resolved the issue. Sensor is now communicating using the latest version of ESPHome. Weird bug but whatever.

fusionstream, 2024-05-03:
Does not work for me