esphome/issues #4414: No Wifi Connection After Upgrade to 2023.3.2

pums12, 2023-04-16:
### The problem

My Sonoff Basics (R2 version) will not connect to WiFi after upgrading to Version 2023.3.2

I have had the same problem on two Sonoff Basics that have been working fine. 

The strange thing is that when I power them from the flasher it will connect to Wifi (although it takes a while). When I look at the logs there are lots of "Association Expired" & "Auth Expired" events.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.4

### What platform are you using?

ESP8266

### Board

Sonoff Basic R2

### Component causing the issue

Wifi

### Example YAML snippet

```yaml
esphome:
  name: my-lamp

esp8266:
  board: esp01_1m

wifi:
  ssid: "XXXXXX"
  password: "XXXXXX"

logger:

api:

ota:
```

ssieb, 2023-04-17:
Some basics have difficulty with wifi.  See if anything in https://github.com/esphome/issues/issues/3816 helps you.

pums12, 2023-04-17:
I managed to get them working again by re-installing the Home Assistant Add-On and then using the Clean Build Files option.

Once I had flashed the devices they started working again.

closed 2023-04-17 by pums12