esphome/issues #4415: Error: Could not find one of 'package.json' manifest files in the package

sns5400, 2023-04-16:
### The problem

Afther make a simple yaml file i try to install this in my esp32  ESP-wroom-32D and receive this screen

[image]

### Which version of ESPHome has the issue?

ESPHome Current version: 2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.4.4

### What platform are you using?

ESP32

### Board

ESP-WROOM-32D

### Example YAML snippet

```yaml
esphome:
  name: esphome-web-997c98
  friendly_name: MyESP32

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "BPnyoqItOQyX/fx1igaBzBjCFr9/4//JuD3tmetDFeE="

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # optional manual ip
  manual_ip:
    static_ip: 192.168.1.26
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-997C98"
    password: "ps707300"

captive_portal:

output:
  - platform: gpio
    id: "my_little_LED"
    pin: 5

light:
  - platform: binary
    output: "my_little_LED"
    name: "My Little LED"
```

### Additional information

I found a simaler error here but there the speak about deleting  packages directory or platformio directory but i can not find such a name in the file system.

I run Home assistant on a rpi4 with unremovable memory e-mmc

How can i fix this problem

ssieb, 2023-04-16:
re-install the addon

closed 2023-04-16 by ssieb

hobojoe, 2023-04-16:
> re-install the addon

Hi, I had the deprecated version of ESPHome and removed it and installed the new one. Now this error happens to me. I have already reinstalled the addon several time, didn't fix it.

Any other suggestions?

Thanks!

sns5400, 2023-04-17:
There is a problem but,  it seems thad it is not importend because already closed and not by me. Reinstallation do not solf the problem.

sns5400, 2023-04-17:
Now afther remove esphome compleet from cmd,  reinstall via adon in HA it function again like description in the documentation.
Now we can go step by step further.

Thank you

hobojoe, 2023-04-17:
> Now afther remove esphome compleet from cmd, reinstall via adon in HA it function again like description in the documentation. Now we can go step by step further.
> 
> Thank you

Hi @sns5400 , thanks for your response ... Can you explain how you removed ESPHome from the cmd? It would be very helpful for me to solve my problem!

Thanks