esphome/issues #4416: Problem with dimming min_power and zero_means_zero

lieven121, 2023-04-16:
### The problem

Hi I have some led spots connected to a shelly RGBW2 flashed with espHome

My spots only go on at about 10% power so i tried to set min_power

It is not clear in the documentation what this number is, i was thinking percent but that doesn't seem right since at 0.01 the min brightness matches the same as 15% (when min_power is 0)

But then the real problem, i quickly discovered that is I set min power it doesn't shut down completely (this of course is intended) 
So i turn on zero_means_zero

The problem lies here, if i now turn on the lights to a low brightness it takes them a half a second to turn on.
Idem when I turn them off they stay at their custom min brightness a while.

What my guess is it that is happening is that the transition doesn't take into account that if min value is e.g 10% brightness 
that it should skip the timing of the first 10 percent, thus resulting in a half off/on time if is start/turn off the lights when brightness is 10% (adjusted so take 20% absolute)

I hope my explanation is clear.

I do only have confirmed this on my shelly RGBW2

Thanks already

### Which version of ESPHome has the issue?

2023.3.2 

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

Shelly RGBW2

### Component causing the issue

light

ssieb, 2023-04-16:
You need to provide more details.  You didn't provide your config.  What are the lights?  Maybe try adding `gamma_correct: 1`.

lieven121, 2023-04-17:
this is my config 
```
substitutions:
  prefix: Porch Kitchen Light
  device_name: porch-kitchen-light
  device_description: "Shelly RGBW2 connected to button and light in the porch on the kitchen side"

esphome:
  name: ${device_name}
  comment: "${device_description}"
  friendly_name: ${prefix}
  on_boot:
    priority: 600
    then: 
      lambda: |-
        id(button_1).publish_state("");

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: !secret ip_porch_kitchen_light
    gateway: !secret ip_gateway
    subnet: !secret ip_subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Porch-Kitchen-Light"
    password: "6iqhsSOh9VvW"

captive_portal:
    
status_led:
  pin: GPIO2
  
light:
  - platform: monochromatic
    restore_mode: ALWAYS_OFF
    name: "${prefix}"
    output: ch_w
    id: light_1

# Example output entry
output:
  - platform: esp8266_pwm
    id: ch_r
    pin: GPIO12
    frequency: 244 Hz
  - platform: esp8266_pwm
    id: ch_g
    pin: GPIO15
    frequency: 244 Hz
  - platform: esp8266_pwm
    id: ch_b
    pin: GPIO14
    frequency: 244 Hz
  - platform: esp8266_pwm
    id: ch_w
    pin: GPIO4
    frequency: 244 Hz
    min_power: 0.005
    max_power: 0.90
    zero_means_zero: true

binary_sensor:    
  - platform: gpio
    name: "${prefix} Button State"
    pin: 
      number: GPIO5
      # mode: INPUT_PULLUP
      # inverted: True
    # on_press:
    # - light.toggle: cupboard_light_sink
    # - light.toggle: cupboard_light_cooktop
    # device_class: motion
    id: button_1_raw
    # internal: true
    filters:
      # - delayed_on_off: 50ms
      - delayed_on: 20ms

    on_multi_click:
      - timing:
          - ON for 20ms to 400ms
          - OFF for at least 300ms
        then:
          - lambda: |-
              if(id(operating_mode_1).state == "Decoupled") return;
              auto call = id(light_1).toggle();
              call.perform();
          - text_sensor.template.publish:
              id: button_1
              state: !lambda 'return "click";'
      - timing:
          - ON for 20ms to 400ms
          - OFF for 20ms to 400ms
          - ON for 20ms to 400ms
          - OFF for at least 400ms
        then:
          - text_sensor.template.publish:
              id: button_1
              state: !lambda 'return "double_click";'
      - timing:
          - ON for at least 500ms
        then:
          - text_sensor.template.publish:
              id: button_1
              state: !lambda 'return "hold";'
    on_release:
      then:
        - lambda: |-
            id(button_1).publish_state("release");
  
sensor:
  - platform: adc
    pin: A0
    name: "${prefix} adc"
    update_interval: 5s
    accuracy_decimals: 4
    id: current_raw
    internal: true
  - platform: template
    name: "${prefix} power"
    unit_of_measurement: W
    accuracy_decimals: 1
    device_class: power
    lambda: return id(current_raw).state;
    filters:
      - calibrate_linear:
        - 0.4000 -> 0.0
        - 0.4043 -> 0.14
        - 0.4053 -> 0.20
      - multiply: 12 # 12V lamp

text_sensor:
  - platform: template
    name: '${prefix} Button action'
    id: button_1
    on_value: # When a state is set
      then:
        - if:
            condition:
              text_sensor.state:
                id: button_1
                state: ''
            else: # If non-empty
              - delay: 100ms
              - text_sensor.template.publish:
                  id: button_1
                  state: !lambda 'return "";' # Reset to empty

select:
  - platform: template
    name: "${prefix} Operating Mode SW"
    id: operating_mode_1
    entity_category: config
    optimistic: true
    options:
      - Decoupled
      - Control Relay
    restore_value: True
    initial_option: Control Relay
    set_action: 
      then:
        - lambda: |-
            return;
```

see GPIO4 for the min_power etc

ssieb, 2023-04-17:
And what is the light?
Did you try my config suggestion?

lieven121, 2023-04-17:
The light is 

```
light:
  - platform: monochromatic
    restore_mode: ALWAYS_OFF
    name: "${prefix}"
    output: ch_w
    id: light_1
```

Ill try the gamma tomorrow

But i don't clearly get what it does

ssieb, 2023-04-17:
I meant what is the physical light that you are controlling?

lieven121, 2023-04-18:
> I meant what is the physical light that you are controlling?

Oh sorry I was confused 😅

Its an array of 12v led spots

lieven121, 2023-04-18:
@ssieb 
Thanks
it does seem to work but i don't really get why 
Could you explain why it is gamma correct and not min power i should use ?

closed 2023-04-18 by ssieb