esphome/issues #4419: ESPHome Firmware Update unavailable after full reboot

alvinchen1, 2023-04-18:
### The problem

Run HAOS

Home Assistant 2023.4.5
Supervisor 2023.04.0
Operating System 10.0
Frontend 20230411.1 - latest

When performing a full system reboot (not simply restarting HA) or if power loss etc, all ESPHome Firmware entities are unavailable.  

[image]

They remain unavailable until I restart HA once more.  Restarting HA (and no other change) restores all the ESPHome Firmware entities.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

Several boards, all ESPhome Devices

### Component causing the issue

N/A

joshuaspence, 2023-04-19:
I am seeing this after updating to HAOS 10

DAVe3283, 2023-04-19:
This is a Home Assistant problem: home-assistant/core#88155

alvinchen1, 2023-04-19:
> This is a Home Assistant problem: [home-assistant/core#88155](https://github.com/home-assistant/core/issues/88155)

Thanks for the link to the HA Core issue!  Would it help to keep this open so others looking for this problem don't create other issues, then close it when resolved at HA Core?

DAVe3283, 2023-04-20:
I think it makes sense to leave this open until HA core is fixed, if nothing else to try and help people find the HA core issue.

mstuij, 2023-05-05:
I have still this issue with HA 2023.5.1

alvinchen1, 2023-05-06:
Rebooted entire OS after 2023.5.2 and seems to be fixed.

closed 2023-05-06 by alvinchen1