esphome/issues #4420: Ethernet PHY on ESP32 board consumes too much power on deep sleep

tracestep, 2023-04-18:
### The problem

When I put the board on deep sleep (e.g,: `deep_sleep.enter`), the board still consumes over ~70 mA. I have traced the issue to the fact that, before entering deep sleep, the ethernet PHY on the board is not powered down / put into low power mode. I will link a PR which works at least for the LAN8720 on my board, but probably others, as it uses the esp-idf HAL.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

Irrelevant

### What platform are you using?

ESP32-IDF

### Board

WT32-ETH01

### Component causing the issue

ethernet

### Example YAML snippet

```yaml
ethernet:
  id: eth
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

deep_sleep:
  id: sleeper
  touch_wakeup: false
  esp32_ext1_wakeup:
    pins: [GPIO39, GPIO35, GPIO14]
    mode: ANY_HIGH

button:
  - id: button_sleep
    platform: template
    name: Sleep
    icon: mdi:sleep
    on_press:
      then:
        - deep_sleep.enter:
            id: sleeper
```

jesserockz, 2023-04-19:
I'm curious as to when a device would have ethernet, but not a power source to just stay awake all the time?

functionpointer, 2023-04-19:
Wow very nice!
Could be useful in an offgrid solar setup

tracestep, 2023-04-19:
It does have a power source, but needs to last a long time waiting and logging events in the event of an extended power outage. Same applies for wi-fi. Ethernet in that case is for more reliable communication.

closed 2023-04-19 by jesserockz