esphome/issues #4424: Home Assistant Time Not Updating

DAVe3283, 2023-04-18:
### The problem

The [Home Assistant Time](https://esphome.io/components/time/homeassistant.html) component only updates when Home Assistant or ESPHome logging first connects to the ESP.

After some time (1-5 days), the local ESP clock has drifted significantly (sometimes a minute), which is really annoying when it is displaying the time.

I can get the time to re-sync by connecting to the logs with ESPHome, but that isn't a reasonable thing to need to do every couple days just to have an accurate clock displayed.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.2

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

time/homeassistant

### Example YAML snippet

```yaml
esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: d1_mini

logger:

api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

globals:
  - id: GLOBAL_LAST_TIMESYNC
    type: char*
    restore_value: no
    initial_value: '"Never\0\0\0\0"'
  - id: GLOBAL_TIMESYNC_COUNT
    type: uint8_t
    restore_value: no
    initial_value: "0"

time:
  - platform: homeassistant
    id: ha_time
    update_interval: 30s
    on_time_sync:
      then:
        - lambda: |-
            std::strncpy(id(GLOBAL_LAST_TIMESYNC), id(ha_time).now().strftime("%H:%M:%S").c_str(), 8);
            id(GLOBAL_TIMESYNC_COUNT)++;
        - logger.log:
            format: "Synchronized system clock"
            level: WARN

display:
  - platform: ssd1306_i2c
    model: SSD1306 64x48
    address: 0x3C
    lambda: |-
      it.fill(COLOR_OFF);
      it.printf(32, 0, id(fnt9), fg, TextAlign::TOP_CENTER, "%d %s", id(GLOBAL_TIMESYNC_COUNT), id(GLOBAL_LAST_TIMESYNC));
```

### Anything in the logs that might be useful for us?

```txt
Notably, "Synchronized system clock" only appears when connecting an additional ESPHome log viewer or (re)connecting Home Assistant. It does not appear every 30 seconds as expected from the YAML. In fact, after running for 24+ hours, it never re-synchronizes.
```

### Additional information

I added some debug code to components/homeassistant/time/homeassistant_time.cpp files and verified that it does call `api::global_api_server->request_time();`. However, the time is not actually updated. Not sure if there is an API problem, or what.

Meisam-kh, 2023-05-18:
Same issue

talakoski, 2023-07-21:
@Meisam-kh is correct. After adding a new ESPHome device you also need to do the integration in Home Assistant  😀

This is even said in the documentation: "Once your configuration is installed on your ESP device and is online, it will be automatically discovered by Home Assistant and offered to set up on your integrations screen"

DAVe3283, 2023-07-21:
I guess I wasn't clear enough in the original issue. The device **is** added to Home Assistant, and I can interact with it. But the time only syncs when it first connects, then never syncs again, unless I do one of the following:

1. remove and re-add the device to Home Assistant
2. restart Home Assistant
3. connect to the device logs with ESPHome

Since I rarely do any of those things, the clock drifts out of sync within a few days. The clock should periodically re-sync to Home Assistant, ideally at least once a day. Timekeeping of my ESPs all appear to drift at least a second a day.

marc-gist, 2023-12-22:
this ever get resolved? I think I have a similar issues, which affects `total_daily_energy` resets :(

DAVe3283, 2023-12-22:
I just confirmed that as of ESPHome 2023.11.6 and Home Assistant 2023.12.1 this is still a problem. The 3 "workarounds" I listed above still work to get the time updated on the ESPHome devices, but it does not actually update the time on the device except just the initial connection.

bdraco, 2023-12-22:
AFAICT, `update_interval` is not a valid option for the `homeassistant` `time` platform

jesserockz, 2023-12-22:
As above `HomeassistantTime` inherits from `RealTImeClock` which inherits `PollingComponent`. The default update interval for all time entities is `15 minutes`. The `update` function inside `HomeassistantTime` sends a time request to HA which will respond with the current time in a separate response message.

```c++
void HomeassistantTime::update() { api::global_api_server->request_time(); }
```

```c++
void APIConnection::on_get_time_response(const GetTimeResponse &value) {
  if (homeassistant::global_homeassistant_time != nullptr)
    homeassistant::global_homeassistant_time->set_epoch_time(value.epoch_seconds);
}
```

So, it should be updating the internal clock every 15 minutes from HA

bdraco, 2023-12-22:
```

2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Got message of type GetTimeRequest: 
2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Sending GetTimeResponse: epoch_seconds: 1703232420

2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Sending frame: [0005250da4438565]

```

setting the time update interval to 30s definitely doesn't work.
The initial time is requested on connect
```

2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Got message of type GetTimeRequest: 
2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Sending GetTimeResponse: epoch_seconds: 1703232420

2023-12-21 22:07:00.662 DEBUG    officenewgli89proxy @ 192.168.107.204: Sending frame: [0005250da4438565]

```

but it doesn't seem to poll it again

bdraco, 2023-12-22:
I don't see any obvious bugs in the code.

Will have to dig a bit more to see if the polling isn't getting set up, setting up incorrectly, or being cancelled somehow

bdraco, 2023-12-22:
Its also not a case of the update interval failing to be overridden as it doesn't happen at 15m either

bdraco, 2023-12-22:
Too late here to make progress on this. Will have to look tomorrow or after the holiday

bdraco, 2023-12-26:
[13:42:14.824][D][homeassistant.time:024]: Updating Home Assistant Time...
[13:42:24.827][D][homeassistant.time:024]: Updating Home Assistant Time...
[13:42:34.821][D][homeassistant.time:024]: Updating Home Assistant Time...
[13:42:44.823][D][homeassistant.time:024]: Updating Home Assistant Time...
[13:42:54.824][D][homeassistant.time:024]: Updating Home Assistant Time...

Its calling request_time

bdraco, 2023-12-26:
```
[13:47:04.769][D][homeassistant.time:024]: Updating Home Assistant Time...
[13:47:04.769][D][api:321]: APIServer::request_time Requesting time from Home Assistant...
[13:47:04.769][D][api:323]: APIServer::request_time Sending time request to aioesphomeapi
```

but it actually sends nothing

bdraco, 2023-12-26:
`    if (!client->remove_ && client->connection_state_ == APIConnection::ConnectionState::CONNECTED)` is probably false

bdraco, 2023-12-26:
I think this has been broken for a long time, maybe 4 years

closed 2023-12-28 by clydebarrow