esphome/issues #4427: Unable to update: Component media_player.i2s_audio requires component i2s_audio.

radek-kafka, 2023-04-20:
### The problem

Hi,

I'm trying to do update to 2023.04 and receiving this error message:

INFO Reading configuration /config/esphome/m5stack-atom-speaker-kit.yaml...
Failed config

media_player.i2s_audio: [source /config/esphome/m5stack-atom-speaker-kit.yaml:49]
  
  Component media_player.i2s_audio requires component i2s_audio.
  platform: i2s_audio
  id: media_out
  name: M5Stack Atom Speaker Kit
  dac_type: external
  i2s_lrclk_pin: GPIO21
  i2s_dout_pin: GPIO25
  i2s_bclk_pin: GPIO22
  mode: mono

### Which version of ESPHome has the issue?

2023.04

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

m5AtomLite

### Component causing the issue

I2_audio

### Example YAML snippet

```yaml
media_player:
  - platform: i2s_audio
    id: media_out
    name: ${friendly_name}
    dac_type: external
    i2s_lrclk_pin: GPIO21
    i2s_dout_pin: GPIO25
    i2s_bclk_pin: GPIO22
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/m5stack-atom-speaker-kit.yaml...
Failed config

media_player.i2s_audio: [source /config/esphome/m5stack-atom-speaker-kit.yaml:49]
  
  Component media_player.i2s_audio requires component i2s_audio.
  platform: i2s_audio
  id: media_out
  name: M5Stack Atom Speaker Kit
  dac_type: external
  i2s_lrclk_pin: GPIO21
  i2s_dout_pin: GPIO25
  i2s_bclk_pin: GPIO22
  mode: mono
```

d-two, 2023-04-20:
https://esphome.io/changelog/2023.4.0.html checkout Breaking Changes

radek-kafka, 2023-04-20:
Thank you very much . Now working perfectly.. For next I will "read before write" :) ..

closed 2023-04-20 by radek-kafka