esphome/issues #4428: Growatt documentation: example is invalid due to use of reserved uart1 uart

pieterbos, 2023-04-20:
### The problem

The documentation of the Growatt Solar Sensor Component lists the following example, at https://esphome.io/components/sensor/growatt_solar.html :

```yaml
# Example configuration entry
uart:
  - id: uart1
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: GPIO3
```

This used to work in an older version. However, this no longer works in current versions, since uart1 is now reserved internally, and the configuration is now invalid. It can be fixed easily, by changing the yaml to:

```
# Example configuration entry
uart:
  - id: uart_modbus
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: GPIO3

modbus:
  uart_id: uart_modbus
  flow_control_pin: GPIO4
```

which is a correct and working configuration. Tested with my own setup, and after changing it works again.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

-

### What platform are you using?

ESP8266

pieterbos, 2023-04-21:
You do not need to rewire. The fix is simply renaming uart1 in the config file to uart_modbus. Like in the example above. Then it works again. I did the same with my Shinewifi-S.

This issue is to update the examples in the documentation to reflect this change.

wfroemgen, 2023-04-21:
Now I get it. Thanks for the hint and the fast reply.

pvhpsb, 2023-04-22:
Thans for the solution

closed 2023-04-23 by nagyrobi