esphome/issues #4430: Unable to compile bme680_bsec sensor on ESP32-C3

fran6120, 2023-04-20:
### The problem

I have tried it with both esphome 2023.4.0 and 2023.5.0-dev.

If I set `framework: esp-idf` I can't compile it and it shows error `This feature is only available with frameworks ['arduino']`.

If I set `framework: arduino` I can't compile either, and this time the error shown is `This feature is only available on ['esp8266'].`

In each attempt always cleaning the build files

``` yaml
esphome:
  name: estacionc3
  friendly_name: estacionc3
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: lolin_c3_mini
  framework:
    type: arduino

logger:

api:

i2c:
  sda: GPIO7
  scl: GPIO6
  scan: True
  frequency: 400kHz

wifi:
  networks:
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1
      manual_ip:
        static_ip: 192.168.1.211
        gateway: 192.168.1.1
        subnet: 255.255.255.0
        dns1: 192.168.1.1
  fast_connect: true

bme680_bsec:
  address: 0x77
  temperature_offset: 0
  iaq_mode: static
  sample_rate: ulp
  state_save_interval: 6h

  - platform: bme680_bsec
    bme680_bsec_id: bme680_internal
    temperature:
      name: "Temperatura"
      id: temperatura_bme680
      sample_rate: lp
      filters:
        - median
    pressure:
      name: "Presión"
      id: presion_bme680
      sample_rate: lp
      filters:
        - median
    humidity:
      name: "Humedad Relativa"
      id: humedad_bme680
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      name: "Resistencia Del GAS"
      filters:
        - median
    iaq:
      name: "Calidad Del Aire"
      filters:
        - median
    iaq_accuracy:
      name: "Índice Calidad De Aire"
    co2_equivalent:
      name: "CO2 Equivalente"
      filters:
        - median
    breath_voc_equivalent:
      name: "VOC Equivalente"
      filters:
        - median

text_sensor: 
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "BME680 Precisión Calidad De Aire"
```

### Which version of ESPHome has the issue?

2023.4.0, 2023.5.0-dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

### Board

ESP32C3 (Lolin C3 Mini)

### Component causing the issue

bme680_bsec

### Example YAML snippet

```yaml
esphome:
  name: estacionc3
  friendly_name: estacionc3
  platformio_options:
    board_build.flash_mode: dio

esp32:
  board: lolin_c3_mini
  framework:
    type: esp-idf

logger:

api:

i2c:
  sda: GPIO7
  scl: GPIO6
  scan: True
  frequency: 400kHz

wifi:
  networks:
    - ssid: !secret wifi_ssid1
      password: !secret wifi_password1
      manual_ip:
        static_ip: 192.168.1.211
        gateway: 192.168.1.1
        subnet: 255.255.255.0
        dns1: 192.168.1.1
  fast_connect: true

bme680_bsec:
  address: 0x77
  temperature_offset: 0
  iaq_mode: static
  sample_rate: ulp
  state_save_interval: 6h

  - platform: bme680_bsec
    bme680_bsec_id: bme680_internal
    temperature:
      name: "Temperatura"
      id: temperatura_bme680
      sample_rate: lp
      filters:
        - median
    pressure:
      name: "Presión"
      id: presion_bme680
      sample_rate: lp
      filters:
        - median
    humidity:
      name: "Humedad Relativa"
      id: humedad_bme680
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      name: "Resistencia Del GAS"
      filters:
        - median
    iaq:
      name: "Calidad Del Aire"
      filters:
        - median
    iaq_accuracy:
      name: "Índice Calidad De Aire"
    co2_equivalent:
      name: "CO2 Equivalente"
      filters:
        - median
    breath_voc_equivalent:
      name: "VOC Equivalente"
      filters:
        - median

text_sensor: 
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "BME680 Precisión Calidad De Aire"
```

mort666, 2023-05-08:
Also suffering from the same issue with both ESP32 S3 boards and RP2040 Raspberry Pico's all of which worked flawlessly prior to 2023.4.0.

maikischa, 2023-07-01:
Same issue here. I guess the BSEC Lib must be updated to 2.X

pszober, 2023-07-18:
Also same for me on ESP32-S3 chip. Is there any update on whether this is planned to be fixed?

clomads, 2023-07-22:
I am also unable to compile for BSEC on C3

pszober, 2023-07-22:
I was able to do it using either Beta or Dev version of esphome in HA (can't remember which one exactly) using the code I published above. It worked until about maybe 3 months ago abd then I wasn't able to compile it anymore.

mharris1984, 2023-08-11:
I am also running into this same issue.... is there any update on the status of resolving this???

ssieb, 2023-08-11:
There's nothing to do until https://github.com/esphome/esphome/pull/4585 is finished.  The bsec library does not support these chips.

pszober, 2023-08-12:
But then why did it compile for me months ago with the exact same code but it won't do it now?

ssieb, 2023-08-13:
I don't know.  As far as I know, it shouldn't have.

jddonovan, 2023-10-12:
@jesserockz  @bisbastuner

Issue related to:
[https://github.com/esphome/esphome/pull/4616](https://github.com/esphome/esphome/pull/4616)
[https://github.com/esphome/esphome/pull/3550](https://github.com/esphome/esphome/pull/3550)

Seems we have a regression or something's changed again.

I tested on C3, cannot compile.

critchards, 2024-02-07:
+1.
I happen to have a BME688 sensor already running fine on an ESP32 so I will resist the urge to update it.
But Im trying to deploy a new one, this time using a XIAO ESP32C3 on ESPHOME2023.12.9. No luck, same error as above, `this feature is only available on ['esp8266']`

dagreatzombi, 2024-04-06:
April bump for the BME680 BSEC on esp32-c3.

gretel, 2024-05-23:
just ran into this with a m5stamp c3 which is pretty recent.

gubagu, 2024-12-18:
Still an issue on  19 Dec 2024 on ESP32 C3 super mini

```
Core                2024.12.4
Supervisor          2024.12.0
Operating System    14.0
```

AlbertoMPinto, 2025-03-14:
I had a very hard time getting the BME680 to work with an esp32-s3 devkitc-1 n32r8.
After many attempts, a lot of research, and using the Bosch 68x libraries, I got it with the following code:

 ```
esphome:
  name: bme680
  friendly_name: BME680
  min_version: 2024.11.0
  name_add_mac_suffix: false
  platformio_options:
    board_build.arduino.memory_type: opi_opi

esp32:
  board: esp32-s3-devkitc-1 #N32
  framework:
    type: arduino
  flash_size: 32MB
  variant: esp32s3
  
wifi:
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.125
    # Set this to the IP address of the router.
    gateway: 192.168.1.254
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  fast_connect: true   

  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
captive_portal:

###

web_server:
  local: true
  version: 3
  
# Enable logging
logger:
  #level: debug
  level: verbose
  
# Enable Home Assistant API
api:
  encryption:
    key: "lF1LSZyzWkKrYnmxhL+H5dumoTWUvKXqvPTKSlys4mQ="

# Allow Over-The-Air updates
ota:
  - platform: esphome

time:
  - platform: homeassistant
    id: esptime

button:
  - platform: restart
    name: "Restart"

### ### ###

# Configuration of I²C BUS entry for ESP32 (em https://esphome.io/components/i2c.html)
i2c:
  sda: GPIO08 #cabo branco
  scl: GPIO09 #cabo amarelo
  scan: true
  #id: bus_a #(Opcional) - Especificar manualmente o ID para este barramento I²C se forem necessários vários barramentos I²C.

# Minimal example configuration with common sensors
bme68x_bsec2_i2c:
  address: 0x76
  model: bme680
  operating_age: 28d
  sample_rate: LP
  supply_voltage: 3.3V
 
sensor:
  - platform: bme68x_bsec2
    # ID of the bme680_bsec component to use for the next sensors.
    # Useful when working with multiple devices
    #bme68x_bsec2_id: bme680_internal

    temperature:
      # Temperature in °C
      name: "Temperature"
      sample_rate: lp
      filters:
        - median
    pressure:
      # Pressure in hPa
      name: "Pressure"
      sample_rate: lp
      filters:
        - median
    humidity:
      # Relative humidity %
      name: "Humidity"
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      # Gas resistance in Ω
      name: "Gas Resistance"
      filters:
        - median
    iaq:
      # Indoor air quality value
      name: "IAQ"
      id: iaq
      filters:
        - median
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "Numeric IAQ Accuracy"
    co2_equivalent:
      # CO2 equivalent estimate in ppm
      name: "CO2 Equivalent"
      filters:
        - median
    breath_voc_equivalent:
      # Volatile organic compounds equivalent estimate in ppm
      name: "Breath VOC Equivalent"
      filters:
        - median

text_sensor:
  - platform: bme68x_bsec2
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "IAQ Accuracy"

##

  - platform: template
    name: "IAQ Classification"
    lambda: |-
      if ( int(id(iaq).state) <= 50) {
        return {"Excellent"};
      }
      else if (int(id(iaq).state) >= 51 && int(id(iaq).state) <= 100) {
        return {"Good"};
      }
      else if (int(id(iaq).state) >= 101 && int(id(iaq).state) <= 150) {
        return {"Lightly polluted"};
      }
      else if (int(id(iaq).state) >= 151 && int(id(iaq).state) <= 200) {
        return {"Moderately polluted"};
      }
      else if (int(id(iaq).state) >= 201 && int(id(iaq).state) <= 250) {
        return {"Heavily polluted"};
      }
      else if (int(id(iaq).state) >= 251 && int(id(iaq).state) <= 350) {
        return {"Severely polluted"};
      }
      else if (int(id(iaq).state) >= 351) {
        return {"Extremely polluted"};
      }
      else {
        return {"error"};
      }  
#######
# eof #
#######
 ```

Hope it helps

jesserockz, 2025-05-26:
The `bsec` library only supports esp8266 and the original esp32 (not c3, s2, s3 etc). If you want to use a variant, you need to use the [bsec2 component](https://esphome.io/components/sensor/bme68x_bsec2).

In ESPHome 2025.6.0, `bsec2` will be available to use when using esp-idf as your framework.