esphome/issues #4432: flip_x commit breaks reversed display

voed, 2023-04-20:
### The problem

The last [commit ](https://github.com/esphome/esphome/pull/4555) from @rafal83 breaks rotated and reversed displays. My clock is installed upside down so i used to rotate & flip it and it worked fine:
```
    rotate_chip: 180
    reverse_enable: true
```
But with that commit the display is completely broken(see details)

### Which version of ESPHome has the issue?

2023.04

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.3

### What platform are you using?

ESP32

### Component causing the issue

MAX7219digit

### Example YAML snippet

```yaml
display:
  - platform: max7219digit
    cs_pin: GPIO25
    num_chips: 4
    intensity: 0
    rotate_chip: 180
    reverse_enable: false
    flip_x: false
    lambda: |-
      it.print(0, 5, id(digit_font), TextAlign::CENTER_LEFT, "1234567890");

```
<details>
[image]
</details>
```
    reverse_enable: false
    flip_x: true

```
<details>
[image]
</details>

```
    reverse_enable: true
    flip_x: true

```
With reverse enable the text is even duplicated from both direction
<details>
[image]
</details>

```
    reverse_enable: true
    flip_x: false

```
<details>
[image]
</details>
```

martin131, 2023-04-20:
same problem

voed, 2023-04-20:
If anyone interested in easy and quick fix, i have created a reverse commit. Just add to your config:
```
external_components:
  - source: github://voed/esphome@dev
    components: [ max7219digit ]
```

sheminasalam, 2023-04-20:
same issue

ajakubek80, 2023-04-20:
same problem

tomber71, 2023-04-20:
same problem

jan-gerard, 2023-04-21:
Similar issue. My display is not rotated, but I need to use reverse_enable: true, and now the individual displays are flipped in X. The flip_x option does not have any effect. I tested all four boolean combinations of reverse_enable and flip-x, and only reverse_enable has an effect, but not the right one. 
The change "Add option flip_x #4555" seems to break the display. By the way, I use an esp8266, no esp32, but that should not matter.

closed 2023-04-22 by jesserockz