esphome/issues #4433: media_player starts with maximal volume

koemat, 2023-04-20:
### The problem

Since update to ESPhome 2023.4.0 media players are starting with max volume. 

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

i2s_audio, media_player

### Example YAML snippet

```yaml
i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO5

media_player:
  - platform: i2s_audio
    name: "${friendly_name}"
    dac_type: external    
    i2s_dout_pin: GPIO26    
    mode: mono
    mute_pin:
      number: GPIO21
      inverted: true
```

d-two, 2023-04-20:
Same here!!! In HA the Volume is correct but the ESP start with 100% Volume...
I think the problem can be this?! Here will the volume set...

[image]

monsieurlatte, 2023-04-20:
Sort of the same here, for me even if I change it in HA it wont stay, it immediately goes back to full volume.

monsieurlatte, 2023-04-23:
Love all the help from the devs on this lol, my home automation notifications are useless until this is fixed, next time I'll be smart and save a copy of the firmware before I update esphome, first time this has ever bit me in the butt but here we are.

alexd321, 2023-04-23:
Same here, I cannot change volume since update, stuck at 100%. Using the new i2s_audio component.

@monsieurlatte if you use Home Assistant and made a backup before the add-on update, you should be able to restore and then re-flash the esp32 (would have to remove the i2s_audio section)

monsieurlatte, 2023-04-24:
Sadly I didnt make a backup, I usually dont bother but for core upgrades.  May have to spin up a sep instance and do that.

monsieurlatte, 2023-04-27:
Nice find!  Good to know about a workaround but I'm still a little bummed there has been zero traction on this.

RoelHermus, 2023-04-27:
With the esphome  update at 27 april, there is something change with I2S. It works better. But when the mediaplayer is in idle mode I can change the volume slider, but the actual volume is not changed.

RoelHermus, 2023-05-03:
Detail of the problem: On playing the sound level was set on was 80% and was switch the idle mode.
I can move the slide but the percent indicator stay on 80%.
When I start the player again, its play the sound on 100%.
After moving the slider, it picks up the new volume setting.

The screen print is in idle mode. Percent indicator on 80% Slider on 25:
[image]

stiw47, 2023-05-05:
I just built my first ESP32 + PCM5102a ESPHome media player, to find that I done that in "wrong moment", lol (regarding this audio issue). I will compile it tonight with older version of ESPHome to see how was act before. Yes, I understand you folks, lot of you are using HassOS. I never used HassOS, so I don't know for sure, but if I remember correct, you are not able to revert ESPHome add-on back to older version. I am running everything (HA, ESPHome, etc.) as Docker containers, so I can install as much as I want separate instances of ESPHome. If anybody wants, I can do a favor, just provide me your yaml, and I can compile binary for you. If I understood this topic correct, last ESPHome version where audio on boot worked correct was 2023.3.2?

khenderick, 2023-05-06:
For people who are now stuck with a mediaplayer that does not work properly, I've created a ["legacy versions" addon repository](https://github.com/khenderick/esphome-legacy-addons) so you can temporarily install a legacy version next to the current one for just those devices.

If you have new insights, or any other means to help to progress this issue, make sure to help :slightly_smiling_face:.

jesserockz, 2023-05-13:
This is fixed in https://github.com/esphome/esphome/pull/4813

closed 2023-05-13 by jesserockz

Zagur, 2023-05-20:
> This is fixed in [esphome/esphome#4813](https://github.com/esphome/esphome/pull/4813)

I have seen that this problem is solved in firmware version 2023.5.1, but my ESP32 has this version but the same thing still happens.

The media_player is always at 100% and even if you lower the volume, either with the slide or otherwise it always goes to 100%.

I am also using the Home Assistant Add-on for RaspiAudio Muse Luxe and I can only turn the volume down when something is playing. At the end of the playback it goes back to 100% volume.

If you need anything else, let me know and I'll let you know.