esphome/issues #4434: New I²S Audio Component forces wasting GPIOs if internal DAC is used.

Rogash1, 2023-04-20:
### The problem

**Before the introduction of  I²S Audio Component:**

media_player:
  - platform: i2s_audio
    name: ESPHome I2S Media Player
    id: dac_speaker
    dac_type: internal #this only uses GPIO25
    mode: stereo
    
  **After the introduction of  I²S Audio Component**  
  
  i2s_audio:
     id: bus_i2s
     i2s_lrclk_pin: GPIO_unused
     i2s_bclk_pin: GPIO_unused
  
  media_player:
     - platform: i2s_audio
       name: ESPHome I2S Media Player
       id: dac_speaker
       dac_type: internal #this only uses GPIO25 or
       mode: stereo
       
  After introducing the new  I²S Audio Component, you need to have an i2s_audio object that requires two GPIOs.
  -If you only use the internal DAC of the ESP32 this forces you two waste two GPIOs. 
  -If you only use a single i2s device you can hard wire its i2s_lrclk_pin to GND or VCC. In this case, you still waste one GPIO.
  
  In my opinion, the GPIOs shouldn't be mandatory but optional.
  
  I currently cant upgrade ESPHome (actually need to downgrade) since I only use the internal DAC in my design and have no free GPIO to waste.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Every

### What platform are you using?

ESP32

### Board

ESP32dev

### Component causing the issue

i2s_audio

scottcopus, 2023-04-22:
I'm also using the internal DAC. I might have some free GPIO to throw at it, but it seems like it should be optional. I agree, or find a workaround?

vyruz1986, 2023-05-03:
The bigger issue here, imho, is that this makes it very unclear how to setup the `media_player` component with the internal DAC, since people (like me) will be looking for which pins they need to assign to `i2s_lrclk_pin` and `i2s_bclk_pin` for the `i2s_audio` component, while they should only be specifying 1 pin

Rogash1, 2023-05-11:
My suggestion would be to make all pins in the I2S component optional.
The components that use the bus must then check whether the pins they need have been declared. In the case that optional pins required by an I2S device have not been declared, errors should be generated when instantiating it.
Each I2S device must specify via an optional attribute whether it uses the left or right audio channel, both channels or none. The I2S Media Player component would then specify no pins to use and a pin attribute to specify which of the internal DACs to use if no external DAC is used.

SCratORS, 2023-07-17:
I wonder when this problem will be fixed?
If an internal DAC is used, then the i2s_audio component should be optional in principle.
I have to put an empty GPIO pin for compilation or roll back to a previous version.
*facepalm*
Every time, I'm afraid I'll update ESPHome. Constantly after updating something breaks.

Tonymac32, 2023-10-31:
I just stumbled across this today attempting to play with my Obsidian ESP32 board and the media_player component, the documentation/examples are wrong/won't build (https://esphome.io/components/media_player/i2s_audio#internal-dac), and the only place I've found reference to it is here.