esphome/issues #4436: Delta filter not working under Kalman-combinator

Aljosha89, 2023-04-20:
### The problem

After update 2023.4.0 is no longer possible to use delta filter under kalman-combinator based sensor:

[image]

Tried to but value in integer or percentage, but there is no change on the problem. Same code under sensor still works. Before update code was working fine.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.4

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

kalman_combinator

### Example YAML snippet

```yaml
- platform: kalman_combinator
  name: "Kalman level"
  id: kalman_level_tank
  unit_of_measurement: "%"
  process_std_dev: 0.020
  sources:
    - source: level_tank
      error: 0.850
    - source: level_tank_bis
      error: 0.850
  filters:
  - filter_out: nan
  - delta: 0.2
```

### Anything in the logs that might be useful for us?

```txt
Delta filter requires a positive number or percentage value.
```

### Additional information

[image]

ssieb, 2023-04-20:
Try compiling it and see what error message that gives.

Aljosha89, 2023-04-20:
Here we are:

[image]

ssieb, 2023-04-21:
Ok, I can reproduce it, but that's very strange!

ssieb, 2023-04-21:
Actually, there is a clue in the compile logs.  It's doing something weird with the parameter, converting to a list and adding a `False`.

jesserockz, 2023-04-21:
I have made a PR to fix the delta filter in https://github.com/esphome/esphome/pull/4723.

You should be able to test with this YAML added to your config:

```yaml
external_components:
  - source: github://pr#4723
    components: sensor
```

Aljosha89, 2023-04-21:
> I have made a PR to fix the delta filter in esphome/esphome#4723<https://github.com/esphome/esphome/pull/4723>.

Yes, now I can compile it and it works properly.

Waiting for definitive fix, thanks

closed 2023-04-23 by jesserockz