esphome/issues #4442: Multiple CORS Access-Control-Allow-Origin

segeda, 2023-04-21:
### The problem

I am trying to read `http://my-esphome.local/events` events stream. But server sends multiple (twice) `Access-Control-Allow-Origin` headers and browser rejects connection. I found out that this reason is https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS/Errors/CORSMultipleAllowOriginNotAllowed.

### Which version of ESPHome has the issue?

2023.3.2

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

m5stack-atom

### Component causing the issue

web_server

### Example YAML snippet

```yaml
esphome:
  name: m5stack-atom-lite
  friendly_name: M5Stack ATOM Lite

esp32:
  board: m5stack-atom
  framework:
    type: arduino

logger:
  level: INFO

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack-Atom-Lite"
    password: !secret ap_password

captive_portal:

web_server:
  port: 80
  ota: False
```

### Anything in the logs that might be useful for us?

```txt
curl -I http://m5stack-atom-lite.local/events                                                                

HTTP/1.1 500 Internal Server Error
Content-Length: 0
Access-Control-Allow-Origin: *
Access-Control-Allow-Origin: *
Connection: close
Accept-Ranges: none
```

### Additional information

Error in browser console

`Acess to resource at 'http://m5stack-atom-lite.local/events' from origin 'http://localhost:8000' has been blocked by CORS policy: The 'Access-Control-Allow-Origin' header contains multiple values '*, *', but only one is allowed.`

snetsplit, 2023-04-27:
I'm only getting one on an ESP8266 device. Unfortunately I don't have any esp32 devices to test it. 
```
curl -I -X GET http://washroom-switch.lan/events 
HTTP/1.1 200 OK
Content-Type: text/event-stream
Access-Control-Allow-Origin: *
Cache-Control: no-cache
Connection: keep-alive
Accept-Ranges: none
```

I tried digging through the source code but I couldn't find any reason it should be different ESP32.  If you try:
`curl -I http://m5stack-atom-lite.local/`

What do you get? 

I personally found I did get two CORS headers for those, but the cause was different.

segeda, 2023-04-27:
The result is the same. 2x CORS header.

```
curl -I http://m5stack-atom-lite.local/
HTTP/1.1 200 OK
Content-Length: 174
Content-Type: text/html
Access-Control-Allow-Origin: *
Access-Control-Allow-Origin: *
Connection: close
Accept-Ranges: none
```

snetsplit, 2023-04-27:
I tried compiling your yaml to see if I could find anything, but no luck.

If you haven't tried it already, I'd try using the "Clean Build Files" option and see if that changes anything

segeda, 2023-05-02:
With sensors is the result the same. (2x CORS for `/`, 1x CORS for `/events`). "Clean Build Files" fixes it. Thanks you 🙏

snetsplit, 2023-05-02:
No problem.  I have a PR open that includes a fix for the / double CORS:
[PR:4691](https://github.com/esphome/esphome/pull/4691/commits/54479b63e55417b24f8dd93761e9f0312a222f7f)

Just waiting for it get accepted/rejected.

segeda, 2023-05-22:
Its weird, but now i get 2 CORS. No reinstal, only reboot.

closed 2023-10-06 by github-actions[bot]