esphome/issues #4443: i2s media player cant play local files anymore, radio, TTS works great

Electronlibre2012, 2023-04-21:
### The problem

When i play a mp3 file from my local /media folder, nothing play. In the esp32 media_player its says "playing" but no sound.

Same mp3 file can be played on any other google home.

My i2s media player can play TTS or a lot of radio without issue, so i am sure its not a hardware problem.

i have not use for a while, so i read some report the local files stop working since 2022.11.0 version of HA

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
esphome:
  name: audionotifier
  friendly_name: audionotifier

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 192.168.1.xxx
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
  fast_connect: on
  power_save_mode: none

status_led:
  pin:
    number: GPIO2 #ESP32 OnBroad LED
    inverted: True

sensor:
  - platform: internal_temperature
    name: "Internal Temperature AudioNotifier"  

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23

voice_assistant:
  microphone: echo_microphone

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO39
      inverted: False
    name: ${friendly_name} Button
    id: echo_button
    on_press:
      - voice_assistant.start:
    on_release:
      - voice_assistant.stop:

media_player: 
  - platform: i2s_audio
    name: audionotifier Media Player
    id: echo_audio
    dac_type: external
    i2s_dout_pin: GPIO22
    mode: mono
```

### Anything in the logs that might be useful for us?

```txt
no error
```

### Additional information

22:34:12][D][media_player:066]:   Media URL: http://192.168.1.50:8123/media/local/sikmu/A%20Reminiscent%20Drive/Ambrosia/Ambrosia.mp3?authSig=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiI2YmQ4ZWFmZGMyZmE0YjNkYWNkODc1Y2ZjZDQxZDU3YiIsInBhdGgiOiIvbWVkaWEvbG9jYWwvc2lrbXUvQSBSZW1pbmlzY2VudCBEcml2ZS9BbWJyb3NpYS9BbWJyb3NpYS5tcDMiLCJwYXJhbXMiOltdLCJpYXQiOjE2ODIxMDIwNTIsImV4cCI6MTY4MjE4ODQ1Mn0.qnS-O37sXv-E58lQsRoAm97rDp7NmbvslURks_ICvws

RynoM, 2024-03-24:
Similar issue, "Component i2s_audio.media_player took a long time for an operation" but has worked once or twice in the past with the same hardware and config.

RynoM, 2024-04-02:
Turns out for me the cause was trying to play https media urls, which silently fail.
See e.g. https://github.com/esphome/feature-requests/issues/2179

newpond, 2024-04-26:
try popping this in your configuration.yaml and putting your media files in /config/www/media folder.