esphome/issues #4444: i2s_audio component missing from esphome all of a sudden

wbarber69, 2023-04-21:
### The problem

getting this error from esphome when trying to update yaml that has not been changed or altered for years

"Component media_player.i2s_audio requires component i2s_audio"

### Which version of ESPHome has the issue?

2023.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

Dont know its the espmuse luxe

### Component causing the issue

i2s_audio

### Example YAML snippet

```yaml
media_player:
  - platform: i2s_audio
    name: muse_luxe
    dac_type: external
    i2s_lrclk_pin: GPIO25
    i2s_dout_pin: GPIO26
    i2s_bclk_pin: GPIO5
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp-muse-luxe-382d3c.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3552/head
WARNING GPIO12 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

media_player.i2s_audio: [source /config/esphome/esp-muse-luxe-382d3c.yaml:52]
  
  Component media_player.i2s_audio requires component i2s_audio.
  platform: i2s_audio
  name: muse_luxe
  dac_type: external
  i2s_lrclk_pin: GPIO25
  i2s_dout_pin: GPIO26
  i2s_bclk_pin: GPIO5
  mode: stereo
  mute_pin: 
    number: GPIO21
    inverted: True
```

ssieb, 2023-04-21:
Read the release notes.

ssieb, 2023-04-21:
Also, i2s audio has not been available for long, so you must have changed it recently.

closed 2023-04-21 by ssieb

wbarber69, 2023-04-21:
It’s part. of esp muse luxe configuration and I’ve not changed any part of that since I bought it.  I recently tried to update it from the esphome dashboard and I am getting these errors. However long it’s been available matters not.  Don’t close my issues unless they are solved.

wbarber69, 2023-04-21:
I see nothing in the decades worth of release notes that says anything about breaking my configuration that was the default for my espmuse luxe device.  Stop being cryptic and try to actually help.

ssieb, 2023-04-21:
https://esphome.io/changelog/2023.4.0.html#i2s-media-player
It's not going to be "solved".