esphome/issues #4446: i2s_audio component missing from esphome

wbarber69, 2023-04-21:
### The problem

Failed config

media_player.i2s_audio: [source /config/esphome/esp-muse-luxe-382d3c.yaml:52]
  
  Component media_player.i2s_audio requires component i2s_audio.
  platform: i2s_audio
  name: muse_luxe
  dac_type: external
  i2s_lrclk_pin: GPIO25
  i2s_dout_pin: GPIO26
  i2s_bclk_pin: GPIO5
  mode: stereo
  mute_pin: 
    number: GPIO21
    inverted: True

### Which version of ESPHome has the issue?

2023.40

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.4.5

### What platform are you using?

ESP32

### Board

Espmuse luxe

### Component causing the issue

I2s_audio

### Example YAML snippet

```yaml
media_player:
  - platform: i2s_audio
    name: muse_luxe
    dac_type: external
    i2s_lrclk_pin: GPIO25
    i2s_dout_pin: GPIO26
    i2s_bclk_pin: GPIO5
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true
```

### Anything in the logs that might be useful for us?

```txt
Failed config

media_player.i2s_audio: [source /config/esphome/esp-muse-luxe-382d3c.yaml:52]
  
  Component media_player.i2s_audio requires component i2s_audio.
  platform: i2s_audio
  name: muse_luxe
  dac_type: external
  i2s_lrclk_pin: GPIO25
  i2s_dout_pin: GPIO26
  i2s_bclk_pin: GPIO5
  mode: stereo
  mute_pin: 
    number: GPIO21
    inverted: True
```

ssieb, 2023-04-21:
Don't open the same issue again just because you don't like the answer.

closed 2023-04-21 by ssieb

wbarber69, 2023-04-21:
How about actually helping people instead of being a dick