esphome/issues #4447: Pause does not work in i2s media player

dab0g, 2023-04-22:
### The problem

Pause does not work in i2s media player (when playing a local file, streaming). Commands are displayed in the esphome logs, but they do not perform any actions. Used - MAX98357A. In older versions, the pause worked.

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.4.6

### What platform are you using?

ESP32

### Component causing the issue

i2s_audio media_player

### Example YAML snippet

```yaml
i2s_audio:
    i2s_lrclk_pin: GPIO5
    i2s_bclk_pin: GPIO2

media_player:
  - platform: i2s_audio
    name: $devicename I2S Media Player
    dac_type: external
    i2s_dout_pin: GPIO4
    mode: mono
    on_state:
      - logger.log: "State updated!"    
    on_play:
      - logger.log: "Playback started!"    
    on_pause:
      - logger.log: "Playback paused!"      
    on_idle:
      - logger.log: "Playback finished!"
```

### Anything in the logs that might be useful for us?

```txt
[09:14:40][D][media_player:059]: 'esp32-audio-4 I2S Media Player' - Setting
[09:14:40][D][media_player:063]:   Command: PAUSE
[09:14:40][D][main:241]: State updated!
[09:14:40][D][main:259]: Playback paused!
[09:14:43][D][media_player:059]: 'esp32-audio-4 I2S Media Player' - Setting
[09:14:43][D][media_player:063]:   Command: PLAY
[09:14:43][D][main:241]: State updated!
[09:14:43][D][main:253]: Playback started!
```

kreamuhely, 2023-04-30:
I confirm, it doesn't work for me either!
The "- media_player.stop" action is works, but "pause" or "toggle" is not.
Nothing has changed in the hardware, esphome does not work after updating.

SeByDocKy, 2023-05-02:
I can confirm too the same bug ....

AlirezaT, 2023-05-06:
same for me

jesserockz, 2023-05-09:
I have only just had time to look into this today. I have found the issue which was a change in the library code, which I upgraded in November 2022, so interesting it was not reported earlier. 

In any case I hope to have it fixed for the beta this week.

closed 2023-05-09 by jesserockz