esphome/issues #4449: Issue with v2023.4.0 and fastled_clockless

asafmatan, 2023-04-22:
### The problem

I get this error while comping the yaml (since v2023.4.0):

INFO Reading configuration /config/esphome/garden_led_strip.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING 'garden_led_strip': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing garden_led_strip (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
|-- FastLED @ 3.3.2
Linking /data/garden_led_strip/.pioenvs/garden_led_strip/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa@2.40802.200502/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/garden_led_strip/.pioenvs/garden_led_strip/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o:(.text._ZN7esphome6socket11LWIPRawImplD2Ev[esphome::socket::LWIPRawImpl::~LWIPRawImpl()]+0x4): undefined reference to `vtable for esphome::socket::Socket'
collect2: error: ld returned 1 exit status
*** [/data/garden_led_strip/.pioenvs/garden_led_strip/firmware.elf] Error 1
========================== [FAILED] Took 4.19 seconds ==========================

### Which version of ESPHome has the issue?

2023.4.0

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

nodemcuv2

### Component causing the issue

socket

### Example YAML snippet

```yaml
esphome:
  name: garden_led_strip
  platform: ESP8266
  board: nodemcuv2
  arduino_version: 2.7.4

wifi:
  ssid: "Home"
  password: "81911234"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "AIiJBbLf9SPt5pfQ0piCNn5VDKG//L/mAcUnG5Cz6v4="

ota:
  password: "8191am00"

# Enables Web Server
web_server:
  port: 80

# Define LED strip connected and amount of LEDS
light:
  - platform: fastled_clockless
    chipset: ws2812b
    pin: GPIO5
    num_leds: 300
    rgb_order: GRB
    name: "GardenDigitalLEDStrip"
    effects:
      - addressable_rainbow:
      - addressable_rainbow:
          name: Rainbow Effect With Custom Values
          speed: 10
          width: 512
      - addressable_rainbow:
          name: Rainbow mega fast
          speed: 20
          width: 150
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_fireworks:
```

### Anything in the logs that might be useful for us?

```txt
Error is in linking stage so no logs beyond the above
```

ssieb, 2023-04-22:
It tells you right at the top: `WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.`
Your Arduino framework version is too old and isn't compatible.

closed 2023-04-22 by ssieb

asafmatan, 2023-04-22:
Thanks, when I remove the manual version I get this:
Failed config

light.fastled_clockless: [source /config/esphome/garden_led_strip.yaml:29]
  
  This feature requires framework version 2.7.4 or lower. Please see note on documentation for FastLED.

What would be a replacement for FastLed?

ssieb, 2023-04-22:
There's ongoing work to get a newer version working, but you could try using the neopixel component instead.

asafmatan, 2023-04-22:
Thanks for your advice, I switched for now to neopixelbus, all works fine.

ssieb, 2023-05-03:
This is temporarily fixed in the latest point release that is just coming out.