esphome/issues #4451: ESP32 will not wake from deep sleep

deanfourie1, 2023-04-23:
### The problem

The ESP32 wakes briefly from deep sleep but then resumes to sleep instantly again.

### Which version of ESPHome has the issue?

latest

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

latest

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

deepsleep

### Example YAML snippet

```yaml
esphome:
  name: rf-transmitter
  friendly_name: rf-transmitter

  on_boot:
    priority: 200
    then:
      - logger.log: Waiting for API connection...
      - wait_until:
          condition:
            api.connected:
          timeout: 10s
      - logger.log: Waiting for time...
      - wait_until:
          condition:
            time.has_time:
          timeout: 10s
      - if:
          condition:
            - sun.is_below_horizon:
          then:
            - deep_sleep.enter:
                id: deep_sleep_1
                until: "23:15:00"
                time_id: ha_time

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:
    
time:
  - platform: homeassistant
    id: ha_time

sun:
 # your location
  latitude: -36.848461
  longitude: 174.763336

deep_sleep:
  id: deep_sleep_1
```

### Additional information

Im trying to setup deep sleep to, if on first boot and sunset in below_horizon, then enter deep sleep until xx:xx time, then exit deep sleep until sun goes below_horizon again.

This is not working, when the ESP boots, it appears to enter deep sleep (goes offline). It then briefly wakes up at the specified until time (comes online) then immediately goes back to sleep (goes offline again).

I'm not sure if this could be a look in the on_boot code, though I do doubt it.

Possibly a bug?

Thanks

ssieb, 2023-04-23:
Watch the serial logs to see what is happening.

donoghdb, 2023-08-29:
@deanfourie1 I have just come across the same issue. I have narrowed it down to when the time is specified to wake up. If I use the normal deep sleep its fine and if I use the specified time to sleep it also works. So its an issue with the specified time. I am triggering off the time component.